trabucayre/openFPGALoader #154: Flash programmation on Arty A7 100T

obpt, 2021-12-20:
I am trying to program the flash of an Arty 100T. The spiOverJtag module does not exist for the FPGA which is on this board, so I created one:
https://github.com/obpt/openFPGALoader/tree/arty-100t

Then I try to program with the following command:
```shell
$ openFPGALoader -b arty --fpga-part xc7a100tcsg324  bitstream.mcs -f --verify
write to flash
Board default fpga part overridden with xc7a100tcsg324
Jtag frequency : requested 10.00MHz  -> real 10.00MHz 
Open file DONE
Parse file DONE
use: /home/obpt/utils/share/openFPGALoader/spiOverJtag_xc7a100tcsg324.bit.gz
load program
Flash SRAM: [===================================================] 100.00%
Done
Detail: 
Jedec ID          : 01
memory type       : 20
memory capacity   : 18
EDID + CFD length : 4d
EDID              : 8001
CFD               : 
Detail: 
Jedec ID          : 01
memory type       : 20
memory capacity   : 18
EDID + CFD length : 4d
EDID              : 8001
CFD               : 
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Fail
Verification failed at 2097152
```

I am able to program the same mcs file with Vivado.

Do you have an idea what I did wrong?

trabucayre, 2021-12-21:
Oups!
I have not seen your issue before and just pushed arty a7 100t support, sorry.
It's weird I have not seen this flash issue before. Could you share your mcs file ?
Thanks!

trabucayre, 2021-12-21:
I have redone tests and I'm able to reproduce your issue.
In fact this device seems not support 4Kb erase everywhere. I have added a workaround.
This update is not in master branch but in rework_spiflash
It's okay for you?

obpt, 2021-12-21:
Just a quick update.

I am sorry, but I cannot provide you the mcs I am using currently (proprietary stuff). But I will try to generate a sharable one in my spare time.

I have tried the version from the branch `rework_spiflash`, openFPGALoader seems happy:

```console
$ openFPGALoader -b arty_a7_100t -f --verify
write to flash
Jtag frequency : requested 10.00MHz  -> real 10.00MHz 
Open file DONE
Parse file DONE
use: /home/obpt/utils/share/openFPGALoader/spiOverJtag_xc7a100tcsg324.bit.gz
load program
Flash SRAM: [===================================================] 100.00%
Done
Detected: spansion S25FL128S 256 sectors size: 128Mb
00000000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
```
But the FPGA does not reconfigure.

It is strange that the verification ends successfully while the FPGA is not able to load the bitstream from the flash.

The only idea (for now) is that maybe (peut-être!) bit or byte swapping is involved somewhere in the process.

Anyway, I will come back to you with a sharable mcs file.

trabucayre, 2021-12-21:
It's a good news for verification.
I may try to generate by myself an mcs file to try, if you can't.
Thanks

trabucayre, 2021-12-21:
I have converted a .bit to a .mcs file according to [these instructions](https://trabucayre.github.io/openFPGALoader/vendors/xilinx.html).
Result is okay with my arty A7 100t. FPGA reload new bitstream from flash without any issues.
mcs is not required, you can use bit or bin files if mcs are definitively not working on your case.

obpt, 2021-12-22:
So... there was a hardware configuration configuration issue on my side: the jumper _SPI mode_ was not connected, preventing the FPGA to load the data from the SPI flash.

So I confirm that the fix from `rework_spi` works for a Arty A7 100T.

Thank you very much for this very useful tool and the time to spend on it.

trabucayre, 2021-12-22:
Great news!
Thanks for your feedback and catch!
Happy to help!

trabucayre, 2022-02-12:
It's okay for  you?
Can I close this issue ?
Thanks

closed 2022-02-18 by trabucayre