trabucayre/openFPGALoader #159: Support for Terasic DE1-SoC board with Cyclone V

sedwards-lab, 2022-01-11:
The Terasic DE1-SoC board has a Cyclone V on it. Specifically, a 5CSEMA5F31C6 connected through an onboard usb-blasterII

Here's the two-line patch to add support for this board:

```
diff --git a/src/board.hpp b/src/board.hpp
index 6750ec0..fefd67e 100644
--- a/src/board.hpp
+++ b/src/board.hpp
@@ -122,6 +122,7 @@ static std::map <std::string, target_board_t> board_list = {
        JTAG_BOARD("de0",             "", "usb-blaster",0, 0, CABLE_DEFAULT),
        JTAG_BOARD("de0nano",         "ep4ce2217", "usb-blaster",0, 0, CABLE_DEFAULT),
        JTAG_BOARD("de0nanoSoc",      "", "usb-blasterII",0, 0, CABLE_DEFAULT),
+       JTAG_BOARD("de1Soc",          "5CSEMA5", "usb-blasterII",0, 0, CABLE_DEFAULT),
        JTAG_BOARD("de10nano",        "", "usb-blasterII",0, 0, CABLE_DEFAULT),
        JTAG_BOARD("ecp5_evn",        "", "ft2232",     0, 0, CABLE_DEFAULT),
        SPI_BOARD("fireant",              "efinix", "ft232",
diff --git a/src/part.hpp b/src/part.hpp
index 55dc6cd..c9934c5 100644
--- a/src/part.hpp
+++ b/src/part.hpp
@@ -60,6 +60,7 @@ static std::map <int, fpga_model> fpga_list = {
        {0x02b150dd, {"altera", "cyclone V", "5CEA2", 10}},
        {0x02b050dd, {"altera", "cyclone V", "5CEBA4", 10}},
        {0x02d020dd, {"altera", "cyclone V Soc", "5CSEBA6", 10}},
+       {0x02d120dd, {"altera", "cyclone V Soc", "5CSEMA5", 10}},
        {0x02d010dd, {"altera", "cyclone V Soc", "5CSEMA4", 10}},
 
        {0x00000001, {"efinix", "Trion",    "T4/T8",            4}},
```

Incidentally, the FPGA is second on the JTAG chain, which the tool seems to handle correctly:

```
USB-Blaster II firmware version: 1.39
index 0:
	idcode   0xba00477
	type     ARM cortex A9
	irlength 4
index 1:
	idcode 0x2d120dd
	manufacturer altera
	family cyclone V Soc
	model  5CSEMA5
	irlength 10
```

Also, your instructions for running quartus_cpf for converting .sof to .svf gave me an error (bad arguments). Here's what worked for me with Quartus Prime 21.1.0:

```
quartus_cpf -c --operation p -g 3.3 --freq 24.0MHz lab1.sof lab1.svf
```

I don't know how to create a private branch that could turn into a pull request

trabucayre, 2022-01-12:
Thanks for these informations.

For svf the correct line is
```
quartus_cpf -c -q -g 3.3 -n 12.0MHz p project_name.sof project_name.svf
```
As mentioned [here](https://trabucayre.github.io/openFPGALoader/vendors/intel.html). But using an `.rbf` it's a bit faster to load.

To create a PR:
1.  you have to fork this repository
2. clone your fork
3. `git checkout -b nameOfYourBranch`
4. apply modification + `git add files` + `git commit -m "message"` + `git push`
5. in github page create a PR.

Thanks again!

trabucayre, 2022-02-13:
Now the board is officially supported. So I can close this issue.

closed 2022-02-13 by trabucayre