trabucayre/openFPGALoader #164: Quiet option

eblot, 2022-01-19:
Hi,

Is there a way to make openFPGALoader. The `--quiet` version (#73) used to work, but with v0.6.1 at least, openFPGALoader reverted back to a very verbose mode whatever the `--quiet` or `--verbose-level -1` is selected.

Thanks.

All the low level JTAG traces are printed out:
```
writeTMS 6 1
writeTMS 4 1
writeTDI len : 32 32 4 0

13 63 10 93
writeTDI len : 32 32 4 0

ff ff ff ff
writeTDI len : 32 32 4 0

ff ff ff ff
writeTDI len : 32 32 4 0

ff ff ff ff
writeTDI len : 32 31 3 7
writeTDI read/write 7 bit
writeTDI last_bit ff size 6

ff ff ff ff
writeTDI move to EXIT1_xx and send last bit 81
writeTMS 6 1
Open file DONE
Parse file DONE
load program
writeTMS 6 1
writeTMS 5 1
writeTDI len : 6 5 0 5
writeTDI read/write 5 bit
writeTDI last_bit b size 4
writeTDI move to EXIT1_xx and send last bit 1
writeTMS 2 1
writeTMS 4 1
writeTDI len : 6 5 0 5
writeTDI read/write 5 bit
writeTDI last_bit ff size 4

writeTDI move to EXIT1_xx and send last bit 81
writeTMS 2 1
writeTMS 4 1
writeTDI len : 6 5 0 5
writeTDI read/write 5 bit
writeTDI last_bit 5 size 4
writeTDI move to EXIT1_xx and send last bit 1
writeTMS 2 1
writeTMS 1 1
writeTMS 2 1
writeTDI len : 306056 306056 38257 0
Flash SRAM: writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 306056 306056 38257 0
writeTDI len : 704 703 87 7
writeTDI read/write 7 bit
writeTDI last_bit 0 size 6
writeTDI move to EXIT1_xx and send last bit 1
writeTMS 1 1
Done
writeTMS 1 1
writeTMS 4 1
writeTDI len : 6 5 0 5
writeTDI read/write 5 bit
writeTDI last_bit c size 4
writeTDI move to EXIT1_xx and send last bit 1
writeTMS 1 1
writeTMS 1 1
writeTMS 6 1
```

trabucayre, 2022-01-19:
Thanks for this issue.
I have fixed verbose signed/unsigned type and now debug message aren't displayed if verbose-level < 2
It's good for you ?
Sorry for the noise :)

eblot, 2022-01-20:
Thanks for the quick fix. I will try to give it a try by the end of this week.

trabucayre, 2022-01-20:
Perfect!
Thanks again!

eblot, 2022-02-08:
Sorry for the late reply. I can confirm that the problem is now fixed. Thanks again.

closed 2022-02-08 by eblot

trabucayre, 2022-02-08:
Thanks for your feedback.