trabucayre/openFPGALoader #167: unknown JTAG ID

kimstik, 2022-01-21:
I got "JTAG init failed with: Unknown device with IDCODE: 0x0000281b"

You may print a bit more about unknown devices. Like it made here:
https://github.com/openocd-org/openocd/blob/master/src/helper/jep106.inc

BTW blue looks terrible on black background.. What about options? bw, green?

trabucayre, 2022-01-21:
It's true. I have to add a better display for unknown device and adding manufacturer ID to help a bit people.
For blue color I have a blak background too :) But I have to search for a ligther blue, green is already used.

0x0000281b is GW2A-55: don't fear when error is displayed for checksum: I have to find howto to compute this one (it's different as GW1N)

trabucayre, 2022-02-09:
I have pushed commit to improve error message and change blue color.
It's good to you ?

closed 2022-04-04 by trabucayre