trabucayre/openFPGALoader #169: Add support for Spartan-6 xc6slx16 with csg324 footprint

DaveBerkeley, 2022-01-26:
I would do a pull-request for this, but I don't have the quartus_sh tool I need to generate the spiOverJtag files. 

You kindly added the xc6slx16 ftg256 recently. This is working very well.

DaveBerkeley, 2022-01-26:
I've just realised that this board currently has a Platform Flash JTAG prom device on it currently, not a SPI flash.

So I can't currently test the hardware either.

trabucayre, 2022-01-27:
for Spartan6 devices it's **ISE** ;-)
I can generates this bitstream if you want.
Which JTAG prom device ? openFPGALoader support XCF parts. Have you public schematics ?
Thanks

DaveBerkeley, 2022-01-27:
I just ran "make" so it tried to build all the targets, hence the problem with quartus. I've now run "make spiOverJtag_xc6slx16csg324.bit.gz" which seems to work. I'll test that today programming the FPGA and confirm if it works. I can do a pull request for this.

I'll see if it is okay to share parts of the schematics, hopefully later today.

Great that it supports XFC parts : if only we could buy some.

DaveBerkeley, 2022-02-13:
Yes, working fine. xc6slx16csg324 also tested with XFC.

Many Thanks

closed 2022-02-13 by DaveBerkeley