trabucayre/openFPGALoader #175: [Request] Add support for Segger J-Link cable (and/or clones) based on STM32

carlosedp, 2022-02-09:
It is a nice generic cable that can be used for multiple boards based on STM32.

How hard would be to add support to it?

Refs:
- https://www.segger.com/products/debug-probes/j-link/
- https://www.aliexpress.com/item/1005002266152514.html?spm=a2g0o.cart.0.0.5d473c004xfzgu&mp=1

trabucayre, 2022-02-09:
Yep, it's true. I think to order one unit (not a clone) since a while.
It's not really hard (theoritically, reality is always less easy) 
- a cable class must inherit from [jtagInterface](https://github.com/trabucayre/openFPGALoader/blob/master/src/jtagInterface.hpp)
- and it must known by the [jtag class](https://github.com/trabucayre/openFPGALoader/blob/master/src/jtag.cpp#L85)

I know a bit of documentation to this topic is required...

trabucayre, 2022-02-11:
I've checked again price and finally I will order one on aliexpress

carlosedp, 2022-02-11:
Awesome, I'm about to get a Kintex7 FPGA (from Ali as well ;) ) and might get the Segger too :)
Let me know once it arrives!

trabucayre, 2022-02-11:
I have to think for kintex too (but it's maybe not reasonable...)
I think I have to wait for around 2 weeks for delivery to receive my Segger :)

trabucayre, 2022-02-26:
I have received my jlink probe \o/
I have tried with a subset of command to see if working: now I have to implement the rest (with algorithm quite similar to cmsisDAP).
I try to do it ASAP but I have a paper to finish and SPI flash issue to fix :-)

trabucayre, 2022-03-12:
I have integrated the code into openFPGALoader and tested with an altera FPGA -> OK
Now I need to check with others devices to fully validate implementation and adding cleanup/comment into code before pushing this support.

trabucayre, 2022-03-12:
Support is now available. It's requires more test to check everything.

carlosedp, 2022-03-14:
That's awesome. I've recently ordered one so it's not here yet.
I'll report back once I test it out.
Thanks a lot for the quickness of implementing it! :D

trabucayre, 2022-03-15:
> I'll report back once I test it out.

Great! It's usually good to have user's report with different use case / setup to validate support or to find bug.

> Thanks a lot for the quickness of implementing it! :D

I'm not sure I was fast to implement that :)

trabucayre, 2022-05-04:
Hi,
Any news ?
Thanks

paller, 2022-07-23:
I'm trying to test with a [J-Link EDU Mini](https://www.segger.com/products/debug-probes/j-link/models/j-link-edu-mini/) against the Tang Nano jtag interface.

First issue is the [PID](https://github.com/trabucayre/openFPGALoader/blob/057ce931f855235ab1f22c8c0a2980db745b07ad/src/jlink.cpp#L23), instead of 0x0105 it's 0x0101. I don't know if they should be compatible, but if so, it would be nice with multiple PIDs allowed for the J-Link cable.

After changing the PID, I get this output
```
% sudo ~/openFPGALoader/build/openFPGALoader -c jlink --freq 1000000 --fpga-part GW1N-1 --verbose-level 2 ~/neopixel_tang_nano.fs
1366:0101 (bus 2, device 32)
b8ea5a33
 0 EMU_CAP_RESERVED
 1 EMU_CAP_GET_HW_VERSION
 4 EMU_CAP_READ_CONFIG
 5 EMU_CAP_WRITE_CONFIG
 9 EMU_CAP_SPEED_INFO
11 EMU_CAP_GET_MAX_BLOCK_SIZE
12 EMU_CAP_GET_HW_INFO
14 EMU_CAP_RESET_STOP_TIMED
17 EMU_CAP_SELECT_IF
19 EMU_CAP_GET_COUNTERS
21 EMU_CAP_GET_CPU_CAPS
22 EMU_CAP_EXEC_CPU_CMD
23 EMU_CAP_SWO
27 EMU_CAP_REGISTER
28 EMU_CAP_INDICATORS
29 EMU_CAP_TEST_NET_SPEED
31 Reserved
device type: J-Link major: 1 minor: 0 revision: 0
Raw IDCODE:
Error write -4
LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
fails to send buffer
JTAG init failed with: fails to send buffer
```

trabucayre, 2022-07-24:
I have to check but it if EDU mini has a different VID/PID it may made sense to add a new entry. It's always possible to allows VID/PID overrides but I'm a bit not convinced here.
I have to search a bit docs about this specific model (features limits or something like this). This driver was written using a clone and sometime those type of copies are a bit differents to the official.

paller, 2022-07-24:
I'm trying to figure out if I did the wiring correctly. Connecting TMS to TMS, TDI to TDI, and so on, really shouldn't be that difficult, but I can't read the IDCODE over jtag with [JTAGLoad](https://wiki.segger.com/JTAGLoad) and a simple .svf file.

trabucayre, 2022-07-24:
Ok: it seems to have two distinct issue:
1. with openFPGALoader an error is produces at libusb level (but it's maybe something wrong at higher (openFPGALoader) level)
2. with official tool you have no errors but you can't access device
True?

paller, 2022-07-25:
With the official Gowin Programmer it works fine for both RAM and flash, which is using the ch552 chip. I was trying to use jlink as an external probe, but I guess it can't work while the ch552 is also connected.

trabucayre, 2022-07-25:
ch552 firmware was implemented to works with gowin programmer (and there is some specifics hooks for gowin).
You can't use external probe with the ch552 not disabled (or not in reset state).

carlosedp, 2022-09-14:
@trabucayre sorry about the absurdely delayed answer. I've moved houses in the past months and some of my stuff got packed away (including the segger cable by accident).

trabucayre, 2022-09-14:
Don't worry. There is sometime good reasons to postpone some tasks :)
I have, also, to find a way to test mini edu model.