trabucayre/openFPGALoader #181: Kintex7 boot time when loading from flash

unbtorsten, 2022-02-22:
_Original discussion in https://github.com/trabucayre/openFPGALoader/issues/180#issuecomment-1048030988_ and following.

Expedite booting from flash by
- add bitstream compression
- configure quad mode

jrrk2, 2022-02-22:
I can see a benefit in download time of the spiOverJtag bit image to ram due to this technique. However I don't think quad mode in the Vivado project will do anything for speed of the eventual BitStream load from flash. We can do diffs between .MCS files from Vivado with and without the quad mode to see what the difference is and whether you can exploit that in your bitstream to flash converter,

unbtorsten, 2022-02-22:
@jrrk2, @trabucayre: Including configuration above does not make a difference for boot time. This agrees with jrrk2's comment above.

trabucayre, 2022-02-22:
> @jrrk2, @trabucayre: Including configuration above does not make a difference for boot time. This agrees with jrrk2's comment above.

Quad mode + compression has no impact to boot time? It's a strange thing. Make sense to dump bitstream for comparison.

jrrk2, 2022-02-22:
But for the bitstream written in flash it is also useless at the moment because the open source toolchain does not know to treat this option.

jrrk2, 2022-02-22:
However I would expect the spiOverJtag bitstream to load quicker. But the compression ratio will be strongly dependent on the number of PIPs that are turned on, which is next to nothing with this design.

jrrk2, 2022-02-22:
write_cfgmem  -format bin -size 32 -interface SPIx4 -loadbit "up 0x00000000 $bitfile" -file "$binfile"

As per this command, we can see the quad mode is introduced during conversion from bitstream to binfile/mcsfile. So this is where we need to look to gain performance, rather than in the spiOverJtag Vivado settings

closed 2022-02-22 by unbtorsten

trabucayre, 2022-02-22:
Feel free to reopen this one if something may be done at openFPGALoader level