trabucayre/openFPGALoader #187: JTAG init failed with: Unknown device with IDCODE: 0x1100481b (TangNano9k)

DxPru, 2022-03-08:
HI

I tried flashing an TangNano9k, but every time it returns the same error.

```
[dxpru@manjaro pnr]$ openFPGALoader -v -m -b tangnano9k bitstream.fs
write to ram
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
JTAG init failed with: Unknown device with IDCODE: 0x1100481b
```

The output doesn't even change when I tried different boards (tangnano, tangnano1k, tangnano4k, littleBee, ...) or even when I didn't specify params.

Thanks in advance

trabucayre, 2022-03-08:
Hi,
Could you provides openFPGALoader's version?
```
openFPGALoader -V
```
9K is a recent addition, no release have this entry -> you must use master branch.
Using -b is to have access to informations like cable, but detection isn't based on that.

DxPru, 2022-03-08:
Hey,

I've got version `v0.5.0` from the offical Repo.

Now I tried with a newer version ( `v0.7.0` ) from the openfpgaloader-git aur repo and it worked.
Thank you

Martoni, 2022-03-09:
Have a problem with tangnano9k : 

```
$ openFPGALoader -m -b tangnano9k impl/pnr/hdmicore.fs 
write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse impl/pnr/hdmicore.fs: 
Done
DONE
Error: Failed to claim FPGA device: mismatch between target's idcode and fs idcode
```

Martoni, 2022-03-09:
I have an id mismatch with tangnano9k with last version : 

```
$ openFPGALoader -f -b tangnano9k impl/pnr/hdmicore.fs 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse impl/pnr/hdmicore.fs: 
Done
DONE
fs_idcode 0x1100581b, idcode 0x100481b
Error: Failed to claim FPGA device: mismatch between target's idcode and fs idcode
```

trabucayre, 2022-03-10:
Hi,
According to http://cdn.gowinsemi.com.cn/UG290.pdf p38, your bitstream is for an **GW1N(R)-9** but your device is a **GW1N(R)-9C**. Are you sure to have checked the good one ?
My Gowin license is expired I have to request for a new one.

trabucayre, 2022-03-10:
In fact this issue is not really openFPGALoader related: fs_idcode come from bit file, and device idcode is read through JTAG in `IDCODE` register.

Martoni, 2022-03-10:
Thanks @trabucayre that works now.
Same gowin part number  (GW1NR-LV9QN88PC6/I5) can designate two different chip ...

trabucayre, 2022-03-10:
It's something easy to manage :)

closed 2022-03-24 by trabucayre