trabucayre/openFPGALoader #188: FTDI JTAG edge selection

smunaut, 2022-03-09:
* The write edge should always be negative edge, there is no reason to swap it at higher close rate. The FTDI drives the clock, so the relative alignement of the data edge vs the clock edge doesn't change depending on clock rate.
* The read edge could be made to always be the negative edge independent of the clock rate. "In theory", JTAG is sampled on the rising edge , but that's really only for the device side, to avoid hold time issues. And at higher clock rate that becomes a problem because between the various board/IC delays, there isn't enough time between the negative edge seen by the device and the internal FTDI rising edge for signal to propagate. But the FTDI being the clock master here, it's internal version of the negative edge is a tiny bit before the external negative edge, and there is no way that the device can change the TDO value before the FTDI samples, so it should always be safe to use the negative edge for capture.

I only tested this on the single device I have, but this worked fine up to the max 30 MHz clock rate.

trabucayre, 2022-03-09:
Default behaviour is exactly as you mention (neg edge for write, pos edge for read).
In fact `config_edge` is a workaround: I have observed not working/corrupted communications with (at least) arty7 35 and artyA7 100T when frequency is higher than 15MHz. This why for these boards read/write edges are inverted when freq > 15Mhz. For any others boards / adapters default behaviour works fine at FTDI's max frequency (maybe issue with level shifters... I don't know)
But maybe this WA has side effect to another ones and instead of usign USB `iProduct` I must add a parameter to let to the user the choice to use normal edges or opposite.

trabucayre, 2022-03-09:
I don't remember...
Anyway. I've tried with arty A7 100. Using neg edge for read and write works as expected with 6MHz, 15MHz & 30MHz.
I have to extend the frequency range and test with all (6) boards to verify.
The question is: is this WA only required for arty or affect all boards with this interface?
I think to improve this WA to handle this specific use case and adding CLI args to change read edge.

smunaut, 2022-03-09:
What do you mean by "WA" ?

My target currently is a Digilent HS-3 connected to a Zynq MP board.

trabucayre, 2022-03-09:
> What do you mean by "WA" ?

Work Around: because JTAG protocol said: write on neg and read on pos. But to achieve max FTDI's frequency you have to play with the read edge.

> My target currently is a Digilent HS-3 connected to a Zynq MP board.

It's interesting to see if it's always true (board and device) or specific to one board (or a subset). In first case a config fix for this probe must be done, in second case it's setup specific and consequently it's to the user to set this according to hardware. Unfortunately I haven't this probe at home to check by myself...

trabucayre, 2022-03-10:
I have updated `config_edge`: now write edge is not anymore modified and is left on neg. read edge is only affected for arty.

smunaut, 2022-03-10:
:+1: 

FWIW, this also switches it on the Digilent JTAG-HS3 that I use.

closed 2022-03-10 by smunaut

trabucayre, 2022-03-10:
Good! But it's a side effect because at the beginning this fix is only for arty. Have to check all my cables.

Maybe this fix has to be done by cable name instead of by USB `iProduct`.
 
Thanks

smunaut, 2022-03-10:
Pretty sure this isn't arty specific because it's something that openocd offers as option for all FTDI cable and as I explained above, it's sort of expected and should work fine everywhere just the way FTDI works.