trabucayre/openFPGALoader #189: Lattice MachXO3LF daisy cahin

kafiabdellah, 2022-03-09:
Hi, 
I am trying to use openFPGALoader to program two MachXO3LF Starter boards in daisy chain configuration using FTDI FT4232H. I am able to program the second board (the last one in the chain), but without success for the the FPGA with the index 0. After some inspection, i am assuming that there some issues with the pollBusyFlag() method. I have carried out another test using 3 devices in a daisy chain and i was able to program the last device in the chain. Also to eliminate any hardware related issues, I have tried the same setup with Lattice diamond and it works fine.

trabucayre, 2022-03-09:
Hi,
I have to test. I have only one machXO3LF but I assume it is possible to mix machXO2 & machXO3. `pollBusyFlag` is one of function where read is required, maybe something here. I do that ASAP.
As a first try could you change [this line](https://github.com/trabucayre/openFPGALoader/blob/master/src/ftdiJtagMPSSE.cpp#L102) by
```
_read_mode = MPSSE_READ_NEG;
```

Thanks

joergfoertsch, 2023-09-14:
Hi,
Is there any update on the issue yet? Because, if I see correctly the problem still persists, right?
Many thanks :)

trabucayre, 2023-09-14:
Hi,
I have received required hardware (thanks @michaeltraxler) and modified to disable onboard jtag (just validated if hack is working).
Now I am able to check/fix this issue. its more or less only a question of free time :)
Thanks!

commit 94b62460 (2023-09-17): jtag: shiftDR: (fix daisy chain) when more than one FPGA, a sequence of '0' before and/or after must be sent instead of '1' (fix #189 and #133
closed 2023-09-17 by trabucayre