trabucayre/openFPGALoader #195: Crash under msys2

balika011, 2022-03-13:
After installing libusb driver with zadig i only get this error 

```
>.\openFPGALoader.exe -c ft2232 --freq 1000000 --ftdi-channel 0 --detect
 0 [] openFPGALoader 646 cygwin_exception::open_stackdumpfile: Dumping stack trace to openFPGALoader.exe.stackdump
```

content of openFPGALoader.exe.stackdump:
```
Exception: STATUS_ACCESS_VIOLATION at rip=7FFBC5F56D5E
rax=0000000000000000 rbx=00000000FFFFBE10 rcx=00000000FFFFBE10
rdx=0000000000000000 rsi=0000000002B2CBEC rdi=0000000002B2CBF0
r8 =00007FFBC5F778E0 r9 =00007FFBC5F77728 r10=0000000000000000
r11=0000000000000246 r12=0000000002B2CBF8 r13=00000000000002B4
r14=00007FFBC5F778E0 r15=00007FFBC5F77728
rbp=0000000002B2CB80 rsp=0000000002B2CB60
program=C:\Users\user\Desktop\OFL\openFPGALoader.exe, pid 646, thread 
cs=0033 ds=002B es=002B fs=0053 gs=002B ss=002B
Stack trace:
Frame        Function    Args
00002B2CB80  7FFBC5F56D5E (7FFC14CD286B, 00002B2CD20, 00002B2CE00, 000FFFFBE88)
00002B2CC00  7FFBC5F59EFA (00000000000, 00000F1A300, 00000000000, 00000000000)
00000000000  7FFC16C7E1F4 (00000F1A300, 00000000000, 00002B2CE00, 7FFC16C7E250)
00000000000  7FFC16C7E2CC (00000000000, 00000000000, 00000000000, 00000000000)
00000000000  00180047716 (00000000000, 00000000000, 00000000000, 00000000000)
00000000000  001800477C4 (00000000000, 00000000000, 00000000000, 00000000000)
00000000000  7FFC16201350 (00000000000, 00000000000, 00000000000, 00000000000)
00000000000  7FFC17291DC8 (00000000000, 00000000000, 00000000000, 00000000000)
End of stack trace
```

I'm using a xFlasher 360 by The Mod Shop and trying to flash a X360ACE V5 (gowin fpga).

On Linux it works just fine.

I'm not familiar with msys2. Do you have any instructions on further debugging?

trabucayre, 2022-03-14:
Hi,
In fact I'm not familiar with msys2 too... Maybe @umarcor could help on this point.
Thanks

umarcor, 2022-03-14:
CI in this repo seems to be good, but the upstream package was last updated in december. @trabucayre can you tag a new release and I will bump the upstream? Then, we can see if a recent build with recent deps works.

@balika011 meanwhile, you can try the nightly packages: https://github.com/trabucayre/openFPGALoader/releases/tag/nightly. Download the one corresponding to your environment (typically MINGW64 or UCRT64) and install: `pacman -U *.zst`.

trabucayre, 2022-03-15:
@umarcor : Yes of course. I have small fixes to apply before doing that but I will tag a new release ASAP.

Thanks

trabucayre, 2022-03-19:
I have pushed a new release. Sorry to the delay

umarcor, 2022-04-28:
The MSYS2 packages were updated some weeks ago, but I forgot to ping. @balika011 can you please update (`pacman -Syu`) and let us know?

trabucayre, 2022-05-04:
@umarcor thanks for this informations!
@balika011 Any news ?