trabucayre/openFPGALoader #198: [Request] Add support for ECP5 45F

el-coder-sb, 2022-03-22:
Trying to program a ECP5 (LFE5UM)-45F I get the following error:

```
simon@simon-VirtualBox:~/Schreibtisch/openFPGALoader_zeugs$ openFPGALoader -c ft2232 /home/simon/Schreibtisch/openFPGALoader_zeugs/helper_blink_LED_blink_unencry.bit 
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file: DONE
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Fail
displayReadReg
	Config Target Selection : 0
	JTAG Active
	Decrypt Enable
	ISC Enable
	Write Enable
	Read Enable
	Decrypt Only
	Std PreAmble
	Preamble ERR
	EXEC Error
Error: Failed to program FPGA: std::exception
detach error -5

```

Any advice welcome how to fix this. Then I will have a look if I can provide a pull request.

trabucayre, 2022-03-22:
In fact the answer is in #199. Your bitstream seems not encrypted but you FPGA wait for one this this feature enabled.

el-coder-sb, 2022-03-23:
Oh, right. Usually I only work with the flash. Just to try openFPGALoader I was like "trying with SRAM is faster". However in this case the bitstream is only accepted if it is encrypted.

Screenshot from Lattice Diamond Progammer GUI:
[image]

Would be great to have the error message somewhat clearer for this case, comparing "Decrypt Only" and the Header information whether the file is encrypted (as given in #199). I could insert such statement and message right after the line  497 of main.cpp (`printError("Error: Failed to program FPGA: " + string(e.what()));`).

What do you think?

Lattice does it this way:

```
function BURST_PROGRAM {
	FuseArray_TDI.row = 1;	
	IF (ISENCRYPTED == 1)
	{
	}
	ELSE
	{
		TRY 1 {
			! Shift in LSC_READ_STATUS(0x3C) instruction;
			SIR Instruction_Length TDI (LSC_READ_STATUS);
		    SDR_VERIFY	32	TDI(#0^32)
							TDO(0x00000000)
							MASK(0x0000B000);
		}
		ELSE
		{
			! Shift in ISC DISABLE(0x26) instruction;
		    SIR	Instruction_Length 	TDI (ISC_DISABLE);
			RUN_TEST	IDLE  TCK 2 DELAY 200;         
			print (1,"The Encryption Only fuses are programmed. Only Encrypted bitstream can be used.");
		}
	}
```

trabucayre, 2022-03-23:
True. Better error message make sense. Current error display (just a status dump) is a bit cryptic for users.
But my idea is to determine, before doing anything, if bitstream is encrypted and which type of bitstream is required by the FPGA. With these informations it's possible to stop ASAP with a correct message (as done for idcode).

el-coder-sb, 2022-03-28:
As better error message does not belong to this topic and as #199 solves the real problem, I just would like to ask for adding ECP5 LFE5UM-45 to FPGAs.yml before closing this issue.

trabucayre, 2022-03-29:
I have updated FPGAs.yml with all ECP5 models (to be coherent with part.hpp).
Thanks

closed 2022-03-29 by el-coder-sb