trabucayre/openFPGALoader #199: [Request] Add support for encrypted ECP5 bit files

el-coder-sb, 2022-03-22:
When trying to configure a ECP5 with an encrypted bitstream I get the following error:
```
simon@simon-VirtualBox:~/Schreibtisch/openFPGALoader_zeugs$ openFPGALoader -c ft2232 /home/simon/Schreibtisch/openFPGALoader_zeugs/helper_blink_LED_blink.bit 
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file: DONE
Error: missing preamble

Parse file: FAIL
Error: Failed to program FPGA: std::exception
detach error -5

```

This patch is a first step to add support:
```
diff --git a/src/latticeBitParser.cpp b/src/latticeBitParser.cpp
index 64e66f0..fc88b10 100644
--- a/src/latticeBitParser.cpp
+++ b/src/latticeBitParser.cpp
@@ -82,7 +82,8 @@ int LatticeBitParser::parse()
                return EXIT_FAILURE;
 
        /* check preamble */
-       if ((*(uint32_t *)&_raw_data[_endHeader+1]) != 0xb3bdffff) {
+       if ((*(uint32_t *)&_raw_data[_endHeader+1]) != 0xb3bdffff && 
+               (*(uint32_t *)&_raw_data[_endHeader+1]) != 0xb3bfffff) {
                printError("Error: missing preamble\n");
                return EXIT_FAILURE;
        }
```

However I run into next error:

```
simon@simon-VirtualBox:~/Schreibtisch/openFPGALoader_zeugs$ openFPGALoader -c ft2232 /home/simon/Schreibtisch/openFPGALoader_zeugs/helper_blink_LED_blink.bit 
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file: DONE
Parse file: DONE
mismatch between target's idcode and bitstream idcode
	bitstream has 0xFFDD3128 hardware requires 0x01112043
Error: Failed to program FPGA: std::exception
detach error -5
```

Any advice how to fix this welcome. Then I will see if I can upload a pull request.

trabucayre, 2022-03-22:
Hi,
Could you share your bitstream (or a similar one with encryption enabled) ?
In fact I have never played with this feature since I mainly use nextpnr/prjtrellis and I have to ask for a new diamond's license.
Thanks

el-coder-sb, 2022-03-23:
Thanks for the quick reaction.

Here you are:

Lattice Programming File Utility Style for encrypted file:
```
--Lattice Semiconductor Corporation Bitstream
--Version:         Diamond (64-bit) 3.9.1.119
--Bitstream Status: Final Version 10.23
--Design name: schaltschwelle_schaltschwelleLED5_6_7_8.ncd
--Architecture: sa5p00m
--Part: LFE5UM-45F-7CABGA381
--Date: Wed Oct 11 15:05:33 2017
--Rows: 9470
--Cols: 846
--Bits: 8011620
--Readback:     Off
--Security:     On
--Bitstream CRC: 0x2F6C
--Usercode:   0x00000000
--Config Mode:None
--
// Header Frame
 FFFF  BFB3 
// Encrypted Block
 75030AD3F18E35F85A9A0410B05911FBF2334B939F030AD3F18E35F85A2E7EE12
// Encrypted Block
E22E3790A3F03CF0A3A83A71438E35F85A39F030AD3F5F85A2E7EE1E35F85A9A0
// Encrypted Block
...
```

Lattice Programming File Utility Style for non-encrypted file:
```
--Lattice Semiconductor Corporation Bitstream
--Version:         Diamond (64-bit) 3.9.1.119
--Bitstream Status: Final Version 10.23
--Design name: schaltschwelle_schaltschwelleLED5_6_7_8.ncd
--Architecture: sa5p00m
--Part: LFE5UM-45F-7CABGA381
--Date: Wed Oct 11 15:02:51 2017
--Rows: 9470
--Cols: 846
--Bits: 8011620
--Readback:     Off
--Security:     Off
--Bitstream CRC: 0x66BA
--
// Header Frame
 FFFF  BDB3FFFFFFFF 
// Reset CRC Frame
 3B  000000 
// Verify ID Frame
 E2  000000  01112043 
// Control Register 0 Frame
 22  000000  40000000 
// Reset Address Frame
 46  000000 
// Write Increment Frame
 82  9124FE 
```

Hex-Editor-Style for encrypted file:
[image]

Hex-Editor-Style for non-encrypted file:
[image]

trabucayre, 2022-03-23:
In fact I need a full file to play with it. Parser has to be verified / updated accordingly.
Thanks

el-coder-sb, 2022-03-23:
Just to mark this twice: The "header frame" part indicates if the bitstream is encrypted or not. It´s just about the **BFh** or **BDh**.

el-coder-sb, 2022-03-23:
Oh, didn´t see you already answered.

OK, I will generate a new encrypted file.

el-coder-sb, 2022-03-23:
Here you go:
[schaltschwelle_lattice_impl1_encry_key_HALLO.zip](https://github.com/trabucayre/openFPGALoader/files/8330813/schaltschwelle_lattice_impl1_encry_key_HALLO.zip)

trabucayre, 2022-03-23:
header value is a good information to know bitstream type: I have to integrate this as part of the info set (as the same level as idcode).
Thanks for the bitstream: I will be able to fix parser.

trabucayre, 2022-03-24:
The reason why the idcode doesn't match it's because bitstream configuration data is encypted. According to this [doc](https://www.latticesemi.com/-/media/LatticeSemi/Documents/ApplicationNotes/EH/FPGA-TN-02039-1-7-ECP5-and-ECP5-5G-sysCONFIG.pdf) I use `0xe2` to retrieve IDCODE, Since prjtrellis doesn't includes header it's the most common way to access to this information.
So now I need to learn howto deal with encrypted bitstream because:
- I suppose sending the bistream is not enough, I assume key must be programmed somewhere
- to check idcode it's required (if possible) to decrypt bitstream (the simpliest way remain to bypass idcode check when bitstream is encrypted)

el-coder-sb, 2022-03-24:
Thanks a lot for your quick reaction.

> According to this [doc](https://www.latticesemi.com/-/media/LatticeSemi/Documents/ApplicationNotes/EH/FPGA-TN-02039-1-7-ECP5-and-ECP5-5G-sysCONFIG.pdf) I use 0xe2 to retrieve IDCODE,

I don´t know the difference (in operation) between 0xE0 (p. 32) and 0xE2 (p. 50), i.e. READ_ID vs. VERIFY_ID?!?

> I assume key must be programmed somewhere

To get an encrypted bitstream run on a virgin FPGA: yes. However programming key (which is non-volatile!) and writing the bitstream are totally separated operations. So I think we can first focus on writing the bitstream. When this is solved, programming the key would be a very nice feature to add :-)

> to check idcode it's required (if possible) to decrypt bitstream

I´m not aware of any way to decrypt the bitstream. And even if this would be possible I´m not sure if IDCODE-opcode information is part of the (then decrypted) bitstream (as this would be a known data which is a good starting point to break encryption; I have no idea why the non-encrypted header of the encrypted bitstream file lacks the IDCODE information, I can´t see any benefit.). 

> the simpliest way remain to bypass idcode check when bitstream is encrypted

Just a suggestion: It needs some kind of "database". As the device name can be read from header of encrypted bitstream file, we can match it with a database entry. For building the database: The idcode for each device can be obtained e.g. from the corresponding bsd-file (for [ECP5](https://www.latticesemi.com/en/Products/FPGAandCPLD/ECP5) ).

[image]

I know this would mean quite a lot of data (as it´s the nature of a database). Nevertheless if implemented well once, it could be maintained easily.

Hope this helps!?

trabucayre, 2022-03-25:
`0xE0` is used to ask device to provides her IDCODE, `0xE2` is used to check matching between one idcode and device idcode. it's the first instruction in a bitstream to check ASAP if the bitstream is targeted for the device.

I have to read encryption docs to see how to deal with it, but if openFPGALoader has no access to idcode contained into the bitstream the only way is to bypass this check. Again bitstream header is only available with diamond, prjtrellis generate a bitstream without this (optional) section so it's a use case (but it's true it's possible to have both).

el-coder-sb, 2022-03-25:
Thanks for explanation. 

Sorry, I just like to clarify: As I understand in case of prjtrellis generated bitstream the 0xE2 command is used!? But where is the IDCODE retrieved from? (or do I misunderstand it and in this case IDCODE check is bypassed?)

trabucayre, 2022-03-26:
0xE2 is used by lattice and prjtrellis (ok it's optional). The idcode stored in the bitstream is the one for the targeted FPGA (when you create a project with diamond or with prjtrellis through nextpnr you specify FPGA model) and it is used to compare between real IDCODE (the one stored in FPGA where the bitstream is used) and bitstream's idcode (added when bitstream was created).

Just before loading into SRAM or writing to the SPI flash, openFPGALoader parse the bitstream to retrieve this information: He know by using 0xE0 the type of the FPGA connected to the computer but has to find if the bitstream is targeted for this FPGA (This avoid load failure).

When I talk about bypass, it's a temporary workaround to be able to load or store, when openFPGALoader hasn't access to IDCODE stored in the bitstream.

trabucayre, 2022-03-26:
After a quick read about bitstream's encryption it seems not, easily, possible to decrypt the bitstream.
So yes, in this situation one solution is to use header (when available) and to convert `Part: ` to idcode. `part.hpp` contains list of devices with `idcode` and `model` so it's possible to use it to have an alternative for an encrypted bitstreams.

trabucayre, 2022-03-27:
I have updated `latticeBitParser` to use *Part:* and *part.hpp* to get idcode from model.
So now openFPGALoader may no more fails with encrypted bitstream to check coherency between FPGA and bitstream when latter is encrypted.

trabucayre, 2022-03-29:
In fact since I have moved to libftdi 1.5 (where attach kernel is handled directly by the library) I've no more sees this error.
Fixed. Thanks.

closed 2022-03-29 by el-coder-sb