trabucayre/openFPGALoader #202: TE0711 S25FL256S QSPI Flash Error: Jedec ID : ff

zhuangzard, 2022-03-23:
Try to flash the TE0711 Model, which is Using Xilinx Arty xc7a100tcsg324 chip, with S25FL256S QSPI chip. Checked the constrain files, they are using same as the /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324.bit.gz files is using. But every time, when try to flash the QSPI, return errors:
openFPGALoader dose support S25FL256S chip, but the Jedec ID is FF is something not pulling chip information?
BTW, I am using Digilent HS2 programmer.

`openFPGALoader --fpga-part xc7a100tcsg324 -c digilent_hs2 Generator_V1.runs/impl_1/Generator.bin -f`
 

```
load program
Flash SRAM: [==================================================] 100.00%
Done
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : 
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
```

trabucayre, 2022-03-23:
It's weird. When 0xffff is seen this mean an issue with SPI flash communication (in fact it's bad to continue in this situation => I have to fix that).
Could you redo with `-v` to have more informations?
I have checked the schematic and the flash is connected with the same pinout.
Thanks

zhuangzard, 2022-03-23:
Thank you for the quick responds. Here is output

I am using master branch and on Ubuntu 20.04.4 LTS platform.

```
parallels@vm:~/Documents/Xilinx/Arty/GeneratorFPGA$ openFPGALoader --fpga-part xc7a100tcsg324 -c digilent_hs2 Generator_V1.runs/impl_1/Generator.bin -f -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bin
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
ff ff ff ff read ffffffff
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
RDSR : fc
WIP  : 0
WEL  : 0
BP   : f
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
parallels@vm:~/Documents/Xilinx/Arty/GeneratorFPGA$ 
```

zhuangzard, 2022-03-23:
After looking into the Trenz Electronic Wiki https://wiki.trenz-electronic.de/display/PD/TE0711+TRM#TE0711TRM-32MByteQuadSPIFlashMemory. It specified the:
_SPI Flash QE (Quad Enable) bit must be set to high or FPGA is unable to load its configuration from flash. By default this bit is set to high at the manufacturing plant._

I did some modification of the Xilinx_spiOverJtag.v file
and Force the csn to 1'b1 always. 

`assign  csn = 1'b1;//fsm_csn;`

Now, I could see some reading output, but Jedec ID become: 00

Here is output 
```
parallels@vm:~/Documents/Xilinx/Arty/GeneratorFPGA$ openFPGALoader --fpga-part xc7a100tcsg324x -c digilent_hs2 Generator_V1.runs/impl_1/Generator.bin -f -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bin
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324x.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
e6 1b 7a bf read e61b7abf
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
EDID + CFD length : 00
EDID              : 0000
CFD               : 00 00 00 00 00 00 00 00 00 80 00 00 00 0c 00 00 00 05 66 1b 7a bf fe 00 00 0d 66 1b 7a bf fe 00 00 06 b6 7e 08 e9 aa 00 00 bf 80 00 00 00 00 00 00 06 45 9e 08 e9 aa 00 00 0a 66 1b 7a bf fe 00 00 0c f5 30 08 e9 aa 00 00 a6 6c 8c 46 ec 86 46 66 06 45 9e 08 e9 aa 00 00 0e 66 1b 7a bf fe 00 00 05 25 9e 08 e9 aa 00 00 01 66 1b 7a bf fe 00 00 0c f5 30 08 e9 aa 00 00 05 66 1b 7a bf fe 00 00 05 25 9e 08 e9 aa 00 00 05 66 1b 7a bf fe 00 00 20 b5 30 08 e9 aa 00 00 05 25 9e 08 e9 aa 00 00 e3 66 1b 7a bf fe 00 00 0b 66 1b 7a bf fe 00 00 44 95 30 08 e9 aa 00 00 12 5a 54 08 
RDSR : 94
WIP  : 0
WEL  : 0
BP   : 5
TB   : 0
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash

```

zhuangzard, 2022-03-23:
Unfortunately, the reading Jedec ID is change every time if I use `assign csn = 1'b1;//fsm_csn;`
Don't think this method is reading correct Jedec ID.

```
parallels@vm:~/Documents/Xilinx/Arty/GeneratorFPGA$ openFPGALoader --fpga-part xc7a100tcsg324x -c digilent_hs2 Generator_V1.runs/impl_1/Generator.bin -f -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bin
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324x.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
4c 7c ac bf read 4c7cacbf
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
EDID + CFD length : 00
EDID              : 0000
CFD               : 00 00 00 00 00 00 00 00 00 80 00 00 00 0c 00 00 00 00 8c 7c ac bf fe 00 00 08 8c 7c ac bf fe 00 00 04 2e 40 7c 65 aa 00 00 bf 80 00 00 00 00 00 00 08 25 bf bc 65 aa 00 00 0d 0c 7c ac bf fe 00 00 71 00 2a bc 65 aa 00 00 2c c6 ec c6 86 c6 46 66 08 25 bf bc 65 aa 00 00 0b 0c 7c ac bf fe 00 00 0c a5 bf bc 65 aa 00 00 07 0c 7c ac bf fe 00 00 71 00 2a bc 65 aa 00 00 00 8c 7c ac bf fe 00 00 0c a5 bf bc 65 aa 00 00 00 8c 7c ac bf fe 00 00 46 7f ca bc 65 aa 00 00 0c a5 bf bc 65 aa 00 00 e4 8c 7c ac bf fe 00 00 0c 8c 7c ac bf fe 00 00 01 5f ca bc 65 aa 00 00 12 5d 8e bc 
RDSR : cc
WIP  : 0
WEL  : 0
BP   : b
TB   : 0
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
parallels@vm:~/Documents/Xilinx/Arty/GeneratorFPGA$ openFPGALoader --fpga-part xc7a100tcsg324x -c digilent_hs2 Generator_V1.runs/impl_1/Generator.bin -f -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bin
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324x.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
d5 5d 33 ff read d55d33ff
Detail: 
Jedec ID          : 95
memory type       : 5d
memory capacity   : 33
EDID + CFD length : 80
EDID              : 0000
CFD               : 00 04 28 cd 7c bc 6a 00 00 09 55 5d 33 ff fe 00 00 04 28 cd 7c bc 6a 00 00 0e 55 5d 33 ff fe 00 00 e3 05 4c 7c bc 6a 00 00 26 55 5d 33 ff fe 00 00 04 28 cd 7c bc 6a 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 0c 00 00 00 01 55 5d 33 ff fe 00 00 09 55 5d 33 ff fe 00 00 04 28 cd 7c bc 6a 00 00 bf 80 00 00 00 00 00 00 02 82 75 7c bc 6a 00 00 0c 55 5d 33 ff fe 00 00 71 04 b4 7c bc 6a 00 00 26 ac ac 26 cc cc 66 66 02 82 75 7c bc 6a 00 00 0a 55 5d 33 ff fe 00 00 02 82 75 7c bc 6a 00 00 06 55 5d 33 ff fe 00 00 71 04 b4 7c bc 6a 00 00 01 55 5d 33 ff fe 00 00 02 82 75 7c bc 6a 00 00 01 55 5d 33 ff fe 00 00 46 78 b4 7c bc 6a 00 00 02 82 75 7c bc 6a 00 00 e5 55 5d 33 ff fe 00 00 0d 55 5d 33 ff fe 00 00 01 58 b4 7c bc 6a 00 00 12 5b 52 7c 
RDSR : b4
WIP  : 0
WEL  : 0
BP   : 5
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
```

zhuangzard, 2022-03-23:
Thanks for the responds, please see attached output after put `-v`

```
~/Documents/Xilinx/Arty/GeneratorFPGA$ openFPGALoader --fpga-part xc7a100tcsg324 -c digilent_hs2 Generator_V1.runs/impl_1/Generator.bin -f -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bin
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
ff ff ff ff read ffffffff
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
RDSR : fc
WIP  : 0
WEL  : 0
BP   : f
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
```

trabucayre, 2022-03-23:
`csn` has the same role as address in `i2c` protocol: it is used to address one peripheral. If `csn` is high peripheral doesn't take into account data in `si` and `sck`. This why you have always '0'.
`Quad enable` is bit 1 from `CR1` register to configure flash into quad mode: required when you build a bitstream with x4 data bus widths.
It's maybe because the device is in quad mode openFPGALoader is unable to communicate with him => I have to check that.

trabucayre, 2022-03-23:
Could you please provides the full code of your device, others option is the bitstream for `xc7a100ticsg324-1L` is not compatible with `XC7A100T-2CSG324C` or `XC7A100T-2CSG324I`

zhuangzard, 2022-03-23:
I removed some of comments above, because I build my own `spiOverJtag` bit file using wrong FPGA family. My comment above is the from correct device family output.

The device is "xc7a100tcsg324-2". I could program Using Xilinx directly.

Regarding quad mode, the openFPGALoader get this information from `_jedec_id`, how could I know it communication have never been build? looks _jedec_id is FFFFFF, which means, chips never been selected.

trabucayre, 2022-03-23:
jedec_id is read from register 0x90. Value read may be because csn never goes low, flash powered down or something wrong into FPGA.
I have to read again the datasheet, to see for something like this.

trabucayre, 2022-03-23:
And If I'm unable to find why, I think I have to order one board....

trabucayre, 2022-03-23:
Ok so the bitstream is worling: good thing.
For team-viewer thanks but my home internet connection it's not possible and it's more easy to have access to the hardware.
first I have to check datasheet and to try with board with the same device to see why jedec read is not working.

zhuangzard, 2022-03-23:
that sounds good. BTW, if you are located at United States, I could send one board to you to test. But looks you are not.

trabucayre, 2022-03-23:
Thanks but I live in France...
Anyway first I have to try with hardware I have :)

zhuangzard, 2022-03-24:
Quick update, just tried to program the TE0711(same SOM) with TE0703 carrier board https://wiki.trenz-electronic.de/display/PD/TE0703+TRM , instead of utilizing Digilent's programer, I could successfully program the Flash without any issue.

Our system is using Jtag : Digilent_hs2 https://www.digikey.com/en/products/detail/digilent-inc/410-249/3902812, TE0703 on board Jtag is Digilent, don't know that will lead the different or not. 

```
~/Documents/Xilinx/Arty/GeneratorFPGA$ openFPGALoader --fpga-part xc7a100tcsg324 -c digilent Generator_V1.runs/impl_1/Generator.bin -f -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bin
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a100tcsg324.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
1 2 19 4d read 102194d
Detected: spansion S25FL256S 512 sectors size: 256Mb
RDSR : 00
WIP  : 0
WEL  : 0
BP   : ca
TB   : 224
SRWD : 0
00000000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
parallels@vm:~/Documents/Xilinx/Arty/GeneratorFPGA$ 
```

trabucayre, 2022-03-24:
It's partially a good news: this mean spi flash is working, and openFPGALoader has no difficulties to communicate! (Could you retry with official `spiOverJtag` for this device to check if it compatible?).
But it's weird: I checked `TE0711` schematic and I have not seen anything explaining why with this carrier it's works and with your setup no... Have you tried to load bitstream in SRAM? `digilent_hs2` is fully supported (it's one of probe I use the most) it's based on `FT232H`, `digilent` on `FT2232H`, and since your able in your setup to get FPGA idcode it's mean `hs2` is working...

zhuangzard, 2022-03-24:
After a long day try, I think I found the reason why system persist me flash the QSPI flash. Last night try was success because I used a new FPGA SOM and flash the new code (not the code I used at office). This morning, I tried again to flash the same hardware use office's code, which give me same FFFFFF output again.  So, not as I thought it was the Digilent programer issue.

I dig into the constrain code, find this line in my office project

`set_property BITSTREAM.CONFIG.PERSIST YES [current_design]`

This prevent me programing using openFPGALoader, but Xilinx Programer could works fine.

This is under the Vivado>Synthesis Tab>Edit Device Properties>Configuration>Prohibit usage of configuration pins as user I/O and persist after configuration should be **NO**.

After changed to No or delete the constrain code line, I tested the code utilizing your latest code directly without any issue.

Again, I appreciate for you help in last two days and try out back and forth with my misleading information. I hope other people who comes into same situation could saw this thread and find the solution.

Thank you! 
Best Regards

trabucayre, 2022-03-25:
Thanks for your feedback: this information must be added to Troubleshooting page and/or xilinx page.
I have to check howto fix that -> if xilinx programmer is able to deal with this situation I have to add the correct configuration to do the same with openFPGALoader.
Thanks again

zhuangzard, 2022-03-25:
Glad could contribute little bit. It is very interesting to know how Xilinx JTAG could flash QSPI event if it pin maintains the configuration logic access `BITSTREAM.CONFIG. PERSIST YES`.

Last night tried to recompile the `spiOverJtag` by adding constrain file `BITSTREAM.CONFIG. PERSIST NO` see it this RAM bit file could unlock the `QSPI persist configuration pin`, but without success. 

UG908 did not give more detail about how Xilinx JTAG server tell FPGA to flash and pull down the `csn` pin after pin been persisted.

zhuangzard, 2022-03-25:
Just an idea about how could bypass it. Is the STARTUPE3 primitive could help to prevent those i/o been persisted? https://www.xilinx.com/support/documentation/user_guides/ug570-ultrascale-configuration.pdf page 115 is showing the STARTUPE3 is have `DI/DO/DTS` and `FCSBO`, which will not like `spiOverJtag` control the `SPI` over I/O pin.

trabucayre, 2022-03-26:
`STARTUPE3` primitive is for `ultrascale` FPGA and not for `serie7`.
I have to find document about persist to see how to reset pins configuration.