trabucayre/openFPGALoader #206: TangNano4k could not communicate since last bitstream flashed.

Martoni, 2022-03-26:
Since I flashed my last bitsream I can't communicate with my tangnano4k anymore.

The bitstream I downloaded was generated with Gowin EDA version 1.9.7.06. And Verilog projet used is provided by my GbHdmi project [here](https://github.com/Martoni/GbHdmi).

Verilog file is generated with following SBT command : 
```shell
$ sbt
sbt:GbHdmi> runMain gbhdmi.TopGbHdmiDriver
```
Warning, the Verilog generated named `TopGbHdmi.v` is dysfunctional.

I flashed bitstream once on my tangnano4k, and since can't communicate with it using openFPGALoader:
```shell
$ openFPGALoader -b tangnano4k --detect
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
$
``` 
Even if I try to flash it with other bitstream I can't : 
```shell
 openFPGALoader -b tangnano4k led_test/project/impl/pnr/led_test.fs
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Error: no device found
```

And If I try to configure tangnano4k with gowin programmer I get this error:
``` 
Info:	"SRAM Program" starting on device-1...
Warn:	Unknown Cable
Error:	ID code mismatch.
Info:	Quit.
Info:	Cost 0.95 second(s)
```

Do you think that my board is «briked» ? The «video part» of my design works if I plug HDMI cable.

Has anyone ever had a similar problem ?

Thanks

kiwanowski, 2022-03-26:
[image]

Did you try pulling down pin 08?

trabucayre, 2022-03-27:
@kiwanowski it's my first idea too

trabucayre, 2022-03-27:
@Martoni Could you fetch new commits and try with:
```
openFPGALoader --detect --verbose-level=2
```
To display raw idcodes
Thanks

trabucayre, 2022-03-27:
I have forgotten to ask that: have you seen 'write error' messages last time you've programmed your board?
I think @pepijndevos has already seens issue like this, where he could no longer reprogramming FPGA. Maybe something with JTAGSEL indeed

pepijndevos, 2022-03-27:
Did the bitstream have any dual mode pins in use? Yea it's possible to soft brick the FPGA that way. You may have to manually hold down some pins to get it in a programmable mode.

Martoni, 2022-03-27:
@trabucayre, with verbose level 2 :

```
$ openFPGALoader -b tangnano4k --detect --verbose-level=2
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0xffffffff
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 0 devices
```

It's the same with pin 08 pulled down.

Martoni, 2022-03-27:
`Did the bitstream have any dual mode pins in use? Yea it's possible to soft brick the FPGA that way. You may have to manually hold down some pins to get it in a programmable mode.`

Hmm, in fact yes. On tangnano4k the led_o pin is used for `MODE` too. 

[image]

Plugged on mode 1 and 2:

[image]

@kiwanowski  I'll try with digilent [HS2](https://digilent.com/shop/jtag-hs2-programming-cable/) jtag programmer and JTAGSEL_N pulled down.

Martoni, 2022-03-27:
Thanks all for your so quick responses. And sorry for this stupid question...

@kiwanowski & @trabucayre, yes the solution was simply to pull down JTAGSEL_N .

```
$ openFPGALoader -b tangnano4k --detect --verbose-level=2
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0x0100981b
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 1 devices
index 0:
	idcode 0x100981b
	manufacturer Gowin
	family GW1NSR
	model  GW1NSR-4C
	irlength 8
```

I can re-configure the fpga without problem now.

closed 2022-03-27 by Martoni

trabucayre, 2022-03-27:
Great! Thanks to all!
It's something to note in `gowin.rst` and `troubleshooting.rst` with big, red, bold, underline font ;-)

Martoni, 2022-06-11:
Just for remember : **JTAGSEL_N** is **MSPI_WP** - PIN08_IOT4B - **15** on tangnano4k board.

trabucayre, 2022-07-26:
I have added a note in the doc.
Thanks for this reminder!