trabucayre/openFPGALoader #214: [Request] Add support for CycloneV 5CEBA9 flash programming

el-coder-sb, 2022-04-11:
Trying to write to the flash of a Intel CycloneV 5CEBA9, I do:

```
openFPGALoader -c ft2232 --fpga-part 5ce523 -f blink.rpd 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
use: /usr/local/share/openFPGALoader/spiOverJtag_5ce523.rbf.gz
Flash SRAM: [==================================================] 100.00%
Done
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : 
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
Error: Failed to program FPGA: Fail to write data
detach error -5
```

It does not run with any of the given spiOverJtag_5cX23.rbf. I suppose I have to generate a new spiOverJtag_XX.rbf file for this FPGA? 

Is spiOverJtag the same as SFL (serial flash loader) of Intel? Do you know how the existing rbf files were generated?

trabucayre, 2022-04-11:
A new bitstream must be built for `5CEBA9`, but I need to have full version (with package).
I have to check (if I have already do it I don't remember) if package is really relevant since SPI pins are accessed through `altserial_flash_loader`
Thanks!

trabucayre, 2022-05-04:
Hi,
Friendly reminder for the full model of your FPGA -> required to add the corresponding spiOverJtag bitstream.
Thanks

trabucayre, 2022-05-05:
I have [pushed](https://github.com/trabucayre/openFPGALoader/commit/3a3fe9a3cbeaa7f46039728a7d562cc6f3c42ec3) corresponding rbf file.
Could you try to check if it's working?
Thanks

el-coder-sb, 2022-05-05:
```
openFPGALoader -c ft2232 --fpga-part 5ce927 -f blink.rpd
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
use: /usr/local/share/openFPGALoader/spiOverJtag_5ce927.rbf.gz
Flash SRAM: [==================================================] 100.00%
Done
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : 
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
Error: Failed to program FPGA: Fail to write data
```

gives me the same output and result as given above...

Anything I can check?

el-coder-sb, 2022-05-05:
I noticed that I get the same result (and runtime) even when I remove `/usr/local/share/openFPGALoader/spiOverJtag_5ce927.rbf.gz`. So I cannot tell if the rbf-file is correctly written into FPGA.

trabucayre, 2022-05-06:
Is it a commercially available board or something custom?
Messages displayed by openFPGALoader means the flash is not accessible or a communication problem appears...
You have exactly same display when the bitstream is removed? It's weird I have to check that. Theoretically when you try to load a not available bitstream, openFPGALoader must stop immediately...

el-coder-sb, 2022-05-06:
It´s a custom board with MSEL[4..0] = 10011 for AS (active serial) Programming (according to "Cyclone V Device Handbook"). With the Quartus Programmer Software writing to flash after loading the serial flash loader  works fine.

> Messages displayed by openFPGALoader means the flash is not accessible

With the oscilloscope I can´t see any real SPI communication between Flash and FPGA.

> You have exactly same display when the bitstream is removed? 

Yepp.

> It's weird I have to check that.

I was suprised, too. Let´s see if you can reproduce this.

trabucayre, 2022-05-07:
It's AS x4 no ? Could you try with 10010. It's maybe an issue with flash configured as single or quad...
But if SPI lines are always idle it's maybe another problem ...

trabucayre, 2022-05-08:
In fact there is another issue (#128) for cyclone V and SPI flash. Unfortunately in both cases boards are not commercially boards so I can't reproduce at home (and with my cycloneV board flash is working...)

el-coder-sb, 2022-05-09:
> You have exactly same display when the bitstream is removed?

I double-checked that today. The behaviour is like this (files always in  /usr/local/share/openFPGALoader/ ):
If there is no spiOverJtag_5ce927.rbf.gz but spiOverJtag_5ce927.rbf (without .gz) the SRAM is written.
If there is no spiOverJtag_5ce927.rbf* at all, the program fails as expected with "unable to open file".
So what I have reported above was when I was trying around and had a spiOverJtag_5ce927.rbf (without .gz). 
As it says "use: /usr/local/share/openFPGALoader/spiOverJtag_5ce927.rbf.gz" I was confused because it does not print any error/warning. However it seems like the program runs fine as well...

> But if SPI lines are always idle it's maybe another problem ...

I checked that again. And I could see that the Flash sends its ID-CODE correctly (measuring only CLK and MISO):
[image]

However openFPGALoader shows no valid ID.

I could do some debugging. For this it would be fine if I could do the following from comand line (if it´s already possible I´m sorry, I didn´t find it):
a) run the "program flash" part without writing the SRAM first (i.e. skipping the SRAM write step when '-f' is selected)
b) run only "read_id" without writing to flash

Do you think you could implement this soon? Would be great. If not I could try to add that.

el-coder-sb, 2022-05-09:
> It's AS x4 no ? Could you try with 10010. 

According to the table 7-2 in "Cyclone V Device Handbook" (Version 2019.10.03) the difference between 1001**0** and 1001**1** is only the POR-Delay (fast vs. standard) and does determine the AS mode.

trabucayre, 2022-05-10:
Yes! Historically spiOverJtag's bitstream was versioned without compress. Since zlib support following @emard's request, all new bitstreams are compressed, but to avoid noise, already present ones are left unchanged. openFPGALoader is trying first to read compressed one, and if not available switchs to uncompressed version. Maybe it's a bad idea to do that silently...
There is no options to run only 'read_id'. Usually I comment piece of code to do that, but it's maybe, for debug purpose, required to add something like that, it's true.

For your chrono, are you sure it's openFPGALoader's request? Before stopping, openFPGALoader, when write flash, sent reset, to force device to read new configuration from flash... It's maybe transaction from the FPGA.

el-coder-sb, 2022-05-10:
> It's maybe transaction from the FPGA.

Yes, it might be that as well. That´s why I asked for only "read_id" command to simplify debugging. (now I have to wait until SRAM is written and then hope I really get the first SPI transaction triggert and if not I have to go for another round...)

el-coder-sb, 2022-05-11:
> a) run the "program flash" part without writing the SRAM first (i.e. skipping the SRAM write step when '-f' is selected)
I startet #222 to add this.

Using that branch I run with the --skip-flash flag to only write to the SRAM. As I understand the code there shouldn´t be any reset after a successful write to SRAM (and an afterwards succesful program ending). However the CONF_DONE is not set active. Doing so with the Quartus Programmer (and another file than the rbf file used by openFPGALoader) CONF_DONE is set active.

Maybe I can do some more tests tomorrow...

trabucayre, 2022-05-12:
It's not clear :)
SRAM is non-volatile storage: if you reset after loading a bitstream the FPGAs read configuration from flash. So device is only reset after writing to the flash (in more or less all class for FPGAs with external flash this step is done in post_flash_access).

trabucayre, 2022-05-12:
I have to check for CONF_DONE

el-coder-sb, 2022-05-12:
Using #222 (state of yesterday) and '--skip-flash' flag, I can observe the following:  
* void Altera::reset() is never called (as expected), a breakpoint inside this function is never triggered (I wasn´t aware of post_flash_access(), that´s why I checked this with breakpoint again)
* using the below attached rbf-file (which I generated) I get at least a CONF_DONE active signal. However without '--skip-flash' flag the flash ID isn´t read neither

Could you report the steps you do to generate the rbf file?

Here are mine:
1) create a new project in Quartus
2) select my CycloneV device
3) add the SerialFlashLoader-IP 
4) compile the project
5) convert sof-file to rbf-file

[spiOverJtag_5ce927.rbf.gz](https://github.com/trabucayre/openFPGALoader/files/8676416/spiOverJtag_5ce927.rbf.gz)

trabucayre, 2022-05-31:
Sorry for delay...
Correct (theoritically) spiOverJtag bitstream for your device is already present in repository so you have to install it.
 The procedure to add a new device may be summaryzed by [this commit](https://github.com/trabucayre/openFPGALoader/commit/3a3fe9a3cbeaa7f46039728a7d562cc6f3c42ec3).

el-coder-sb, 2022-05-31:
> The procedure to add a new device may be summaryzed by [this commit](https://github.com/trabucayre/openFPGALoader/commit/3a3fe9a3cbeaa7f46039728a7d562cc6f3c42ec3).

Sorry, there might be some misunderstanding: I´m already using spiOverJtag bitstream for the 5CEBA9.
Nevertheless the CONF_DONE is not active after loading that spiOverJtag bitstream.

That´s why I asked how you did to generate the `spiOverJtag_5ce927.rbf.gz`?

el-coder-sb, 2022-06-01:
Thanks for the informations.

> For CONF_DONE pin could you add set_global_assignment -name ENABLE_INIT_DONE_OUTPUT ON in spiOverJtag/constr_cycloneV.tcl?

Works fine. If there is no arguments against it, would be nice to add this line to the project.

I keep trying to find the source of failure...

trabucayre, 2022-06-02:
Yep! I have to add this option to the altera tcl. But to avoid noise and too big commit this will be applied to future bitstreams (or to all if something must be fixed).

el-coder-sb, 2022-06-10:
Installing Quartus 20.1.1 Lite and running `make spiOverJtag_5ce927.rbf.gz` (+ build + make install) gives SPI-Signals and the JEDEC-ID is read 😄 

Here the file which is generate by running make:

[spiOverJtag_5ce927.rbf.gz](https://github.com/trabucayre/openFPGALoader/files/8876745/spiOverJtag_5ce927.rbf.gz)
 
So may be you just used an outdated version of Quartus which has a bug??

Furthermore there ocurrs another error (and I do not understand why there are four bytes read for the ID...). Here the output:

```log
openFPGALoader -c ft2232 --fpga-part 5ce927 -f /light.jic -v
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x2b140dd
	manufacturer altera
	family cyclone V
	model  5CEBA9
	irlength 10
File type : jic
use: /usr/local/share/openFPGALoader/spiOverJtag_5ce927.rbf.gz
Flash SRAM: [==================================================] 100.00%
Done
20 ba 18 10 read 20ba1810
Detected: micron N25Q128 256 sectors size: 128Mb
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash overflow
Error: Failed to program FPGA: Fail to write data
```

trabucayre, 2022-06-10:
WTF! I have the same version but it's prime instead of lite!
There is something strange with that! But the most important thing is: the spiOverJtag bitstream is working for your model!
Maybe it's the same thing for #128 ...

You have no others modifications for spiOverJtag? (I have updated constr_cycloneV to enable init done pin).
Could you create a PR to update the file?

Currently `.jic` are not supported only `rpd` or `rbf` may be used.

Thanks!

el-coder-sb, 2022-06-14:
> I have the same version but it's prime instead of lite!

I think it´s not prime vs. lite but lite vs. standard vs. pro ;-)

> You have no others modifications for spiOverJtag? 

No, when I compiled the stuff I had this diff:

```log
git diff master test-read-flash-ID 
diff --git a/spiOverJtag/constr_cycloneV.tcl b/spiOverJtag/constr_cycloneV.tcl
index afcd80e..5ce45cf 100644
--- a/spiOverJtag/constr_cycloneV.tcl
+++ b/spiOverJtag/constr_cycloneV.tcl
@@ -1,2 +1,3 @@
 set_global_assignment -name STRATIXV_CONFIGURATION_SCHEME "ACTIVE SERIAL X4"
 set_global_assignment -name ON_CHIP_BITSTREAM_DECOMPRESSION ON
+set_global_assignment -name ENABLE_INIT_DONE_OUTPUT ON

I´ll prepare a PR and do tests with rpd next days.
```

trabucayre, 2022-06-14:
It's hard to believe the same soft with more or less limitations produces working/not working bitstream for the same code :-)
But experience shows it's possible...

Since `ENABLE_INIT_DONE_OUTPUT` is now by default there is nothing more to do than pushing new bitstream revision.

rbf is another fine option too for both RAM and flash (for RAM compression must be disabled).

el-coder-sb, 2022-06-14:
Just in case you want to compare the rpt files for any hints for diffs...
[spiOverJtag.fit.rpt.txt](https://github.com/trabucayre/openFPGALoader/files/8897244/spiOverJtag.fit.rpt.txt)
[spiOverJtag.flow.rpt.txt](https://github.com/trabucayre/openFPGALoader/files/8897245/spiOverJtag.flow.rpt.txt)
[spiOverJtag.map.rpt.txt](https://github.com/trabucayre/openFPGALoader/files/8897246/spiOverJtag.map.rpt.txt)

trabucayre, 2022-06-14:
Thanks. I will try to see if something differs between both quartus flavors....

el-coder-sb, 2022-08-03:
Finally I got time to final-test it with rpd-file. Works fine. The bitstream is correctly loaded by the fpga afterwards.

```log
$ sudo openFPGALoader -c ft2232 --fpga-part 5ce927 -f light.rpd
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
use: /usr/local/share/openFPGALoader/spiOverJtag_5ce927.rbf.gz
Flash SRAM: [==================================================] 100.00%
Done
Detected: micron N25Q128 256 sectors size: 128Mb
00000000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
```

Thanks!

closed 2022-08-03 by el-coder-sb