trabucayre/openFPGALoader #215: Arrow DECA support

gergoerdi, 2022-04-15:
I'm trying to use `openFPGALoader` with an [Arrow DECA](https://www.arrow.com/en/products/deca/arrow-development-tools) development board. Its product page claims it has a USB-Blaster II included, so I thought it'd be as simple as using the `-c usb-blasterII` flag; alas, it fails on `--detect`:

```
$ openFPGALoader -c usb-blasterII  --probe-firmware ~/sdk/fpga/Altera/20.1/quartus/linux64/blaster_6810.hex  --detect
write to ram
USB-Blaster II firmware version: 1.39
JTAG init failed with: Unknown device with IDCODE: 0x031050dd
```

gergoerdi, 2022-04-15:
Out of curiosity, I tried extending `part.hpp`:

```
diff --git a/src/part.hpp b/src/part.hpp
index 1c02fdf..af3198e 100644
--- a/src/part.hpp
+++ b/src/part.hpp
@@ -85,6 +85,7 @@ static std::map <int, fpga_model> fpga_list = {
        {0x02d020dd, {"altera", "cyclone V Soc", "5CSEBA6", 10}},
        {0x02d010dd, {"altera", "cyclone V Soc", "5CSEMA4", 10}},
        {0x02d120dd, {"altera", "cyclone V Soc", "5CSEMA5", 10}},
+        {0x031050dd, {"altera", "MAX 10", "10M50DAF256", 10}},
 
        {0x00000001, {"efinix", "Trion",    "T4/T8",            4}},
        {0x00210a79, {"efinix", "Trion",    "T8QFP144/T13/T20", 4}},
```

But it's not enough to get it working:

```
USB-Blaster II firmware version: 1.39
found 1 devices
index 0:
	idcode 0x31050dd
	manufacturer altera
	family MAX 10
	model  10M50DAF256
	irlength 10
File type : svf
FREQUENCY 2.40E+07 HZ 
frequence valeur 2.40E+07 unite HZ24000000
USB-BlasterII has a 24MHz fixed frequency
TRST ABSENT 
trst value : ABSENT
ENDDR IDLE 
enddr value : IDLE
ENDIR IRPAUSE 
endir value : IRPAUSE
STATE IDLE 
state value : IDLE

SIR 10 TDI (2CC) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST IDLE 8400003 TCK ENDSTATE IDLE 
```

and then just hangs there.

trabucayre, 2022-04-16:
Your modification looks good.
I have a trenz board with a max10 (but never already used). I will try in my side to see why SVF hang.
Thanks

trabucayre, 2022-04-16:
Tried with my @Trenz-Electronic _analogMax_: it's works. So I have to check SVF<->usbBlasterII

gergoerdi, 2022-04-17:
Could it be that I created my `.svf` file incorrectly?

```
$ quartus_cpf  -c -n p -g 3.3  -q 24MHz _build/arrow-deca/synth/Top.sof Top.svf
```

trabucayre, 2022-04-19:
I have tried with a _de10nano_ (usb blaster II) and the same command to generate svf-> it's works
So:
- max10 + svf + openFPGALoader: ok
- usbBlasterII + svf + openFPGALoader: ok

I have to find a deca board because the issue is not clearly evidente.
But when you said "and then just hangs there" you wait for a long time before killing app? SVF is a bit slow.

gergoerdi, 2022-04-19:
I didn't know SVF upload was supposed to be slow! I killed it after 1 or 2 seconds because that's how long it took to upload the original `.sof` file from the Quartus GUI.

But how slow is too slow? I tried it again now and it's been running for two minutes, the last output being 

```
RUNTEST IDLE 8400003 TCK ENDSTATE IDLE 
```

trabucayre, 2022-04-19:
Since there is no loop aspect in SVF, it's usual to use **runtest** to wait instead of polling bit(s) on a specific register. Some vendors adds garbage to obfuscate code to avoid simple reverse. This why this approach is usually slower than a code with device knowledge (and it's why quartus_pgm is faster).
Since max10 is not officially supported (I have to study bitstream file) you have to use SVF, but my real problem is why it's not working with this board...

gergoerdi, 2022-04-20:
Let me know if I can help -- e.g. by attaching the `.sof` or `.svf` file, running a special version of `openFPGALoader` with the board with loads of extra debug trace messages, etc.

trabucayre, 2022-04-21:
Yes you can attach your files to allows me to analyze them. But without the board I can't see if something is wrong specifically for this hardware. But deca seems not available on classic store...
Anyway I have to try to add support for max10 and, consequently, add a way to use use file other than SVF

trabucayre, 2022-04-21:
Thanks.
I will check content and svf player behaviour to see if something go wrong with this file.

trabucayre, 2022-05-06:
I have analyzed a bit the svf file and openFPGALoader's svf player seems correct... I have some additionnal check to do and maybe small improvement but I can't see trivial issues...
I have searched for one deca board  but this one seems currently impossible to bought / not available...

boogermann, 2022-12-28:
@trabucayre I have this board and was trying to add support to it and got stuck at the same spot as @gergoerdi at `RUNTEST IDLE 8400003 TCK ENDSTATE IDLE`. try left it running for about 30min and nothing.
We use Quartus to generate the svf output and notice that by default the frequency is set to 25mhz, either way sof->svf or the generated file has the same issue of locking up.
Files are located [here](https://github.com/DECAfpga/DECA_binaries)

The board has a QSPI flash where there's a known bug that will clear what's there when flashing an svf file as described here: 
[Why is the security key erased in the Intel® MAX® 10 device during Serial Vector Format (.svf) programming?](https://www.intel.co.uk/content/www/uk/en/support/programmable/articles/000086778.html)
Even after removing that from the svf file I had no success.

openFPGALoader seems to be interacting with the board correctly, lights goes up for programming and can be reset.
You think you could point me to where to look so I can try to fix it?
OpenOCD can flash both svf files, would a verbose dump help identify what could it be?

trabucayre, 2023-01-01:
I have to retry with your bitstream, but with a sort of fake interface because I haven't the board (but it's enough if issue is related to an overflow or something similar). Last time I haven't seen any issues (but maybe it's because it's not a real interface/board).
I have started to add native support for this FPGA device family: pof parser is now mainlined, I have to add algo to load the content to max10 device.

I have, also to check if stall is not cable related (I have boards with usbBlasterII interface so It's possible to check behaviour with a large runtest wait).

trabucayre, 2023-01-02:
@gergoerdi @boogermann : I'm a dumb. The issue was not at the svf  parser level but in usb blasterII driver. A 16bits variable was used to store number of Bytes to send but an overflow appears  when log2(clk_len) > 16.
It's now fixed and this method ends correctly.

boogermann, 2023-01-02:
@trabucayre Awesome, Thanks for the fix!
I just gave it a test, it's not getting stuck anymore, but a new error has showed up. Tried svf output from quartus @ 25Mhz and converted one to @ 24Mhz, with and without the DSM Clear, but they all seem to error out on the same spot.

```
openFPGALoader -b deca galaga_deca_220814.svf --probe-firmware /opt/altera/quartus/linux64/blaster_6810.hex -r -v
USB-Blaster II firmware version: 1.36
found 1 devices
index 0:
	idcode 0x31050dd
	manufacturer altera
	family max 10
	model  10M50DA
	irlength 10
FREQUENCY 2.40E+07 HZ 
frequency value 2.40E+07 unit HZ24000000
USB-BlasterII has a 24MHz fixed frequency
TRST ABSENT 
trst value : ABSENT
ENDDR IDLE 
enddr value : IDLE
ENDIR IRPAUSE 
endir value : IRPAUSE
STATE IDLE 
state value : IDLE

SIR 10 TDI (2CC) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST IDLE 8400003 TCK ENDSTATE IDLE 

SIR 10 TDI (203) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 123 TCK 

SDR
	len   : 23
	tdo   : 0
	tdi   : 24
	mask  : 0
	smask : 0

SIR 10 TDI (205) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 123 TCK 

SDR
	len   : 32
	tdo   : 32
	tdi   : 32
	mask  : 32
	smask : 0

SIR 10 TDI (203) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 123 TCK 

SDR
	len   : 23
	tdo   : 0
	tdi   : 24
	mask  : 0
	smask : 0

SIR 10 TDI (3F2) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 8400003 TCK 

SIR 10 TDI (307) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 123 TCK 

TDO value 80 isn't the one expected: 1
Cannot proceed because of error(s) at line 39
```

trabucayre, 2023-01-02:
This check is a recent addition, but it looks like bit reversion is wrong `0x01` seems `0x80` in opposite order...
I have to check with my analogmax (and maybe with another boards).

trabucayre, 2023-01-04:
Ok. I have found (and fixed) this issue. Again at usbblaster driver level: wrong align when the read transaction is only 1bit length (same issue was found for ftdi driver). It's weird I suppose until now all read was by a multiple of 8bits.

boogermann, 2023-01-04:
@trabucayre awesome, I'm gonna give it a test and submit another PR with the board.
Thanks for the fix!

trabucayre, 2023-01-04:
And thanks for the PR :)

@gergoerdi Could you test too? Thanks

boogermann, 2023-01-04:
@trabucayre submitted!
Should we put a note on the documentation saying that the MAX10 doesn't support RBF files?

trabucayre, 2023-01-04:
Yes it make sense [intel doc](https://github.com/trabucayre/openFPGALoader/blob/master/doc/vendors/intel.rst) must be updated to add max10's details.

boogermann, 2023-01-04:
Ok. I'll update the docs and probably submit another PR for the de10-lite, another Max 10 but uses usb blaster 1.