trabucayre/openFPGALoader #216: Flash programming not working with the Colorlight i9

hansfbaier, 2022-04-23:
Hi, flashing does not work with the colorlight i9.
I use a FT232H. Flashing works on the i5 there:
```
$ openFPGALoader -c ft232 build/top.bit -f --verify -r 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detail: 
Jedec ID          : c8
memory type       : 40
memory capacity   : 15
EDID + CFD length : c8
EDID              : 1540
CFD               : 
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
Refresh: DONE
```
When I try on the Colorlight I9, I get:
```
$ openFPGALoader --verbose-level 2 -c ft232 build/top.bit -f --verify -r 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0x41112043
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 1 devices
index 0:
	idcode 0x1112043
	manufacturer lattice
	family ECP5
	model  LFE5UM-45
	irlength 8
File type : bit
IDCode : 41112043
displayReadReg
	Config Target Selection : 0
	Done Flag
	Std PreAmble
	No err
Open file DONE
Parse file DONE
bitstream header infos
Part: LFE5U-45F-8CABGA381
idcode: 41112043
Enable configuration: DONE
SRAM erase: DONE
ef 40 17 0 read ef401700
Detected: Winbond W25Q64 128 sectors size: 64Mb
RDSR : 1c
WIP  : 0
WEL  : 0
BP   : 1a
TB   : 220
SRWD : 0
nr_sectors : 64
00000000 00400000 007fffff 1c
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Fail
Verification failed at 1
Refresh: DONE
Error: Failed to program FPGA: std::exception

```
And the bitstream in Flash has not changed, when I do a reset.
Loading the bitstream into the SRAM of the FPGA, works fine, though.

trabucayre, 2022-04-23:
It's weird. Your flash seems have block's protection enabled but openFPGALoader seems don't care about that.
I have to check/fix that.
Thanks

trabucayre, 2022-04-23:
Ok.
status register is wrong -> it's fixed.
This correction didn't fix your issue but at least informations are true: could you dump again?

hansfbaier, 2022-04-28:
@trabucayre Now the programming succeeds, but the flashed bitstream does not work ...

hansfbaier, 2022-04-29:
Unfortunately the same result:
```
$ openFPGALoader --verbose-level 2 --index-chain 0  -f --unprotect-flash -c ft232 build/top.bit  --verify
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0x41112043
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 1 devices
index 0:
        idcode 0x1112043
        manufacturer lattice
        family ECP5
        model  LFE5UM-45
        irlength 8
File type : bit
IDCode : 41112043
displayReadReg
        Config Target Selection : 0
        Done Flag
        Std PreAmble
        No err
Open file DONE
Parse file DONE
bitstream header infos
Part: LFE5U-45F-8CABGA381
idcode: 41112043
Enable configuration: DONE
SRAM erase: DONE
ef 40 17 0 read ef401700
Detected: Winbond W25Q64 128 sectors size: 64Mb
RDSR : 1c
WIP  : 0
WEL  : 0
BP   : 7
TB   : 0
SRWD : 0
nr_sectors : 64
00000000 00400000 007fffff 1c
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Fail
Verification failed at 1
Refresh: DONE
Error: Failed to program FPGA: std::exception
```

trabucayre, 2022-04-30:
Ok. Seems to have a problem with this chip or with unprotect step... Maybe I have to order one i9, or to try another board with similar flash chip.

trabucayre, 2022-04-30:
Could you try:
`openFPGALoader --verbose-level 2 --index-chain 0  --unprotect-flash -c ft232 `
followed by
`openFPGALoader --verbose-level 2 --index-chain 0  -f -c ft232 build/top.bit  --verify`

hansfbaier, 2022-05-01:
That worked! Many thanks!

closed 2022-05-01 by hansfbaier

trabucayre, 2022-05-02:
Thanks for your feedback.
A complete check of this chip is mandatory: seems to have something wrong with the protection. Your flash was protected but this is not correctly discover by openFPGALoader. It's maybe a bug around `TB` or something wrong internally to this chip (another chip does not not taken into account TB information).