trabucayre/openFPGALoader #217: FT232 USB tests fail if no iProduct field set.

DaveBerkeley, 2022-04-26:
I'm trying to use a standard FT232H breakout board as a JTAG programmer.

It uses the same FTDI device as my jtag-hs3. However, the open device fails with an error trying to read the iProduct field. The call to libusb_get_string_descriptor_ascii() fails.

Not sure why this test was added to FTDIpp_MPSSE().

Would a clearer error message help?

Is there a way of programming the FTDI device so that iProduct has a value?

Can this behaviour be (optionally?) disabled so that off-the-shelf FT232H devices can be used?

trabucayre, 2022-04-30:
In my mind a check of return code is required to see if it's a real error (access) or if it's only because this field is empty.
Thanks to point that!

DaveBerkeley, 2022-05-01:
It looks like it is returning an error code (ret < 0), not bytes_read==0. So I don't think the call can distinguish between an error and a string that has not been set. "lsusb -v" also says (error) for the field.
It would be good to be able to use cheap FT232H adapters, but perhaps they should have the data set? I'm not sure how to do this. I've always been wary of reprogramming the ROM on FTDI devices.

trabucayre, 2022-05-01:
Same conclusion.
I have updated ftdipp_mpsse: now a warning is displayed and `_iproduct` is set as an empty string. No more exception.
In fact this attribute is used for digilent probe and ch552 clone for Workarounds.
It's just weird, usually with adapters a custom iProduct is configured or default one is left unchanged, it's the first time I see that. It's always possible to rewrite eeprom with ftdi_eeprom or a code like [this](https://github.com/trabucayre/ft232_cbus_config)

DaveBerkeley, 2022-05-01:
That works fine, thanks.

closed 2022-05-01 by DaveBerkeley