trabucayre/openFPGALoader #229: configuration unsuccessful, if LiteX design contains DDR3

hansfbaier, 2022-05-19:
Hi, 
I struggled with this strange issue for a couple of days and finally found out that,
for some weird reason, configuration failed (DONE pin on the FPGA not coming high)
when I tried to load a LiteX design that contains DDR3 on a Kintex7 XC7K160T design,
with the DDR being on the high performance banks.
See https://github.com/litex-hub/litex-boards/pull/392
How could we approach to debug this?

trabucayre, 2022-05-20:
I assume without this commit everything is OK?
If so my first idea is to revert this commit, and firstly (without using DDR3) adding one by one properties.
Unfortunately if the problem remain and only appears when DDR3 is used the last solution (and not funny) is to dump USB traffic using wireshark and to analyze this.
I have received yesterday a Xilinx Platform cable (a clone) and I am currently working on it (but my current knowledge is actually a bit limited).
Definitively I have to order one kintex (if price is reasonable) to have basic support for this archictecture....

hansfbaier, 2022-05-21:
What commit are you referring to? I can't find any...

trabucayre, 2022-05-21:
True, my sentence is a bit unclear. When I talk about commit it the one you mention ie. litex-hub/litex-boards#392
It introduce some pins modifications and new properties: maybe one of them change the global behaviour of the FPGA.

hansfbaier, 2022-05-21:
Ah but before the commit the same symptoms showed.

trabucayre, 2022-05-21:
Could you **bisect** to see when this symptoms appeared ?

hansfbaier, 2022-05-24:
@trabucayre bisect on the openFPGALoader tree?

trabucayre, 2022-05-24:
It's after a flash write or loading into SRAM?

When I talk about bisect it's about modifications applied to the board at LiteX level, openFPGALoader's xilinx class has not being modified for some time

The idea is to see if by the past flashing or loading bitstream was working, where (which commit) the symptoms appeared (ie if it's related to a modification around FPGA configuration) and, consequently, to see if something is missing or if an adaptation is required for this device/kintex with this configuration.

I have search for your board and unfortunately it is really expensive :(

Thanks

hansfbaier, 2022-06-15:
Will need to try this too for the enclustra board

trabucayre, 2022-06-15:
It's interesting starting point! Maybe improving timings is required?
What it is newest acquisition? If it's a board with a reasonable price I can bought one to see why a slow clock is required because I have never seen with boards I have at home 

Thanks!

hansfbaier, 2022-06-15:
Yes this is a very good value for money board with a great price:
https://www.aliexpress.com/item/1005001275162791.html

trabucayre, 2022-06-16:
Yes seems a bit more realistic (maybe a bit high for hobbies) but it can't be shipped to france...

hansfbaier, 2023-01-03:
Hi I could reproduce this issue with the $99 QMTech board.
With the attached bitstream which has been generated by the open source toolchain.
[qmtech_xc7k325t.bit.gz](https://github.com/trabucayre/openFPGALoader/files/10333882/qmtech_xc7k325t.bit.gz)

hansfbaier, 2023-01-03:
Oh, no it actually works with latest openFPGALoader. So you can ignore this last post then.

hansfbaier, 2023-01-03:
Even works with 30MHz. Great.

trabucayre, 2023-01-04:
Maybe I have to order in the near futur one qmtech to improve support (current xilinx's support is primarily implemented for artix7).

hansfbaier, 2023-01-05:
@trabucayre The best value for money K325T is the STLV7325, if you want to spend $100 more
https://www.aliexpress.com/item/1005001275162791.html
But you get so much more out of that chip with it