trabucayre/openFPGALoader #230: Add support for libgpiod

niklasekstrom, 2022-05-22:
Would it be hard/possible to add support for libgpiod, similar to how openocd does, https://openocd.org/doc/html/Debug-Adapter-Configuration.html#index-linuxgpiod, so that a Raspberry Pi can be used as a JTAG programmer?

I would be interested in working on this, but it would be good to first get your insights about how hard/easy you think it would be.

trabucayre, 2022-05-23:
Hi,
I think this addition is not a too hard task (it's always easy to say that before starting ;-)).
Since this is bitbanging it's possible to inspire by [ftdiJtagBitbang](https://github.com/trabucayre/openFPGALoader/blob/master/src/ftdiJtagBitbang.cpp).

To summarise you have to create a class:
- inheriting to JtagInterface and implement (at least) `writeTDI`, `writeTMS` `setClkFreq`, ...
-  add your implementation for each of them by using `gpiod_line_set_value_bulk` when more than one pin have to be changed or `gpiod_line_set_value` when only one.

Thanks

niklasekstrom, 2022-05-23:
Thanks for your response and your comments!

I made a simple driver based on what you wrote, it's available here: https://github.com/niklasekstrom/openFPGALoader/blob/master/src/libgpiodJtagBitbang.cpp

I haven't been able to test it yet, other than that it compiles and runs on a Raspberry Pi, but I haven't connected it to an FPGA and programmed anything yet.

There are probably many things that could be done nicer in the code (for example I think it should be a preprocessor variable which says if this driver should be included in the build or not, and many other things) but until I have tested that it works I will leave it like this for now.

niklasekstrom, 2022-05-23:
I was able to test the driver with an XC95288XL CPLD, in so far that it reads and recognizes the idcode.

[image]

trabucayre, 2022-05-24:
Your implementation lokks good. Just small remarks / idea
- you open chip by num: is it possible to open by /dev ? And reuse **--device** option instead of new option?
- same question for **pins number** an option is already present, maybe reusing it with a small adaptation **FT232RL** vs **gpiod** 
- in method `update_pins` maybe tms and tck can be updated at the same time: it will reduce access userspace <-> kernelspace. Maybe to reduce calling functions it's more efficient to bufferize pins (more or less like for ft232) states.

for building this support only when asked by the user (and it's make sense since it's not possible to access gpio on x86 cpu) you can look for [cmsis support](https://github.com/trabucayre/openFPGALoader/blob/master/CMakeLists.txt#L13).

Thanks

trabucayre, 2022-05-25:
I have reread your code: seems good.
For `--pins` option my only worry is about time spend with shift and unshift at the `main` level but it's looks good

I think having a method used to play with both clock edges and read may be a good idea to reduce code redundency.
Theoriticaly if you update `tms` and `tdi` on falling/low `tck` level this must works but the `read_tdo` must be called after setting `tck` high since rising edge is the "read edge": it's more secure to have the `tdo` pin stable. You can't do predicate about device's delay to update `tdo`.

niklasekstrom, 2022-06-24:
Hi Gwenhael!

Thanks for reminding me. I got around to testing the libgpiod code today, both reading and writing to a Xilinx XC95288XL and to a Gowin GW1N-9C, and it seems to be working.

I added the change that you suggested, with reading after setting TCK high. It actually was working as it was before, but I agree that this seems more correct. https://github.com/niklasekstrom/openFPGALoader/commit/7e08b4c54df55cd5dfe8356ac3daf831a7d494ec

There's one thing I feel I haven't thought about enough, and that is what to set the _clkHZ variable to, which is used when timing erase and write operations. Setting _clkHZ to 5 MHz works fine, but it's possible that a lower value (meaning less waiting/faster writing) could work as well.

Of course it would be nice if someone else could test this code, but I think I'm fine with merging as it is now, as long as you think that the changes to the code not in libgpiodJtagBitbang.cpp/.hpp are fine. Those changes are mostly in that the pins_config fields are actual GPIO pin numbers instead of bit masks, and that the validation of GPIO pins are moved from the main file into the individual JTAG interface classes (FtdiJtagBitBang and LibgpiodJtagBitbang).

trabucayre, 2022-06-29:
I have to test your implementation but seems to be good to me.
clkFreq isn't really relevant but it's true for getClkFreq it's required to provides something (could you try with simply 1Hz or 1MHz)? In fact it's interesting: bsd talk about the delay in uS but if jtag is really slow I don't know if the delay (or at least a clock serie) is required or not.
Thanks

cr1901, 2022-06-30:
I've been testing the libgpiod branch locally, and while I _rarely_ am able to successfully read the JTAG ID from my attached MachXO2 device, the vast majority (like literally 99%) of `openFPGALoader` invocations result in the following:

```sh
sudo ./openFPGALoader -v 2 -d /dev/gpiochip5 --fpga-part machxo2  -c libgpiod --pins 10:12:11:19  ~/store/incoming/blinky-fast.bit
sudo ./openFPGALoader -v 2 -d /dev/gpiochip5 --fpga-part machx  -c libgpiod --pins 10:12:11:19  ~/store/incoming/blinky-fast.bit
```

It could be that the clock speed is too high for my GPIO breakout cable length; i2c from my SBC works fine attached to my breadboard. But I'd still like more info to debug. Is there a way to get `openFPGALoader` to give me more information, such as the data that was actually read back?

trabucayre, 2022-07-01:
@cr1901 thanks for your feedback. I will take profit to the week-end to test by myself (I haven"t machxo2 without onboard jtag).
Maybe it's time to open a PR to have a more easy way to fix / comment code.

cr1901, 2022-07-02:
@trabucayre With the following change:

```diff --git a/src/libgpiodJtagBitbang.cpp b/src/libgpiodJtagBitbang.cpp
index ccb0308..b737ad9 100644
--- a/src/libgpiodJtagBitbang.cpp
+++ b/src/libgpiodJtagBitbang.cpp
@@ -132,6 +132,8 @@ gpiod_line *LibgpiodJtagBitbang::get_line(unsigned int offset, int val, int dir)

 int LibgpiodJtagBitbang::update_pins(int tck, int tms, int tdi)
 {
+       static int cnt = 0;
+
        if (tdi != _curr_tdi) {
                if (gpiod_line_set_value(_tdi_line, tdi) < 0)
                        display("Unable to set gpio pin tdi\n");
@@ -147,10 +149,14 @@ int LibgpiodJtagBitbang::update_pins(int tck, int tms, int tdi)
                        display("Unable to set gpio pin tck\n");
        }

+       display("cnt: %d, tck: %d, tms: %d, tdi %d\n", cnt, tck, tms, tdi);
+
        _curr_tdi = tdi;
        _curr_tms = tms;
        _curr_tck = tck;

+       cnt++;
+
        return 0;
 }
```

and the following command-line :

```sh
sudo ./openFPGALoader -r --verbose-level 2 -d /dev/gpiochip5 -c libgpiod --pins 10:12:11:19  --fpga-part machxo2 ~/store/incoming/blinky-fast.bit
```

I was able to successfully program a new bitstream to my MachXO2's flash. Obviously, this is an unacceptable impl of a software delay except for debugging :P, but it does indicate that my SBC is not driving the GPIO pins with proper timing.

I guess the question now is... why isn't libgpio driving the pins with the proper timing? FWIW, I have my LA set up now, so I can capture traces. Here's an example of a trace with the following command line, the above patch, and a bit of extra logic to exit early: `sudo ./openFPGALoader --detect --verbose-level 2 -d /dev/gpiochip5 -c libgpiod --pins 10:12:11:19  --fpga-part machxo2 ~/store/incoming/blinky-fast.bit`

[image]

And here is another trace with the command line (`--verbose-level 2` removed): `sudo ./openFPGALoader --detect -d /dev/gpiochip5 -c libgpiod --pins 10:12:11:19  --fpga-part machxo2 ~/store/incoming/blinky-fast.bit`
[image]

AFAICT, it looks like TMS isn't timed properly unless I add a delay in the form of a `display` statement.

cr1901, 2022-07-02:
EDIT: The following patch gets me a _consistently working_ `libgpiod` driver for my SBC:

```diff --git a/src/libgpiodJtagBitbang.cpp b/src/libgpiodJtagBitbang.cpp
index ccb0308..c0fb9c0 100644
--- a/src/libgpiodJtagBitbang.cpp
+++ b/src/libgpiodJtagBitbang.cpp
@@ -175,6 +175,8 @@ int LibgpiodJtagBitbang::writeTMS(uint8_t *tms_buf, uint32_t len, bool flush_buf
                tms = ((tms_buf[i >> 3] & (1 << (i & 7))) ? 1 : 0);

                update_pins(0, tms, 0);
+               gpiod_line_set_value(_tck_line, 0);
+               gpiod_line_set_value(_tck_line, 0);
                update_pins(1, tms, 0);
        }
```

_Both_ dummy writes are required; without both, TMS will TCK's posedge, and with only one, I think there's a setup time violation or something (because while TMS comes before TCK's posedge, the MachXO2 only returns all ones).

Maybe we could workshop an acceptable patch from this for adding delays in terms of dummy `libgpiod` writes?

trabucayre, 2022-07-02:
Thanks for these informations: it's really usefull.
My guess with successive calls of `gpiod_line_set_value` was: each of them takes enough time to be stable/completed before next call.
For ftdi using [bitbanging](https://github.com/trabucayre/openFPGALoader/blob/master/src/ftdiJtagBitbang.cpp:133) this worls and I suppose all pins are updated more or less at the same time.
Maybe simply another work around is to write tck each time update_pins is called instead of conditionning that update to a state change.

trabucayre, 2022-07-12:
@cr1901 I have tried a subset of commands used in this driver, latencies between each pins changes are quite large, I can't see this invertion. But it's maybe hardware (Linux driver) related: I have tried with a pizero2
@niklasekstrom I have tried to use `xxx_bulk` for all output pins: latencies a shortest and consequently it's allows to discard test about pins states and to send inconditionnaly the value for all 3 pins at the same time.
I have do cleanup my test class and create a gist.

cr1901, 2022-07-13:
@trabucayre If you can't duplicate, then it's probably a hardware or kernel bug on my end. How stable are command-line options supposed to be? Is it possible to add e.g. a `--tmsdelay` option that adds dummy waits so TMS has time to settle before sampling on the next TCK posedge?

trabucayre, 2022-07-13:
This [gist](https://gist.github.com/trabucayre/2f31c0b46bd4d95af1287d4e6eecd3e6) show an example of bitbanging with _bulk functions. Timings matches requirement.
@cr1901 I'm not opposite to add new CLI options but with a logic analyzer I see around 200kHz clock period with large delays between TDI/TMS updates before TCK rising edge (TMS/TDI are updated around 250ns after TCK falls). So I'm not sure adding delays is really required.

cr1901, 2022-07-13:
>Have you tried my demo with your board to check timing/sequence?

No, I will do that tonight. Right now, I'm trying to reach a good stopping point on my current project.

>Another question: which is your machXO2 board?

[TinyFPGA AX2](https://tinyfpga.com/)

cr1901, 2022-07-14:
Okay, so just to confirm that things are working _with my patch_, I've programmed my MachXO2 just a few seconds ago:

```
wjones@DietPi:~/src/FPGA/openFPGALoader/build$ ./openFPGALoader -d /dev/gpiochip5 -c libgpiod --pins 10:12:11:19  --fpga-part machxo2 ~/store/incoming/blinky-slow.bit
Open file: DONE
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
Disable configuration: DONE
```

From the above command line, I needed to use the following defines in your gist:

```cpp
// TDI:TDO:TCK:TMS
// /dev/gpiochip5 --fpga-part machxo2  -c libgpiod --pins 10:12:11:19 
#define DEV_GPIO "/dev/gpiochip5\0"
/* PIN 32 */
#define TCK 11
/* PIN 34 */
/* GND */
/* PIN 36 */
#define TDI 10
/* PIN 38 */
#define TDO 12
/* PIN 40 */
#define TMS 19
```

When I compile and run your gist, I get the following:

```sh
wjones@DietPi:~/src/debug/libgpiod$ g++ -o gpio gpio.cpp -lgpiod && ./gpio 
Error requesting gpio lines
Error set default value
Error requesting gpio line 12
Erreur init
```

Hmmm, wonder what went wrong?

My `udev` rule for `/dev/gpiochip5` looks like this (i.e. using `sudo` doesn't make a difference):

```sh
wjones@DietPi:~/src/debug/libgpiod$ cat /etc/udev/rules.d/60-gpiod.rules 
# udev rules for gpio port access through libgpiod
SUBSYSTEM=="gpio", KERNEL=="gpiochip5", GROUP="gpiod", MODE="0660"
```

trabucayre, 2022-07-14:
I'm currently building a buildroot system for my rockpro64 (based on a rockchip too). A soon as I have a working system I try too.
Could you comment line 184->187 and 206: `open_device` and `close_device` are now handled in CTOR/DTOR (sorry for this mistake).
I have to try to get a tinyFPGAAX2 it's interesting to me to have a machXO2 board without dedicated interface.

cr1901, 2022-07-14:
With the relevant lines commented, this is the full LA trace I get:

[image]

Nothing on the TDO line. I'm not actually sure what the `main` program is doing, nor am I sure what a "correct" trace should look like. But I'm assuming this isn't correct :P.

trabucayre, 2022-07-14:
`main` program is just used to send a 16 bits sequence without anything clever for jtag. `TDI` must toggle each period and `TMS` is left unchanged during two periods.
Second loop is to send one char with TMS low and goes high with last bit.
It's true in your case `TMS` transition is really (too) near TCK rise. What is the delay between `TMS` transition and `TCK` transition?
In fact it's seems to follow pins order: 1 tdi, 2 tck, 3 tms. It seems to add a delay (ie more than one transition between tms/tdi changes and tck high) make sense. With the question to hardcode this for security or let user choise? First may be more transparent than failure and the need to use a LA to verify timings.

trabucayre, 2022-07-15:
> Approximately 0.5-2.5us, when I zoom in on the relevant trace between TMS transition, and the _next_ TCK posedge. TMS posedge transitions seem to be closer to 0.5us, and negedge transitions seem to be closer to 2.5us.
> 
According to **MachXO2 Family datasheet** tBTS is min 10ns so your timings may be compatible
> What about, if reading the IDCODE fails, the GPIO driver adds an increasing number of delays automatically to reset into Test Logic Reset (or Run Test Idle) and try to read the IDCODE again, failing if the delay amount exceeds a certain amount?
I'm not really happy when high level has to know to many thing about lowlevel.

cr1901, 2022-07-15:
>I'm not really happy when high level has to know to many thing about lowlevel.

Then let's just add a dummy delay of two `gpiod_line_set_value(_tck_line, 0);` with a comment that it's a "guard for misbehaving boards where TMS may sometimes transition after TCK posedge and miss timing".

cr1901, 2022-07-16:
I have determined that the reason I required this patch was because I forgot that the GPIO I was using for the `TMS` line (19( was _already_ attached to another circuit. The load must've been unacceptable for the GPIO pin (there's a pullup somewhere), which made it take a very long time to settle into the correct state.

Using the following pins/cmdline, `openFPGALoader` now works _without_ my patch:

```sh
./openFPGALoader -d /dev/gpiochip5 -c libgpiod --pins 10:11:12:14  --fpga-part machxo2 ~/store/incoming/blinky-fast.bit
```

niklasekstrom, 2022-07-18:
Hi guys. Sorry for the radio silence, I was away on vacation, but I came home yesterday.

I will read what you have written, and then rebase my fork on the head of the upstream, and create a PR.

niklasekstrom, 2022-07-18:
I read your comments, although not that carefully tbh. I get a sense that much (most?) of what @cr1901 writes are related to the TMS pin having a too high load, and with that problem fixed then the original code works.

You @trabucayre mentioned that using the _bulk functions work well for you. Would you mind doing a quick branch with those changes, and then I can try that on the XC95288XL CPLD I have?

I rebased my additions on top of the current head, and created a PR:  https://github.com/trabucayre/openFPGALoader/pull/242. Maybe it's easier to comment there, as you mentioned.

closed 2022-07-27 by niklasekstrom