trabucayre/openFPGALoader #241: Tang Nano board unable to flash

paller, 2022-07-14:
I have two Tang Nano boards both give the same error when trying to flash. As it's terrible to get them working on Windows with "official" tools I really hope to get openFPGALoader to work. The output below is from Fedora 35.

This is the output when writing to memory:
```
sudo openFPGALoader -b tangnano -v neopixel_tang_nano.fs
write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices
idcode 0x900281b
manufacturer Gowin
model  GW1N-1
family GW1N
File type : fs
Parse file Parse neopixel_tang_nano.fs:
Warning: Unknown IDCODE
checksum 0x0000
Done
DONE
bitstream header infos
CRCCheck: OFF
CheckSum: a7b8
Compress: OFF
ConfDataLength: 548
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 12005036
loading_rate: 0
Jtag frequency : requested 2.50MHz   -> real 2.00MHz
displayReadReg 00011422
	Bad Command
	Memory Erase
	Non-jtag is active
	Gowin VLD
	POR
pollFlag: 110a2
erase SRAM pollFlag: 110a2
Done
pollFlag: 11422
pollFlag: 110a2
Flash SRAM: [======================================            ] 74.794922%pollFlag: 110a2
pollFlag: 110a2
pollFlag: 110a2 <-- will continue to print this over and over again.
```

Output when writing to flash:
```
sudo openFPGALoader -b tangnano -v -f neopixel_tang_nano.fs
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices
idcode 0x900281b
manufacturer Gowin
model  GW1N-1
family GW1N
File type : fs
Parse file Parse neopixel_tang_nano.fs:
Warning: Unknown IDCODE
checksum 0x0000
Done
DONE
bitstream header infos
CRCCheck: OFF
CheckSum: a7b8
Compress: OFF
ConfDataLength: 548
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 12005036
loading_rate: 0
Jtag frequency : requested 2.50MHz   -> real 2.00MHz
pollFlag: 110a2
erase SRAM pollFlag: 110a2
Done
pollFlag: 11422
pollFlag: 110a2
erase Flash Done
pollFlag: 11422
write Flash: [==================================================] 100.000000%
Done
pollFlag: 11422
CRC check: Success
displayReadReg 00011422
	Bad Command
	Memory Erase
	Non-jtag is active
	Gowin VLD
	POR
```
While this goes to 100% it doesn't seem to work either.

I have used both boards in the past and they worked fine. I don't know if the FTDI-clones used on these boards may have been bricked by the official FTDI drivers after having plugged the boards into a Windows PC. I hope it's not the case.

trabucayre, 2022-07-14:
This is not due to a FTDI clone bricked, otherwise openFPGALoader wose not able to detect and use it...
Could you share your .fs ? 
```
Warning: Unknown IDCODE
checksum 0x0000
```

Is weird to me.

Don't try to write into internal flash this board: the ch552 firmware is a nightmar, I try frequently to find a way to deal correctly with this device but never found the ultimate solution :-/

paller, 2022-07-14:
Sorry, pressed the wrong button.

trabucayre, 2022-07-14:
I've downloaded the file. Now I need to check this to see if something is wrong in the parser or in the fs. Thanks.
Yes an issue is opened but project seems dead and sipeed for new tang nano (1k, 4k,9k) has changed jtag interface for something more stable (only fails with used on a pi4...)

trabucayre, 2022-07-15:
I have analyzed the fs you was provides and everything seems to be coherent. This file is correctly analyzed, informations are extracted.
Could you provides openFPGALoader's version and if you use windows or a Linux distro (which one)?

paller, 2022-07-15:
Used the fedora-35-x86_64 rpm from https://copr.fedorainfracloud.org/coprs/mobicarte/openFPGALoader/build/3770136/ which builds from https://github.com/trabucayre/openFPGALoader/archive/v0.8.0/openFPGALoader-v0.8.0.tar.gz

I will try
- Building openFPGALoader from source
- Update ch522 firmware

Also, this is [hilarious](https://twitter.com/SipeedIO/status/1547759294723944448)
[image]

trabucayre, 2022-07-15:
I use (of course) HEAD but there is nothing about fsParser between v0.8.0 and HEAD... It's weird and I don't see why fs is not correctly parsed...
Yes new tangnano have a BL702 instead of ch552. It's a good thing because the new interface works better (there is an issue with rpi4 but it's not at openFPGALoader level but kernel level). But my problem is: it's not a fix for previous release using ch552 device.
How to update is explained [here](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sipeed_tang_nano.py#L11) and I have written a *stupid* tool to swith device in programming mode [here](https://github.com/trabucayre/jump_ch552_to_bl)

paller, 2022-07-22:
I have previously used openFPGALoader for Xilinx which is why I started using it when the GoWin Programmer failed me on Tang Nano.
It's a little odd, between 3 laptops only one is able to program the board.

Laptop 1: AMD Ryzen, Windows 11 Home, can't initialize the USB device correctly. Error code 43 every time.
Laptop 2: Intel i5, Windows 11 Pro, work's perfectly without any problems using the GoWin programmer and FTDI drivers.
Laptop 3: Intel i3, Fedora 35, dmesg indicates the board is setup with the correct driver, openFPGALoader unable to program RAM and flash.

I managed to compile and load the alternative ch552 firmware from https://github.com/diodep/ch55x_jtag unfortunately it didn't help. It still works on _Laptop 2_ with the GoWin Programmer but not with openFPGALoader.

Only progress is that I now know the .fs file works and will make the RGB diode blink when it is loaded correctly.

Do you have any suggestions for what I can try in order to debug this?

paller, 2022-07-22:
I'm sorry, there's no problem after all.

I tried building from source and it works just fine. Then I compared version numbers and I happened to have an old version (0.2.6) in /usr/local/bin which got called instead of the one installed through the packet manager.

closed 2022-07-22 by paller

trabucayre, 2022-07-23:
My fault: I have missed to ask for openFPGALoader version :-)
Thanks!