trabucayre/openFPGALoader #245: unable to open ftdi device: -4 (usb_open() failed)

infinitesteps, 2022-08-12:
Problem:

```
$ openFPGALoader -b arty_a7_35t /home/foo/git/external/f4pga-examples/xc7/counter_test/build/arty_35/top.bit
unable to open ftdi device: -4 (usb_open() failed)
JTAG init failed with: unable to open ftdi device
```

More info:

```
$ uname -a
Linux baz 5.10.0-16-amd64 #1 SMP Debian 5.10.127-2 (2022-07-23) x86_64 GNU/Linux

$ openFPGALoader -V
openFPGALoader v0.8.0

$ lsusb
Bus 004 Device 007: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
...

$ sudo dmesg |grep -i usb
...
usb 4-6: Detected FT2232H
usb 4-6: FTDI USB Serial Device converter now attached to ttyUSB1

$ ls -l /dev/ttyUSB1
crw-rw---- 1 root dialout 188, 1 Aug 12 13:38 /dev/ttyUSB1

$ getent group dialout
dialout:x:20:foo

$ picocom -b 115200 --imap lfcrlf /dev/ttyUSB1
...
**        Avnet/Digilent Arty Evaluation Board        **
**        LEDs and switches GPIO Demonstration        **
********************************************************
********************************************************
**
Choose Task:
BTN0: Print PWM value.
BTN1: 'Cylon' LED display.
...

$ openFPGALoader -b arty_a7_35t --detect
unable to open ftdi device: -4 (usb_open() failed)
JTAG init failed with: unable to open ftdi device

```

trabucayre, 2022-08-12:
This issue is similar to #246. Everything seems fine (group, rw permissions).
I have to find a way to reproduce this issue. On all my computers this works fine.

trabucayre, 2022-08-12:
Could you confirm nothing else use /dev/ttyUSB0?

cfriedt, 2022-08-12:
I'm on Ubuntu Jammy in #246 if that helps

infinitesteps, 2022-08-12:
When I connect, I get these kernel messages. Sorry I didn't realize that it used both /dev/ttyUSB0 and /dev/ttyUSB1

```
[ 3898.979556] usb 1-6: new high-speed USB device number 9 using ehci-pci
[ 3899.136500] usb 1-6: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 7.00
[ 3899.136502] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3899.136505] usb 1-6: Product: Digilent USB Device
[ 3899.136506] usb 1-6: Manufacturer: Digilent
[ 3899.136507] usb 1-6: SerialNumber: 2<snip>
[ 3899.139777] ftdi_sio 1-6:1.0: FTDI USB Serial Device converter detected
[ 3899.139792] usb 1-6: Detected FT2232H
[ 3899.140203] usb 1-6: FTDI USB Serial Device converter now attached to ttyUSB0
[ 3899.140380] ftdi_sio 1-6:1.1: FTDI USB Serial Device converter detected
[ 3899.140393] usb 1-6: Detected FT2232H
[ 3899.140704] usb 1-6: FTDI USB Serial Device converter now attached to ttyUSB1
```

No Linux process seem to have control of it, the following command gives no output:

```
$ lsof /dev/ttyUSB*
```

I am on: Debian GNU/Linux 11 (bullseye)

infinitesteps, 2022-08-12:
I fixed it. I had to add the following udev rule:

File: `/etc/udev/rules.d/99-ftdi.rules`

```
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="0666"
```

For others who arrive here with different boards, change the vendor and product to match the hex values produced by lsusb. For example here is mine:

`Bus 002 Device 004: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC`

trabucayre, 2022-08-13:
Thanks!
A  udev rule is provided with openFPGALoader (https://trabucayre.github.io/openFPGALoader/guide/install.html#udev-rules) but maybe this note is not correctly highlity and maybe with the binary version this file is not available.

cfriedt, 2022-08-13:
@infinitesteps - slight modification (does not grant access to users not in the 'dialout' group). Oddly, the apparent permissions were exactly the same for me before and after adding this udev rule but it does indeed work :-)

```bash
# add user to dialout group
sudo usermod -a -G $USER dialout
# gain new group access without logging out/in
exec su -l $USER
# add udev rules
sudo cat <<< EOF > /etc/udev/rules.d/99-ftdi.rules
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="0660", GROUP="dialout"
EOF
sudo udevadm control --reload-rules && sudo udevadm trigger
```

trabucayre, 2022-08-13:
It's true: when you add your user to a new group you have to logout and login for modifications takes effects... Or using sudo to gain group/permission update locally.
The rule you add and commands are exactly ones mentionned into openFPGALoader's documentation but it's seems clear this section have not enought visibility.
Thanks!

trabucayre, 2022-08-16:
> * I came to openFPGALoader indirectly through F4PGA. I was following their documentation and so it might not be a fault of the this projects documentation.
>
openFPGALoader's documentation must be improved: you haven't found required information this mean documentation is not clear or too limited
> 
> * Thanks for the tip on the 0660 @cfriedt - I thought it was odd that it needed world permission. I don't know how udev works but the filesystem permissions were 0660 without the udev rule and it was not working for me (even after logout). I even changed the filesystem perms to 0666 without luck. It wasn't until the udev rule that things started working.
>
This is weird yes. rules file provided by openFPGALoader set right for others to 4. Content of this issue must be resumed in [faq](https://trabucayre.github.io/openFPGALoader/guide/troubleshooting.html).
>
> * @trabucayre I got a clue about the problem when I built and ran [simple.c](http://developer.intra2net.com/git/?p=libftdi;a=blob;f=examples/simple.c;h=8abd7b3944a88ebf9a0e5bc9791f3b1872d07295;hb=6935ef7bd3cd0666438423285736cece988c2c7f) from the libftdi project (that corresponded to my systems version of libftdi). The error message was more informative. I can't remember exactly how the `ftdi_usb_open()` is called but I don't think it uses the error message provided by libftdi - that might be helpful.
>
Currently, openFPGALoader displays libftdi message too, but a more explicit message must be added with some tips to fix issue (it's true here but for all error too).

verilog-indeed, 2022-09-04:
Hello, this problem also occurs to me on two different Windows machines (7 and 10), I get the exact same error when using --detect on a Tang Nano 4K board, is there an equivalent fix for Windows? let me know if you need further debugging information.

Chandler-Kluser, 2022-09-05:
> I fixed it. I had to add the following udev rule:
>
> File: `/etc/udev/rules.d/99-ftdi.rules`
>
> ```
> ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="0666"
> ```
>
> For others who arrive here with [...]

Thanks a lot!! It helped me, too

CoruNethron, 2022-12-03:
Hello, thx for debug tips. Same issue as @verilog-indeed , unable to flush Tang Nano on Windows 10. openFPGALoader is running from commandline, so there is no /etc/udev/rules.d
Running CMD with administrator priviledges doesn't seem to help as well
In the "device manager" there are two devices named "USB Serial Converter A/B", both with vid:pid 0403:6010
Just in case, under MSYS2 "ls /dev" shows ttyS[0-9] device instead of ttyUSB[0-9] when board is plugged in.
Please let me know as well if I can help to continue debug under windows.

CoruNethron, 2022-12-03:
Managed to workaround this, as mentioned here:
https://esp32.com/viewtopic.php?t=15469

Using this tool:
https://zadig.akeo.ie/

[image]

Replacing driver for Sipeed-JTAG from FTDIBUS to WinUSB seems to solve the issue. openFTDILoader starts to correctly detect device and flush it.

Not sure, what this tool really does, cause of lack of knowledge about USB stack.

trabucayre, 2022-12-04:
I'm absolutely no knowledge about windows but as far I know to be able to use the libftdi with windows it's required to replace driver: and it's job of this tool.
Thanks for this information