trabucayre/openFPGALoader #246: Unable to connect to arty-a7-100t

cfriedt, 2022-08-12:
Hi - I was just following the instructions here
https://f4pga-examples.readthedocs.io/en/latest/building-examples.html

and everything works except for openFPGALoader. I turned on verbose `display()` messages, and this is the output I get

```bash
(xc7) cfriedt@ubuntu:~/workspace/f4pga-examples/xc7$ TARGET="arty_100" make download -C counter_test
make: Entering directory '/home/cfriedt/workspace/f4pga-examples/xc7/counter_test'
if [ arty_100='unsupported' ]; then \
  echo "The commands needed to download the bitstreams to the board type specified are not currently supported by the F4PGA makefiles. \
    Please see documentation for more information."; \
fi
The commands needed to download the bitstreams to the board type specified are not currently supported by the F4PGA makefiles.     Please see documentation for more information.
openFPGALoader -b arty_a7_100t /home/cfriedt/workspace/f4pga-examples/xc7/counter_test/build/arty_100/top.bit
try to open 403 6010 -1 -1
unable to open ftdi device: -4 (usb_open() failed)
JTAG init failed with: unable to open ftdi device
make: *** [/home/cfriedt/workspace/f4pga-examples/xc7/counter_test/../../common/common.mk:86: download] Error 1
make: Leaving directory '/home/cfriedt/workspace/f4pga-examples/xc7/counter_test'
```

Any suggestions?

Note: it's not a permissions issue (i am part of the `dialout` group and have verified write permission)

cfriedt, 2022-08-12:
I'm on Ubuntu Jammy if that helps

cfriedt, 2022-08-12:
Apparently this is also an issue with `x3csprog` (so likely libftdi is the culprit)

```bash
$ xc3sprog -c nexys4 counter_test/build/arty_100/top.bit
XC3SPROG (c) 2004-2011 xc3sprog project $Rev$ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

Could not open FTDI device (using libftdi): usb_open() failed
Unable to access FTDI device with either libftdi or FTD2XX
```

trabucayre, 2022-08-13:
Yes this error is at `libftdi` or `libusb` level and error message a bit cryptic.
Is the solution provided in #245 is working for you?

cfriedt, 2022-08-13:
@trabucayre - a similar solution, also similar to what is mentioned in https://trabucayre.github.io/openFPGALoader/guide/install.html#udev-rules

Oddly, the actual permissions / groups on `/dev/ttyUSB*` (and my account) are the same before and after this, which is somewhat confusing. Not sure what additional magic the udev rule provides, but it does work. Thanks :-)

```bash
# add user to dialout group
sudo usermod -a -G $USER dialout
# gain new group access without logging out/in
exec su -l $USER
# add udev rules
sudo cat <<< EOF > /etc/udev/rules.d/99-ftdi.rules
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="0660", GROUP="dialout"
EOF
sudo udevadm control --reload-rules && sudo udevadm trigger
```

closed 2022-08-13 by cfriedt

trabucayre, 2022-08-13:
Thanks for your feedback!