trabucayre/openFPGALoader #249: ULX3S: Invalid pin ID

mole99, 2022-08-23:
Previously I could upload to the ULX3S without problems:

```
> openFPGALoader --board=ulx3s ulx3s.bit
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Open file: DONE
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
Disable configuration: DONE
```

But after updating to the latest release, I get:

```
> openFPGALoader --board=ulx3s ulx3s.bit
Invalid pin ID
JTAG init failed with: std::exception
```

I don't have the exact commit where it broke, but I can look for it if needed.

trabucayre, 2022-08-23:
This regression was introduces when **libgpiod** support was added. pins configuration are now described by it's ID instead of shift (`4` instead of `1 << 4`).
I have pushed a fix: works with my ulx3s but could you confirm?
Thanks

mole99, 2022-08-23:
Yes, I can confirm that it fixed the issue. Thank you very much :)

trabucayre, 2022-08-23:
Thanks for your feedback!

closed 2022-08-23 by trabucayre

trabucayre, 2022-09-14:
I did not have planned to create a new release soon, but it's true with this regression now fixed and GW2A's MSPI mode added it make sense to create at leas a minor release (0.9.1).
I will doing that ASAP.
Thanks!

trabucayre, 2022-09-19:
@carlosedp with a bit of late: done!
Thanks!

carlosedp, 2022-09-19:
Thanks, sent a PR bumping Homebrew.
https://github.com/Homebrew/homebrew-core/compare/master...carlosedp:homebrew-core:patch-1