trabucayre/openFPGALoader #250: tangprimer20k program slow and stucked

Zepan, 2022-08-26:
Hi, thank you for support tangprimer20k~
I just tested "tangprimer20k" options, and seems it have some bug,
when program flash, it use 23s program 1%...
when program sram, it stucked at about 78%
how can I figure out the problem? did it appears on your board?

trabucayre, 2022-08-26:
Hi,
for SRAM it's strange: I have my board since 2 weeks and not observed this type of behaviour, I assume @Icenowy (whos has added GW2A support) and @enjoy-digital have not seen that too: could you provide your setup, sometime bl702 programmer has issue and/or could you try with another programmer (if it's possible)?
for flash: I know currently flashing takes around 10mins: it's awfully slow. It's one of task I have to resolved but gowin datasheet provides nothing specifics for this device... WIP

trabucayre, 2022-09-04:
Any news?
I have played a lot with my tangPrimer (SRAM) and I have never seen any issue.
For Flash I read again Gowin programming datasheets: GW2A is mentionned but it's lite and many informations are just missing... I have to find a way to obtain required params...

Chandler-Kluser, 2022-09-05:
I am facing a lot of issues with Tang Primer 20k too, I received mine in 2 days ago.

the most annoying issue is the device is not being recognized, even when root permissions.

the cases I am able to program (5% of total cases) flash memory it takes 10 minutes

trabucayre, 2022-09-05:
Could you add -v option and provides dump? It's weird because it's exactly the same interface as tang nano 4k and 9k...
For Flash: SPI access was added for GW1N and is based on *Boundary scan*. It's a bitbang and take lot of time. I'm working to add native mode (something similar to SPI access for lattice ECP5) but docs are really light about this topic.

Chandler-Kluser, 2022-09-05:
> Could you add -v option and provides dump? It's weird because it's exactly the same interface as tang nano 4k and 9k... For Flash: SPI access was added for GW1N and is based on _Boundary scan_. It's [...]

The issue in #251 was solved by erasing embdFlash in a windows machine with gowin's programmer, the issue of 5% flashing times (with permissions) I could solve now thanks to #245 , as I [commented there](https://github.com/trabucayre/openFPGALoader/issues/245#issuecomment-1236584995).

but the slow flash time is a fact, now. I will paste with verbose activated soon!

Chandler-Kluser, 2022-09-05:
```
$ openFPGALoader -b tangprimer20k -v -f tang-primer-demo.fs
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
File type : fs
Parse file Parse tang-primer-demo.fs: 
checksum 0x4876
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 1342
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 0000081b
loading_rate: 0
Jtag frequency : requested 2.50MHz   -> real 2.00MHz  
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
pollFlag: 60a0
erase SRAM pollFlag: 4080
pollFlag: 4080
pollFlag: a0
Done
b 40 16 b read b40160b
Detail: 
Jedec ID          : 0b
memory type       : 40
memory capacity   : 16
EDID + CFD length : 0b
EDID              : 1640
CFD               : 0b 40 16 0b 40 16 0b 40 16 
b 40 16 b read b40160b
Detail: 
Jedec ID          : 0b
memory type       : 40
memory capacity   : 16
EDID + CFD length : 0b
EDID              : 1640
CFD               : 0b 40 16 0b 40 16 0b 40 16 
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
pollFlag: 20
```

trabucayre, 2022-09-05:
Thanks for your feedback.
Documentation and troubleshoothing have to be updated to be more clear/detailed. Last time I have installed a new distro on a computer I haven't seen issues (but clearly I know exactly howto do -> I'm not objective... I'm interessted by an external contribution).
I'm working on MSPI access (instead of bitbanging) but documentation are limited to a state machine and a chrono -> I have some resultts but I'm far to have something correctly working -> WIP.

Chandler-Kluser, 2022-09-05:
keep the great work, my friend

if there is something I can help, here I am!

trabucayre, 2022-09-06:
@Zepan, @Chandler-Kluser: I have updated repository with a first MSPI draft: now it's takes ~9s to write the bitstream.

@Chandler-Kluser If you want you, @cfriedt and/or @infinitesteps, updating [documentation](https://github.com/trabucayre/openFPGALoader/blob/master/doc/guide/install.rst#udev-rules) and [troubleshooting](https://github.com/trabucayre/openFPGALoader/blob/master/doc/guide/troubleshooting.rst} with your findings it maybe useful for everyone.

Chandler-Kluser, 2022-09-07:
@trabucayre I have recompiled the repo and reinstalled it in my PC with [this commit](https://github.com/trabucayre/openFPGALoader/commit/fe259fb78d185b3113661d04cd7efa9ae0232425) and the flash time in Tang Primer is way more faster, thanks a lot!!

About the documentation and troubleshooting, I will make some pull requests to update these markdowns.

trabucayre, 2022-09-07:
Yep around 60 time faster on my computer ;-)
Finally this time two figures are enough to add necessary code to access SPI flash :-)
Thanks for your feedback!
@zepan: any update?

Zepan, 2022-09-08:
Hi, I just pull and compile, test ok~ 
Program flash and sram is fine and fast, thank you for your work!

trabucayre, 2022-09-13:
@Zepan thanks for your feedback. MSPI mode is clearly better when it's available (for devices booting on external Flash). For SRAM I haven't modified anything: I suspect some glitches on JTAG lines.

closed 2024-06-13 by trabucayre