trabucayre/openFPGALoader #251: Cannot flash Tang Nano 9k

Chandler-Kluser, 2022-08-31:
I cannot flash my Tang Nano 9K with a default [led example project provided by Sipeed](https://github.com/sipeed/TangNano-9K-example/tree/main/led).

I have fully compiled the bitstream file and used openFPGALoader to write to flash memory:
```
$ openFPGALoader -b tangnano9k -f 9K_LED_project.fs
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
JTAG init failed with: Unknown device with IDCODE: 0x00fa03fa (manufacturer: 0x1fd (), part
```

I have read issues #245 and [this question from Stack Overflow](https://stackoverflow.com/questions/69912368/unable-to-open-ftdi-device-with-vid-0403-pid-6010), my user has all permissions to access FTDI device, which is listed by:
```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse
Bus 003 Device 008: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Bus 003 Device 002: ID 214b:7250 Huasheng Electronics USB2.0 HUB
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID c0f4:01b0 USB usb keyboard
Bus 001 Device 003: ID 8087:0025 Intel Corp. Wireless-AC 9260 Bluetooth Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Here it is my rules.d file for this device:
```
$ cat /etc/udev/rules.d/70-ttyusb.rules
KERNEL=="ttyUSB[0-9]*", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28e9", ATTRS{idProduct}=="0189", MODE="0666"
SUBSYSTEM=="usb_device", ATTRS{idVendor}=="28e9", ATTRS{idProduct}=="0189", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", MODE="0666"
```
This error happened only once, and then I couldn't even access the device anymore:
```
$ openFPGALoader -b tangnano9k -f 9K_LED_project.fs
write to flash
unable to open ftdi device: -3 (unable to set usb configuration. Make sure the default FTDI driver is not in use)
JTAG init failed with: unable to open ftdi device
```
Neither with root priviledge:
```
$ sudo openFPGALoader -b tangnano9k 9K_LED_project.fs
unable to open ftdi device: -3 (unable to set usb configuration. Make sure the default FTDI driver is not in use)
JTAG init failed with: unable to open ftdi device
```
Is there something I missed? Thanks in advance!

trabucayre, 2022-08-31:
Could you provides openFPGALoader version and your setup (computer/SBC and OS).
The first result shows an align error: it may appear sometime when using fly wire but with an onboard interface it's a bit weird.
Could you try to `sudo modprobe -r ftdi_sio` ?
Maybe a `dmesg` dump will be useful.
Thanks

Chandler-Kluser, 2022-08-31:
- my computer:
[image]
- openFPGALoader version:
```
$ openFPGALoader -V
openFPGALoader v0.7.0
```
- openFPGALoader was built from source.
- `sudo modprobe -r ftdi_sio` didn't give any output.
- dmesg dump:
```
(...)
[25480.743278] usb 3-1.4: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 5.00
[25480.743282] usb 3-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[25480.743284] usb 3-1.4: Product: JTAG Debugger
[25480.743285] usb 3-1.4: Manufacturer: SIPEED
[25480.743287] usb 3-1.4: SerialNumber: FactoryAIOT Pro
[25480.763931] usbcore: registered new interface driver usbserial_generic
[25480.763943] usbserial: USB Serial support registered for generic
[25480.766579] usbcore: registered new interface driver ftdi_sio
[25480.766595] usbserial: USB Serial support registered for FTDI USB Serial Device
[25480.766660] ftdi_sio 3-1.4:1.0: FTDI USB Serial Device converter detected
[25480.766708] usb 3-1.4: Detected FT2232C
[25480.766993] usb 3-1.4: FTDI USB Serial Device converter now attached to ttyUSB0
[25480.767042] ftdi_sio 3-1.4:1.1: FTDI USB Serial Device converter detected
[25480.767102] usb 3-1.4: Detected FT2232C
[25480.767265] usb 3-1.4: FTDI USB Serial Device converter now attached to ttyUSB1
[25481.559410] usbserial: USB Serial deregistering driver FTDI USB Serial Device
[25481.559512] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[25481.559567] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[25481.559583] usbcore: deregistering interface driver ftdi_sio
[25481.559628] ftdi_sio 3-1.4:1.1: device disconnected
[25481.559660] ftdi_sio 3-1.4:1.0: device disconnected
[25481.597589] usbserial: USB Serial deregistering driver generic
[25481.597607] usbcore: deregistering interface driver usbserial_generic
```
===============================================================================
I have also tried to flash through a virtual machine with USB port forwarding:
- VM specs:
[image]
- openFPGALoader version:
```
$ openFPGALoader -V
openFPGALoader v0.5.0
```
- built via pacman (`sudo pacman -S openfpgaloader`).
- `sudo modprobe -r ftdi_sio` didn't give any output either.
- dmesg dump:
```
(...)
[  476.231814] usb 1-3: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 5.00
[  476.231819] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  476.231822] usb 1-3: Product: JTAG Debugger
[  476.231824] usb 1-3: Manufacturer: SIPEED
[  476.231825] usb 1-3: SerialNumber: FactoryAIOT Pro
[  476.795321] usbcore: registered new interface driver ftdi_sio
[  476.795331] usbserial: USB Serial support registered for FTDI USB Serial Device
[  476.795348] ftdi_sio 1-3:1.0: FTDI USB Serial Device converter detected
[  476.795365] usb 1-3: Detected FT2232C
[  476.796004] usb 1-3: FTDI USB Serial Device converter now attached to ttyUSB0
[  476.796030] ftdi_sio 1-3:1.1: FTDI USB Serial Device converter detected
[  476.796041] usb 1-3: Detected FT2232C
[  476.796479] usb 1-3: FTDI USB Serial Device converter now attached to ttyUSB1
(...)
[  485.831139] usbserial: USB Serial deregistering driver FTDI USB Serial Device
[  485.941562] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[  485.941634] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[  485.941648] usbcore: deregistering interface driver ftdi_sio
[  485.941670] ftdi_sio 1-3:1.1: device disconnected
[  485.941693] ftdi_sio 1-3:1.0: device disconnected
```

trabucayre, 2022-08-31:
You use an intel x86 64bits computer: I have not seen issue in this situation (I have seen issue with some ARM boards).
Sorry my question was not clear: `sudo modprobe -r ftdi_sio` is used to remove driver used for FTDI devices. You must retry to load bitstream (Could you try in a first time to load into SRAM?) and after then make a dmesg dump.
v0.7.0 is quite old (9k support was added after this release): could you try with master branch ?
Thanks.

Chandler-Kluser, 2022-08-31:
"Could you try in a first time to load into SRAM?" - yes, I have already tried...

I will compile from source and try again with the newer version

trabucayre, 2022-09-01:
loading to SRAM is working or same issue appear?

Chandler-Kluser, 2022-09-02:
I compiled from source again in a new Manjaro with kernel 5.15.60-1-MANJARO

I am going to make the same procedure and paste the output here

Chandler-Kluser, 2022-09-02:
I have installed Manjaro in a brand new native machine and I have got the same error:
[image]
openFPGALoader version:
[image]

Chandler-Kluser, 2022-09-02:
I have also added the `99-openfpgaloader.rules` from the repository folder using the command:
```
$  sudo cp 99-openfpgaloader.rules /etc/udev/rules.d/
```
considering that my device access group is not `dialout` neither `plugdev` groups, but `root` as described below:
```
$ lsusb                                                                                          
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 2a7a:6a18 CASUE CASUE USB Keyboard
Bus 003 Device 003: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse
Bus 003 Device 010: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Bus 003 Device 002: ID 214b:7250 Huasheng Electronics USB2.0 HUB
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0025 Intel Corp. Wireless-AC 9260 Bluetooth Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ stat /dev/bus/usb/003/010                                                                      
    Arquivo: /dev/bus/usb/003/010
    Tamanho: 0         	Blocos: 0          bloco de E/S: 4096   arquivo especial de caractere
Dispositivo: 0,5	Inode: 1129        Ligações: 1     Tipo de dispositivo: 189,265
     Acesso: (0664/crw-rw-r--)  Uid: (    0/    root)   Gid: (    0/    root)
     Acesso: 2022-09-02 02:28:37.370181621 -0300
Modificação: 2022-09-02 02:28:37.370181621 -0300
  Alteração: 2022-09-02 02:28:37.370181621 -0300
    Criação: -
```
the only modification I did in `99-openfpgaloader.rules` was to edit the group id from `plugdev` to `root`, and added my own user to that group:
```
# Original FT2232 VID:PID
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="664", GROUP="root", TAG+="uaccess"
```
I ran:
```
$ sudo udevadm control --reload-rules && sudo udevadm trigger
$ sudo usermod -a $USER -G root
```
and then tried to flash to SRAM, but what I have got is:
[image]

Now my machine specs are:
[image]

trabucayre, 2022-09-02:
Ok: your previous error seems due the rules configuration (using **root** group may be not good for security).
Now problem seems related to communication: it's weird because openFPGALoader is working with the BL702 (I use it with tang primer 20k) but I think I have to bought a 9k and trying with a second computer I have with archlinux to see if it's related to the OS/kernel or something else...
Have you another computer to test? Some people have reported wrong/good behaviour according to hardware (mainly with the ch552 used for first tangNano board).

Chandler-Kluser, 2022-09-02:
I am aware of the risks of using root permissions in that case, I only want to know what is going on.

I have also tried to use the official gowin programmer but I could not detect the board either....

I will try to test it in another computer, but I am not so confident because I have got the same issue from an entire brand new machine with manjaro linux installed (I have also had problems with Ubuntu 22.04 LTS, too)

Chandler-Kluser, 2022-09-02:
Problem solved by using a second computer with MS Windows and GoWin Official Programmer as admin by erasing SRAM and embedded flash.

I could flash the [example project](https://github.com/sipeed/TangNano-9K-example/tree/main/led) in the embedded flash memory using the second PC.

The FPGA has successfully presented the led demo, so I went back to the original linux machine and used `openFPGALoader`...

This was the result I have got in the manjaro linux machine:
[image]

I did not make any configuration change from the last unsuccessful flash, just erased SRAM + embdFlash and come back again to original PC, I don't have idea what was the big deal so far to disable the flashing from linux.

but now it is working! Thanks to @trabucayre attention

closing the issue...

closed 2022-09-02 by Chandler-Kluser

trabucayre, 2022-09-02:
Maybe it's related to the reboot: when you update user group for a user you have to logout and login again... Or simply reboot the computer.
Thanks!

SelfhostedPro, 2024-07-26:
Is there not a way to erase this flash/resolve this issue without a windows device? I don’t have any available currently.

trabucayre, 2024-07-27:
Have you experimented the same issue recently? This issue is a bit old and outdated. Gowin's driver have received lot of improvement.

fedor-goncharov, 2024-08-09:
I had a very similar sequence of problems, I think it started with my incorrect rules for the USB port and at some point 
I got ```mpsee_read/mpsee_write()``` errors as above. I followed advice of @Chandler-Kluser : on Gowin website one can 
download "Programmer" soft (Educational Version) and it has commands Erase SRAM, Erase Flash memory. 

After erasing and setting rules for USB as advised on Sipeed website it works correctly now.

trabucayre, 2024-08-09:
Thanks for your feedback. I have to see why doing the same thing with openFPGALoader.

fedor-goncharov, 2024-08-16:
@trabucayre @Chandler-Kluser After two dense days I found that my Tang Nano 9k is flashed perfectly fine if .... 

I simply use the original usb-cable from the little box how it is shipped, otherwise - CRC errors / and further problems 
with FTDI drivers. So my advice would be try to simply change the USB cable to original one and ensure that the board is 
accurately isolated (in my case I pinned it to my breadboard). 

I have no idea what had happened .....

trabucayre, 2024-08-17:
Yes cable is an usual case of failure.
For isolated board: one possible thing (not limited to this board) may be related to the computer with or without ground pin. Also sometime USB interface contains an isolator (or something similar) just after usb connector. Using a short cable may reduces some ECM issue.