trabucayre/openFPGALoader #252: Error: Failed to claim FPGA device: mismatch between target's idcode and fs idcode

neuberfran, 2022-09-03:
Hi,

I stay trying this: https://trabucayre.github.io/openFPGALoader/vendors/gowin.html with my tangnano 1k by sipeed with issue:

Error: Failed to claim FPGA device: mismatch between target's idcode and fs idcode

Everything looks OK and the installation of openFPGALoader looks OK.

[image]
Can anyone help

trabucayre, 2022-09-03:
Your install seems ok. This error is displayed when you create a gateware for an FPGA and you try to load it into another, different, FPGAs. For example you have a project for **GW1N-2** and your target is **GW1N-1**.
Each model has a specific idcode (a unique number by model) and this information is available into the bitstream.
Could you use `-v` option to have more verbose ouput and recheck physical device?
Keep in mind there is two tangNano1K:
- original one with a **GW1N-1** and with a **ch552** as jtag interface
- new one with a **GW1NZ-1** and with a **bl702**.

So if you use an existing project for the original: risks are great to have a gateware configured to the wrong FPGA.

neuberfran, 2022-09-03:
@trabucayre Tks
https://gist.github.com/neuberfran/2645508bd4ac11f8c254d8cdf480281e
[]s
What about 100mA issue too
https://github.com/sipeed/Tang-Nano-examples/issues/5

trabucayre, 2022-09-03:
This issue is tangNano first generation related. If you have bought recently your board I think you have an bl702 interface.
According to your screenshot you use [Tang-Nano-examples](https://github.com/sipeed/Tang-Nano-examples)

But if you have new one you have to use
[TangNano-1K-examples](https://github.com/sipeed/TangNano-1K-examples)

neuberfran, 2022-09-03:
@trabucayre Solved. Nothing change in my board because already comes with this program installed(I think)

[image]

Yes. I have new one

trabucayre, 2022-09-03:
Great!
Most important thing is: openFPGALoader has passed idcode check. But I'm agree message is a bit cryptic and it's requires to be reviewed with more relevants informations!
Thanks!
Can I close this issue?

closed 2022-09-03 by neuberfran