trabucayre/openFPGALoader #255: test_sfl.svf redistribution

smoe, 2022-09-30:
The file https://raw.githubusercontent.com/trabucayre/openFPGALoader/master/test_sfl.svf is not covered by the Apache license and from how how I interpret
```
!Copyright (C) 2018  Intel Corporation. All rights reserved.
!Your use of Intel Corporation's design tools, logic functions 
!and other software and tools, and its AMPP partner logic 
!functions, and any output files from any of the foregoing 
!(including device programming or simulation files), and any 
!associated documentation or information are expressly subject 
!to the terms and conditions of the Intel Program License 
!Subscription Agreement, the Intel Quartus Prime License Agreement,
!the Intel FPGA IP License Agreement, or other applicable license
!agreement, including, without limitation, that your use is for
!the sole purpose of programming logic devices manufactured by
!Intel and sold by Intel or its authorized distributors.  Please
!refer to the applicable agreement for further details.
```
may not even be redistributable. I was preparing a Debian/Ubuntu package oy openFPGALoader. Would it seem reasonable to just omit that file from that package? Other thoughts?

Many thanks!

trabucayre, 2022-09-30:
This file may be perfectly omitted (It hasn't been used for a while and I've just remove this).
Thanks for your work to add openFPGALoader as Debian/Ubuntu package!

smoe, 2022-09-30:
Credits for spotting that go to the Debian FTPmasters.

trabucayre, 2022-10-03:
I good to you now?

smoe, 2022-10-03:
I did not get around to reuild and reupload - later this week :) Now that I know that the file can go, everything is easy. Thank you!

trabucayre, 2022-12-21:
I have created a new release: now this svf is no more present.

trabucayre, 2023-02-17:
I think this issue could be closed. `test_sfl.svf` has been removed from the repository.

closed 2023-02-17 by smoe