trabucayre/openFPGALoader #258: Add option to scan attached probes

cr1901, 2022-10-12:
[Context](https://libera.irclog.whitequark.org/openFPGALoader/2022-10-12#1665607174-1665608617;)

There should be an option to scan probes, just like there's an option to scan attached devices. At minimum, for FTDI devices, I think this should return the serial number so that 

Furthermore, `libftdi` failing to open a device _on Windows_ does not indicate a fatal error. There are multiple types of device drivers for FTDI devices on Windows, and a failure to open may indicate that a probe is using a driver that `libftdi` doesn't understand. Therefore, I believe scanning probes should have an option to keep going on error.

trabucayre, 2022-11-01:
Is it good to you? Can I close this issue?
Thanks

cr1901, 2022-11-01:
Yes, it's safe to close this issue, the new functionality does what I want :).

closed 2022-11-01 by trabucayre