trabucayre/openFPGALoader #259: GW1N-9C: flash CRC error

jeanbie0, 2022-10-13:
Hello,
I've CRC error when I try to write the flash of the FPGA.

I succeed to write the SRAM without any problem and the code is running.

```
openFPGALoader --unprotect-flash -f -v -c libgpiod --pins 65:28:20:19 ./code.fs

write to flash
found 1 devices
index 0:
        idcode 0x100481b
        manufacturer Gowin
        family GW1N
        model  GW1N(R)-9C
        irlength 8
File type : fs
Parse file Parse ./code.fs:
checksum 0xb8e4
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 712
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 1100481b
loading_rate: 0
pollFlag: 350a2
erase SRAM pollFlag: 35082
pollFlag: 35082
pollFlag: 35082
pollFlag: 310a2
Done
pollFlag: 31422
pollFlag: 310a2
erase Flash Done
pollFlag: 11422
write Flash: [==================================================] 100.00%
Done
pollFlag: 11422
CRC check : FAIL
b8e4 0000
displayReadReg 00015421
        CRC Error
        Memory Erase
        Non-jtag is active
        Gowin VLD
        Security Final
        POR
```

Thank you

trabucayre, 2022-10-13:
Everything looks good the `--unprotect-flash` is not required here because is related to external flash.
But it's weird, last time I have seen CRC equal to 0 is with a too short sleep before reading (ie bitstream not completly loaded).
But with libgpiod (slow) I'm not sure if it's the reason. This device (seems the same as sipeed tang nano 9k) has been validated by many people.
Is it a commercial board (ie easy to order) and it's possible to you to test with a jtag cable ?
Thanks

trabucayre, 2022-10-14:
It's interesting: I always take into account max freq because it's not possible to work with an higher rate, but not min freq.
I have to do tests to see if this requirement must __absolutely__ be true: at least to print a warning when not met.

jeanbie0, 2022-10-14:
It was this.

I create a new cables based on libgpiod cable. This cable "jetson-nano-gpio" is only usable with Jetson Nano (Tegra X1) because of the structure of internal memory. but it can be easily adapted for other CPU.

With this I can have a clock frequency of 1.2MHz. Still too slow for the datasheet but the writing in flash is working.

I can make a pull request if you want. [Fork](https://github.com/jeanbie0/openFPGALoader)

closed 2022-10-14 by jeanbie0