trabucayre/openFPGALoader #262: Xilinx flash programming fails using release

gregdavill, 2022-10-22:
## Summary
After installing openFPGALoader package in MSYS2, loading a bitstream into a xillinx's config SRAM works, but loading into FLASH fails.

```
$ openFPGALoader.exe -b qmtechKintex7 -c tigard -f fpga/qmtech_k7/iqmc510x_fpga.bit
write to flash
Board qmtechKintex7 has not default cable
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices

...

use: D:/a/_temp/msys64/mingw64/share/openFPGALoader/spiOverJtag_xc7k325tffg676.bit.gz
Error: fail to open D:/a/_temp/msys64/mingw64/share/openFPGALoader/spiOverJtag_xc7k325tffg676.bit.gz
Error: Failed to program FPGA: Error: fail to open D:/a/_temp/msys64/mingw64/share/openFPGALoader/sp
```
Note the file does exist, and has been installed.. But is located at `/msys64/mingw64/share/openFPGALoader/spiOverJtag_xc7k325tffg676.bit.gz`

## Detailed issue

The extra `D:/a/_temp/` prefix on the directory seems to originate from a hard coded path in the source code. `DATA_DIR` in a prefix for the bitfile.

https://github.com/trabucayre/openFPGALoader/blob/6a00466137ca8c20b519c7a5193b338f7e246935/src/xilinx.cpp#L315-L341

`DATA_DIR` is defined by CMAKE

https://github.com/trabucayre/openFPGALoader/blob/6a00466137ca8c20b519c7a5193b338f7e246935/CMakeLists.txt#L35

On the CI builds that end up as releases this value for MSYS2 builds becomes `D:/a/_temp/msys64/mingw64/share/`

trabucayre, 2022-10-22:
It's true: **spiOverJtag** bitstreams are installed into a directory based on `CMAKE_INSTALL_FULL_DATAROOTDIR` and code uses this path but if openFPGALoader (executable + bitstreams) is moved this path become wrong and writing a bitstream for **Xilinx** and **Intel** fails.
The question is to find the best way to introduce an alternate policy:
- at buildtime: allowing user to provides explicitly `DATA_DIR` and using this parameter instead of `CMAKE_INSTALL_FULL_DATAROOTDIR`
- at runtime: adding an arg to allow user to provides correct `DATA_DIR`

The first option seems better because openFPGALoader behavior is kept as now (it's always possible, of course, to add both solutions).

gregdavill, 2022-10-22:
The hardcoded path should work for default installs, right now it doesn't because cmake? Is passing in a windows path, which includes `D:/a/_temp/msys64` prefix which isn't needed when the program is running under msys2.

This page (https://www.msys2.org/wiki/Porting/) mentions this under "Hard-coded paths and relocation"
> Sometimes these paths are baked into shell scripts or pkg-config's .pc files. In that case, you should use sed in the PKGBUILD package() function to correct this back to the msys2 version of the path. 

Adding a arg for runtime would be useful as an additional feature.

trabucayre, 2022-10-22:
the path is absolute this why "D:/a/_temp/msys64" is used. In fact there is another approach by replacing `DATA_DIR` absolute path by relative path -> if structure is kept this must works everywhere.

Your PR looks good but I have no way to test So I have to assume it's works for you and resolves this issue.

Guru1904, 2022-10-22:
https://github.com/Xilinx/embeddedsw/issues/225#issue-1415371661
Can you help me with this, please????

closed 2022-10-23 by trabucayre

trabucayre, 2022-10-23:
@gregdavill Thanks again

@Guru1904 it seems out of context since you use xilinx tools. And it's a bit difficult to answer: it's a access to the PL: are you sure to have a gateware with the processing-system wrapper and something mapped to this address?

Guru1904, 2022-10-25:
@trabucayre Thanks a lot for responding,
I am fresher in the embedded software domain, 

I have been provided with the bit file(RTL design) by the hardware team and I have the application code to test it on the Zynq board(ZED700). I am able to write the data in this address but when I try to read the data, I get AP transition timeout. I am just trying to understand why it is causing errors and what may be the possible solutions. Meanwhile, I have cross-checked base addresses are correct or not.  I want to know will the RCC clock and reset caused this error ? (RTL Design). Just the possible scenarios which may cause this error.    THANK YOU

ottelo9, 2023-09-11:
I installed openFPGAloader in MSYS2 (pacman -S mingw-w64-ucrt-x86_64-openFPGALoader). 

$ openFPGALoader -V
openFPGALoader v0.10.0

If I try to flash I get the error:

```
$ openFPGALoader -c dirtyJtag --fpga-part xc6slx9tqg144 -f "FWUPD_64HD_BETA7_v2.2.0.bin"
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Open file DONE
Parse file DONE
use: D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx9tqg144.bit.gz
Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx9tqg144.bit.gz
Error: Failed to program FPGA: Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJt
ag_xc6slx9tqg144.bit.gz
```

I thought this bug was fixed in v.10 ?

PR #263 (2022-10-23): msys2: Fix windows paths
Fixes #262 

Following the example from msys2 (https://www.msys2.org/wiki/Porting/) "Hard-coded paths and relocation"
> Sometimes these paths are baked into shell scripts or pkg-config's .pc files. In that case, you should use sed in the PKGBUILD package() function to correct this back to the msys2 version of the path. 

The package step can strip the `D:/a/msys64/` or equivalent prefix that's added at compile time.

```
 -- Version installed from msys2 packages
$ strings /mingw64/bin/openFPGALoader.exe | grep mingw64/share
D:/a/msys64/mingw64/share/openFPGALoader/spiO [... cut]