trabucayre/openFPGALoader #268: Error writing to flash

Onicc, 2022-11-04:
Hi,

I write bitstream to RAM everything works fine, but write to flash I get an error.

```bash
openFPGALoader -c ft232 -b zybo_z7_20 -f led_twinkle.bit
write to flash
Board default cable overridden with ft232
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7z020clg400.bit.gz
Error: fail to open /usr/local/share/openFPGALoader/spiOverJtag_xc7z020clg400.bit.gz
Error: Failed to program FPGA: Error: fail to open /usr/local/share/openFPGALoader/spiOverJtag_xc7z020clg400.bit.gz
```

Host: Ubuntu20.04 amd64
Board: xc7z020clg400-2

trabucayre, 2022-11-04:
As mentionned in the documentation write to flash is not available through JTAG. The SPI Flash is connected to the PS part and can't be acceded from FPGA.
I'm agree the error is not explicit for this specific case and a check must be done at CTOR level with a better error message.

trabucayre, 2022-11-20:
with [commit](https://github.com/trabucayre/openFPGALoader/commit/13fa66bc2219f06341216487ab1eff3bf90b48bb) openFPGALoader will no more tried wrongly to write bitstream for **xc7z** devices.

trabucayre, 2022-12-01:
It's good to you? can I close it?

Onicc, 2022-12-05:
Yes, it's safe to close this issue, added error messages are more appropriate.

closed 2022-12-05 by Onicc