trabucayre/openFPGALoader #270: MachXO2 Internal Flash Programming using `.bit` files

cr1901, 2022-11-12:
Context from IRC channel:

```
[17:52:52] <cr1901> At some point I'll add MachXO2 Flash programming if no one has done it yet, but right now I've been preoccupied
[17:54:51] <trabucayre> Could you open an issue to that? It's a good todo list :)
[17:55:28] <trabucayre> MachXO2 flash is working (at least last time I have tried)
[17:56:27] <cr1901> internal flash?
[17:58:40] <trabucayre> with jed yup https://github.com/trabucayre/openFPGALoader/issues/31
[17:58:59] <cr1901> Ohhh right, I wanted .bit internal flash programming
[17:59:36] <cr1901> and seem to vaguely recall that "doing the right thing automatically based on file extension" gets tricky once .bit support is added
[18:01:34] <trabucayre> Yep currently combo .bit + machXO2 isn't supported.
```

`.bit` file programming is useful because it's what the open source flow (via `yosys`, `nextpnr`, and [`ecppack`]) emits; Lattice prefers emitting `.jed` files. Note that [`ecppack`] _can_ emit SVD files directly; I have never tried this functionality for MachXO2.

An example of `.bit` file programming that I've confirmed to work can be found in the [Tinyfpga-A-Programmer repository](https://github.com/tinyfpga/TinyFPGA-A-Programmer/blob/master/python/tinyfpgaa/tinyfpgaa.py#L1111-L1208).

[`ecppack`]: https://github.com/YosysHQ/prjtrellis/blob/master/libtrellis/tools/ecppack.cpp

trabucayre, 2022-11-26:
I have pushed required modifications to flash a **machXO2** with a `.bit` file.
Tested with my **machXO2EVN** (**LCMXO2-7000HE**)

trabucayre, 2022-12-01:
It's good to you? can I close it?

cr1901, 2022-12-01:
@trabucayre Yes, good to close. If there's problems, I'll make another issue.

closed 2022-12-01 by cr1901