trabucayre/openFPGALoader #271: JTAG Device ID code version field is required for some lattice parts

mini-ninja-64, 2022-11-14:
Hi, really like the work done by everyone for this project, it's so useful and a massive benefit to the open ecosystem!

I was working to write to a LCMXO2-2000HC through JTAG and found that although the chip was not recorded in the parts list it was instead recognised as a LCMX03LF-1300C.

As noted [here](https://github.com/trabucayre/openFPGALoader/blob/d5190a3bd42fa7dfcd42bff6110e7a3c5d9fec1b/src/part.hpp#L20) IEEE 1149.1 dictates the upper nibble should specify version, however maybe lattice decided to ignore this as the Lattice LCMXO2-2000HC has the id code: `0x012BB043`while the Lattice LCMX03LF-1300C has id code: `0xE12BB043`.

As such when the id code upper nibble is masked during the part lookup process an LCMXO2-2000HC will be mistakenly identified as a LCMX03LF-1300C.

As a workaround i just commented out the LCMX03LF-1300C entry and added an LCMXO2-2000HC entry in `part.hpp` and it worked flawlessly (⁠ﾉ⁠◕⁠ヮ⁠◕⁠)⁠ﾉ⁠*⁠.⁠✧.

I'm unsure on the best way to proceed with this but maybe part entries could have a way to mark if the version field is required or ignore it and fallback if multiple matching entries are found 🤔, thoughts?

trabucayre, 2022-11-14:
Thanks for your feedback and encouragement !
Yes some devices must have this nibble unmasked to avoid this type of issue. The algorithm seems a bit too much naive and must improved:
- first thing is to re-add the nibble for all LCMX03LF to have distinction between devices
- second thing seems to searchs for devices by using masked and unmasked version. If both are found -> select correct one.

I'm not really happy by the second point: maybe switching search masked and not masked. Have to try...

mini-ninja-64, 2022-11-17:
Thank you for the response!

That makes sense, I can have a look at doing this and raise a PR, if it would be helpful?

trabucayre, 2022-11-17:
Yes of course!
Since you have hardware it's maybe useful/easy to validated algorithm update.
Thanks!