trabucayre/openFPGALoader #272: Erasing/Flashing only <file size> part of the flash

TheAnimatrix, 2022-11-16:
It would be great in terms of speed and efficiency if only <file size> part of the flash at whatever offset was erased and flashed instead of the whole thing. Writing takes quite a bit of time on larger modules.

trabucayre, 2022-11-17:
It a bit strange because it's the case: before a write only sectors (and subsectors) impacted are [erased](https://github.com/trabucayre/openFPGALoader/blob/master/src/spiFlash.cpp#L357).
Flash access is always slow depending of size of datas to write: it must be done by [pages](https://github.com/trabucayre/openFPGALoader/blob/master/src/spiFlash.cpp#L367)

TheAnimatrix, 2022-11-18:
Oh so it does this already ? Interesting.
Also, is there any way i can force it to do what it does in QSPI ? I'm not sure if there's much to be gained though since erase/write have a lot of dead time for the commanding module anyways.

trabucayre, 2022-11-20:
It's depend on your device:
- lattice (ECP5/nexus), gowin, anlogic: flash access uses hardware functionalites and I'm not sure this it's possible.
- ice40: it's not possible due to the FTDI's limitations
- altera, xilinx: these ones need a custom bitstream (**spiOverJtag**) to bridge jtag and SPI. But to have something simple and generic it support only x1 SPI transfer.

Yes using QSPI may benefical for large write but for erase and wait for enable it's limited by the SPI flash.

closed 2022-11-21 by TheAnimatrix