trabucayre/openFPGALoader #278: What kind of preamble do I need to prefix into my bitstream?

francis2tm, 2022-11-28:
Hey!
When I try to flash my "image" (image contains a bitstream+ some memory contents I need stored in my flash memory) into the external flash memory of a ECP5 FPGA this error appears:

```
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file Wrong File 4a4a
Error: missing preamble

FAIL
Error: Failed to program FPGA: std::exception
```

I'm using a yosys to synthesize my bitstream, I have no idea what kind of preamble the file must have. I searched through the docs, I haven't found anything either. Maybe you can provide me with some pointers?

Thank you

francis2tm, 2022-11-28:
UPDATE: I just tried to load the bitstream only to RAM, and it works:
```
openFPGALoader -c ft232 build/gsd_orangecrab/gateware/gsd_orangecrab.bit
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file: DONE
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
Disable configuration: DONE
```
So, the problem was the other contents I want to put in the SPI flash as well...

Also, if I try to flash the bitstream only but into flash, this happens:
```
openFPGALoader -c ft232 -f build/gsd_orangecrab/gateware/gsd_orangecrab.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
flash chip unknown: use basic protection detection
timeout: 0 0 1000
0
wait: Error
write en: Error
Erasing: [==================================================] 100.00%
Fail
Refresh: FAIL
displayReadReg
        Config Target Selection : 0
        SPIm Fail1
        Preamble ERR
        EXEC Error
Error: Failed to program FPGA: std::exception
```

trabucayre, 2022-11-28:
For the first post I needs a bit more explanation: your `.bit` contains the bitstream + data ? I'm astonish because your command didn't provides a board name but I see `Board colorlight has not default cable` ?

For the second post: I assume you use an **orangeCrab** ? If so I have just tried with a *tigard* cable and my *orangeCrab*: the flash is seen. (*W25Q128* is already in `spiFlashdb`).
In fact the flash is not seen (idcode = 0).

francis2tm, 2022-11-28:
Exactly, if you're familiar with orangecrab and litex, my .bit includes gateware + firmware ;) And yes! I'm sorry, that command was actually `openFPGALoader -c ft232 -b colorlight -f build/image.bit`. But the output is the same if I don't specify any board. And since you don't have listed orangeCrab, I removed the `-b colorlight` to not assume I have this board. Sorry for the misunderstanding.

What can be causing the flash not being recognize? Maybe my connections to ft232?

francis2tm, 2022-11-28:
Very weird... I tried several times and in one of them the flash was recognized:
```
$openFPGALoader -c ft232 -f build/gsd_orangecrab/gateware/aa.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detected: Winbond W25Q128 256 sectors size: 128Mb
00000000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Refresh: DONE
```

trabucayre, 2022-11-28:
For lattice it's true, the board is only used to know onboard cable. When no default cable you have to specify it manually and -b is useless.
Seems something wrong with your bit (maybe a wrong concat since bit preamble is not found).
Last result is interresting: which cable you use ? You use fly wires? My tigard has a 1.27mm connector so connection is direct. I have already seen sometime unstable behaviour with fly wires.

francis2tm, 2022-11-28:
If the bitstream is alone, the flash works. But if in the same .bit I have the bitstream (gateware) + firmware, it doesn't work

Yes I literally use jumper wires... Maybe thjat's the problem?

trabucayre, 2022-11-29:
> If the bitstream is alone, the flash works. But if in the same .bit I have the bitstream (gateware) + firmware, it doesn't work

I have never tested to merge gateware+firmware: could you explain how to do that?

> Yes I literally use jumper wires... But I don't think that's the problem, since I can flash the bitstream just fine to SRAM. The problem is flashing into SPI Flash memory

Is it possible for you to test with a more robust connection? Just to check if you have stability improvement? I don't know why but sometime SPI flash is more sensible to glitches....