trabucayre/openFPGALoader #279: Gowin SRAM Flash: FAIL error with non-default user code setting in bitstream

jrobgh, 2022-12-02:
Nice work on this tool- I can report success flashing 2x GW1N-9 in a chain via FT2232D (running openFPGALoader for each index-chain). 

I noticed that applying a non-default user code in the Gowin project bitstream setting results in an 'SRAM Flash' error though the design runs fine.

Jtag frequency : requested 2.50MHz   -> real 2.00MHz
erase SRAM Done
Flash SRAM: [==================================================] 100.00%
Done
SRAM Flash: FAIL
578b 12345678

I am using the prebuilt windows binary from https://colognechip.com/downloads/cc-toolchain-win.zip

trabucayre, 2022-12-04:
It's a use case I never tried: usually `usercode` register is read to check maching between local `checksum` and the one into the FPGA.
By default if no `usercode` is specified it is filled with the `.fs` `checksum,` but it's not true for your specific case: I have to check if another register provides this information, or trying to match with `checksum` or `usercode`.
(And in fact I'm not sure this test is really relevant: *status register* has one bit to specify a CRC error.)

trabucayre, 2023-02-16:
Sorry for the long delay. I have updated gowin's class to check if usercode register match usercode/checksum area when this register doesn't match the computed checksum.
It's good to you?