trabucayre/openFPGALoader #281: Please add an option to direct specify bridge

UweBonnes, 2022-12-05:
Hello,

at the moment, the bridge file is looked up deciphered from the board or the part argumenr, but always in  DATA_DIR given at compile time. This requires recompile if user has no access rights to DATA_DIR and wants to use another bridge. I would appreciated either an explicit argument for the bridge file like -B (--bridge) arg or a search directory could be given. xc3sprog used -I for the bridge file, but naming in xc3sprog was not to consistant.

trabucayre, 2022-12-07:
True.
Until now I have always thinked users have access to this directory and could be copy bitstream but I logical to add a way to explicitely provides bridge to use (and bypassing detection based on directory and fpga_part).

trabucayre, 2022-12-10:
I have pushed [this option](https://github.com/trabucayre/openFPGALoader/commit/fb8c1a5f97014631ccf53a4d19e7480b4d89ee51) it's good to you?

UweBonnes, 2022-12-12:
Yes, the option is fine. Many Thanks!

closed 2022-12-12 by UweBonnes