trabucayre/openFPGALoader #284: [Request] Add support for Anlogic EG4X20

knicklicht, 2022-12-14:
Hello, I saw that this project already supports Anlogic EG4S20. I would be very interested in the X20 model. There are cheap LED receiver cards with this FPGA on it. I think many projects would benefit from a cheap board with a Ethernet and lots of I/Os. Are there any plans? What is needed to add a new FPGA + board to the project?

trabucayre, 2022-12-14:
Hi,
True: openFPGALoader has a basic support for **anlogic** FPGAs.
This [commit](https://github.com/trabucayre/openFPGALoader/commit/0855efb29f47e3c0c91a68331a489936405b7329) is an example for adding a new part (ie FPGA model) and a new board.
But I'm not sure  anlogic is alive: website is currently down, english page was never released (and last time I have checked has simply totally disapeared).
But anyway don't hesitate to open a PR to add this model and corresponding board. I can, of course, add those by myself but I need more details and to be sure openFPGALoader is working with it.

trabucayre, 2024-11-26:
I close this issue. Feel free to open new one or PR :)
Thanks

closed 2024-11-26 by trabucayre