trabucayre/openFPGALoader #288: found 0 devices

wdevore, 2022-12-31:
Hi, I'm trying to upload a bit-stream to a relatively new board [Schoko](https://github.com/machdyne/schoko) via Tigard. It's an ECP5 45K.

```
$ openFPGALoader -V

openFPGALoader v0.10.0
```
On the **left** is the Schoko and the **right** is the Tigard.
[image]

I can scan the Tigard and it is detected.

```
$ openFPGALoader --scan-usb
found 17 USB device
Bus device vid:pid       probe type      manufacturer serial               product
001 037    0x0403:0x6010 FTDI2232        SecuringHardware.com TG1106c8             Tigard V1.1
```

I can also run *detect* and it shows:
```
$ openFPGALoader --detect
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
```
However, when I try to **upload** I get:
```
$ openFPGALoader -v -c tigard /media/RAMDisk/hardware.bit
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 0 devices
Error: no device found
```

both ttyUSB0/1 devices are mounted. I even tried *--reset* and get:
```
$ openFPGALoader --reset
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Error: no device found
```

Any Ideas or suggestions?
I'm guessing that I need to make some sort of enhancement to openFPGALoader...perhaps?

Thanks

trabucayre, 2023-01-01:
According to [jtag pinout](https://github.com/machdyne/schoko#jtag-header) (a column is odd, second is even), its
| pin | pin |
|-----|------|
| TCK | TDI |
| TDO | TMS |
| 3v3 | GND |

but according to the picture you have connected as:

| pin | pin |
|-----|-------|
| TCK | TMS |
| TDI | 3v3 |
| TDO | GND |

trabucayre, 2023-01-11:
Any news?
Thanks

trabucayre, 2023-01-11:
Weird. I have a tigard and it's a great board. Unfortunately I haven't schoko board (and can't currently order one) to test the combo.

wdevore, 2023-01-12:
Great galactic exoplanets! **It worked!** I swear I tested it with the current wire configuration *confused*. LOL. I guess I don't need the usb-blaster after all. Sorry about the confusion-argh!

```
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
        idcode 0x1112043
        manufacturer lattice
        family ECP5
        model  LFE5UM-45
        irlength 8
File type : bit
Open file: DONE
Parse file: DONE
bitstream header infos
Part: LFE5U-45F-6CABGA256
idcode: 41112043
IDCode : 41112043
displayReadReg
        Config Target Selection : 0
        Done Flag
        Std PreAmble
        No err
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
userCode: 00000000
Disable configuration: DONE
displayReadReg
        Config Target Selection : 0
        Done Flag
        Std PreAmble
        No err
```
[image]

wdevore, 2023-01-12:
I must have had the wiring wrong. I had spoken with the Machdyne engineer and he indicated that the image I was following was label incorrectly.

The correct wiring is:

pin | pin
-- | --
TCK | TDI
TDO | TMS
3v3 | GND

<br class="Apple-interchange-newline">

closed 2023-01-12 by wdevore

trabucayre, 2023-01-12:
Great!
Wrong pinout is one of recurrent cause of failure :)
Don't hesitate to open a PR to add this board :)
Thanks!

wdevore, 2023-01-12:
I've never opened a PR before but I did review some of the PRs already
submitted. Not sure what I need to do.

Should I add a new <vendor>.rst file? Update some of the *.yml* files?

trabucayre, 2023-01-13:
To create a PR: you have to fork the repository, clone you fork, create a new branch, modify files, add+commit+push and github will offer you to create a PR.

To add this board: You have only to modify *doc/boards.yml* and *src/board.hpp* as done [here](https://github.com/trabucayre/openFPGALoader/commit/67563983f926f6c62c519fd9e96a0d71b358e066)
Thanks

wdevore, 2023-02-09:
Sorry for taking so long to create a PR. I was waiting for Machdyne's newer
Konfekt (Ecp5) based board and wanted to make sure it worked as well so I
could submit them both. 😏

trabucayre, 2023-02-10:
Don't worry! And thanks!