trabucayre/openFPGALoader #300: tangprimer20k flash failure

ZhekaS, 2023-01-26:
Hello. Just got my new Tang Primer 20K, verified it is working with Gowin programmer on windows. Unfortunately with openFpgaLoader I am facing an issue consistently. The flashing is stopping at ~48% and then it starts showing an error: "Error: ftdi_read_data in mpsse_read" indefinitely. After the process is interrupted, the board is in some bad state and any subsequent attempts to program it fail even earlier, and in order to recover I need to use the Gowin programmer. 
I am using the build from commit _b1c843acdb549c17c15525df649e02f88960d06b_
Here is the dump from the session:

```
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
File type : fs
Parse file Parse /redacted/TangPrimer-20K-example/DDR-test/LicheeTang20K_DDR_Test/impl/pnr/ddr3_ref_design.fs: 
checksum 0xaf49
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 2110
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 0000081b
loading_rate: 0
Jtag frequency : requested 2.50MHz   -> real 2.00MHz  
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
pollFlag: 60a0
erase SRAM pollFlag: 4080
pollFlag: 4080
pollFlag: 4080
pollFlag: a0
Done
pollFlag: 20
b 40 16 b read b40160b
Detail: 
Jedec ID          : 0b
memory type       : 40
memory capacity   : 16
EDID + CFD length : 0b
EDID              : 1640
CFD               : 0f 40 16 0b 40 12 0b 60 16 
9 40 6 b read 940060b
Detail: 
Jedec ID          : 0b
memory type       : 40
memory capacity   : 16
EDID + CFD length : 09
EDID              : 0640
CFD               : 0b 00 16 0b 40 16 0b 40 16 
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [=======================                           ] 45.51%Error: ftdi_Writing: [=========================                         ] 48.83%Error: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_read^C

```

trabucayre, 2023-01-27:
I have to check if there is not related to a regression in ftdi support: never seen this error with my tang primer 20k 
But could you provides more information about your env (ie your computer). I know BL702 used as usb<->jtag interface may be unstable (or not working on rpi4). Could you try with another cable? Do you use an USB switch?

Thanks

trabucayre, 2023-01-27:
Seems to have no visible issue.
Could you just paste dmesg after failure?
weird :-/

ZhekaS, 2023-01-27:
Will do, but I believe it is the same.
Could do anything with timing as my hardware is quite old? Let me know if I can enable any extra debug messages in compilation that might be helpful.

trabucayre, 2023-01-28:
My primary computer is ~10 years old. I'm have not really idea on how to see why it's not working: never found solution for ch552 device used for tang nano and bl702 has issues with linux on rpi4. So I suspect something internal to the jtag<->usb converter. But it's may be wrong.

trabucayre, 2023-02-16:
Hi,
Any news?
Thanks

bentwire, 2023-02-18:
I'm having the exact same issue. Where it fails seems to vary and sometimes I can make it succeed just by commenting out a line of code.  I am on a fairly modern laptop so I don't think its timing, just in case I hacked the gowin.cpp to set the clock to 100KHz and it still happens.  I think it may be bitstream specific as it hags in different places and I can make it go away by undoing my last change or comment out some other large piece of code (like change a read from the DDR to a fixed set of bytes when writing to a FIFO), sometimes commenting out the code does not make the hang go away but 'moves' it.

I have not figured out any pattern yet, but I can supply code and bitstreams if needed.

bentwire, 2023-02-18:
Actually... I take it back, this is slightly different... It hangs with flash but I tried sram and I get this (and it ends up running...?)

After stashing my clock change and updating to latest git:

sudo openFPGALoader --board tangprimer20k --verbose-level 2 impl/pnr/dk_video.fs 
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0x0000081b
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 1 devices
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
File type : fs
Parse file Parse impl/pnr/dk_video.fs: 
checksum 0x4ed2
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 2110
ProgramDoneBypass: ON
SPIAddr: 00fff000
SecurityBit: ON
idcode: 0000081b
loading_rate: 0
Jtag frequency : requested 2.50MHz   -> real 2.00MHz  
displayReadReg 00006020
	Memory Erase
	Done Final
	Security Final
pollFlag: 60a0
erase SRAM pollFlag: 4080
pollFlag: 4080
pollFlag: 4080
pollFlag: 4080
pollFlag: a0
Done
pollFlag: 20
pollFlag: a0
Flash SRAM: [=======================================           ] 76.48%pollFlag: 68e0
Flash SRAM: [==================================================] 100.00%
Done
pollFlag: 6020
Error key checkSum not found
CRC check : FAIL
Read: 0x0000777f checksum: 0x4ed2

displayReadReg 00006020
	Memory Erase
	Done Final
	Security Final

bentwire, 2023-02-21:
This occurs with 0.10.0 BTW.  I did read the troubleshooting guide.  Btw Awesome job on all the docs!

vpecanins, 2023-04-21:
Also, programming to external flash works correctly with this bitstream: https://github.com/sipeed/TangPrimer-20K-example/tree/main/UART

trabucayre, 2023-04-21:
I have retried with master branch and with *LicheeTang20K_DDR_Test* without issue :(
SRAM and flash works (also with verify).
I have a tangPrimer20k with dock lite but I doesn't use the cable provided with the board and it's plugged on thinkpad's dock.
I haven't any idea on how to reproduce your issue :(

wingrime, 2023-09-04:
Some bitstreams hung for me with this error, reverting commit 
https://github.com/trabucayre/openFPGALoader/commit/94d0ad6c851f691b27ed48884744dcaeb051e734 
works

trabucayre, 2023-09-05:
As mentionned previously this commit is unrelated to gowin class so the problem is between this commit and HEAD (this mean around 139 commits).
Could you provides hash you use or distro/package version to reduce amount of commits to check.
Thanks.

wingrime, 2023-09-09:
> As mentionned previously this commit is unrelated to gowin class so the problem is between this commit and HEAD (this mean around 139 commits). Could you provides hash you use or distro/package version to reduce amount of commits to check. Thanks.

I didn't rollback tree, I revert exactly that commit. And one more thing, `dmesg` shows sudden disconnect of bl702 without reverting commit
I think it's something with ft232 comparability with bl702

trabucayre, 2023-09-10:
disconnect is the usual behavior: when you plug an ftdi (or clone) **ftdi_sio** took an exclusive access to the peripheral. First thing done by the *libftdi* is to detach device (ie the driver lost access). Without this step user space application can't directly communicate with the peripheral.
When openFPGALoader stop, the driver is reattached so a new message about "new device detected" is shown in *dmesg*. But if openFPGALoader crash or is manually stopped this last step is bypassed and the device is not detected.

Commit mentioned (archlinux) has introduces an issue in JTAG state machine behavior (fixed by commit https://github.com/trabucayre/openFPGALoader/commit/61b59ce8271a65f38d608fb8337215cffed5587b).
And in fact I'm a bit perplex about the fact reverting commit mentioned at this thread beginning fixes Gowin flash programming since it's related to a class used only when a direct access to flash must be done (as required by ice40, some efinix boards and cologneChip).