trabucayre/openFPGALoader #302: [Request] Add support for openocd remote_bitbang protocol

jeremyherbert, 2023-02-06:
Hi,

OpenOCD has an adapter/driver called `remote_bitbang` which allows JTAG to be bitbanged over a TCP connection. It is useful for devices which are otherwise unaware of the JTAG protocol. For example, the [glasgow](https://github.com/GlasgowEmbedded/glasgow) project allows an openocd server to be created in order to communicate with various programmable devices.

Would it be possible to add support for this cable/protocol? The protocol is extremely simple, you can find the description here: 

https://github.com/openocd-org/openocd/blob/master/doc/manual/jtag/drivers/remote_bitbang.txt

Thank you!

trabucayre, 2023-02-06:
Hi,
Yes it's possible. This protocol seems not complex to implement. Maybe make sense to have both client and server support.

trabucayre, 2023-02-15:
Hi,
Have you an example or something similar to *glasgow* code: it's to have a base to develop support client support in a first time.
Thanks.

jeremyherbert, 2023-02-15:
Is this what you are looking for? It is some python code I wrote to convert the openocd data into GPIO accesses on a simple USB to GPIO device I have made:

```python
import time
import socket
import struct
import os
import shutil
import threading
import select

from enum import IntEnum, Enum
from typing import Optional

from polyglot_turtle import PolyglotTurtle, PolyglotTurtleXiao, PinDirection, PinPullMode

class JtagServer(object):
    def __init__(self, pt: PolyglotTurtle, tdi_pin: int, tdo_pin: int, tms_pin: int, tck_pin: int, host="localhost", port=33333):
        self._pt = pt

        if not 0 <= tdi_pin <= 3:
            raise ValueError("TDI pin out of range")
        if not 0 <= tdo_pin <= 3:
            raise ValueError("TDO pin out of range")
        if not 0 <= tms_pin <= 3:
            raise ValueError("TMS pin out of range")
        if not 0 <= tck_pin <= 3:
            raise ValueError("TCK pin out of range")

        self._tdi_pin = tdi_pin
        self._tdo_pin = tdo_pin
        self._tms_pin = tms_pin
        self._tck_pin = tck_pin

        self._pt.gpio_set_direction(self._tdi_pin, PinDirection.OUTPUT)
        self._pt.gpio_set_direction(self._tms_pin, PinDirection.OUTPUT)
        self._pt.gpio_set_direction(self._tck_pin, PinDirection.OUTPUT)

        self._pt.gpio_set_direction(self._tdo_pin, PinDirection.INPUT)
        self._pt.gpio_set_pull(self._tdo_pin, PinPullMode.NONE)

        self._host = host
        self._port = port

        self._terminate = threading.Event()
        self._run_thread = threading.Thread(target=self._run_thread_func, daemon=True)

    def start_server(self):
        if self._terminate.is_set():
            raise RuntimeError("Server has already stopped")

        self._run_thread.start()

        return self

    def stop_server(self):
        self._terminate.set()
        self._run_thread.join()

    def __enter__(self):
        self.start_server()

    def __exit__(self, exc_type, exc_val, exc_tb):
        self.stop_server()

    def _run_thread_func(self):
        with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as server_socket:
            server_socket.bind((self._host, self._port))
            server_socket.listen()

            read_list = [server_socket]
            while True:
                readable, _, _ = select.select(read_list, [], [], 1.0)
                for sock in readable:
                    if sock is server_socket:
                        client_socket, address = server_socket.accept()
                        client_socket.setblocking(False)
                        read_list.append(client_socket)
                    else:
                        data = sock.recv(1024)

                        if data:
                            response = self._process_data(data)

                            if response:
                                if response[-1] is None:
                                    sock.send(bytes(response[:-1]))
                                    break
                                else:
                                    sock.send(bytes(response))
                        else:
                            sock.close()
                            read_list.remove(sock)
                            break

                if self._terminate.is_set():
                    for sock in read_list:
                        if sock == server_socket:
                            continue
                        sock.close()
                    break

    def _process_data(self, data: bytes):
        output = []
        for cmd in data:
            cmd_str = chr(cmd)
            if cmd_str == "B" or cmd_str == "b":
                print("blink ignored")
            elif cmd_str == "R":
                if self._pt.gpio_get_level(self._tdo_pin):
                    output.append(ord('1'))
                else:
                    output.append(ord('0'))
            elif cmd_str == "Q":
                print("quit")
                # output.append(None)
                # break
            elif cmd_str in ["0", "1", "2", "3", "4", "5", "6", "7"]:
                tck = (int(cmd_str) & (1 << 2)) != 0
                tms = (int(cmd_str) & (1 << 1)) != 0
                tdi = (int(cmd_str) & (1 << 0)) != 0

                self._pt.gpio_set_level(self._tck_pin, tck)
                self._pt.gpio_set_level(self._tms_pin, tms)
                self._pt.gpio_set_level(self._tdi_pin, tdi)
            elif cmd_str in ["r", "s", "t", "u"]:
                print("reset ignored")
            else:
                raise ValueError("Got invalid command" + str(cmd))

        return output

if __name__ == "__main__":
    pt = PolyglotTurtleXiao()
    with JtagServer(pt=pt, tdi_pin=2, tdo_pin=1, tms_pin=3, tck_pin=0):
        print("jtag server started")
        while True:
            time.sleep(10)
```

The `self._pt.gpio_...` lines just set the GPIO pin state.

trabucayre, 2023-02-15:
Thanks for informations and code.
Which USB to gpio is it? Just to be able to reproduce your setup and have a base to start implementation.
Thanks!

jeremyherbert, 2023-02-16:
The USB device is a tool I developed for my students: https://github.com/jeremyherbert/polyglot-turtle-xiao

It uses the low cost board from here: http://wiki.seeedstudio.com/Seeeduino-XIAO/

Note that this device is not particularly fast - flashing an ECP5-12F takes about 50 seconds. The glasgow is much faster, but they are harder to get at the moment.

The python library to use the board can be found here: https://github.com/jeremyherbert/python-polyglot-turtle  or `pip install polyglot-turtle` - you can either use the code in the previous response, or use this example to run the openocd interface: https://github.com/jeremyherbert/python-polyglot-turtle/blob/master/openocd_jtag_fast.py (this passes the commands to the MCU rather than interpreting them in python). If you'd like to see exactly what the commands do, you can find the firmware implementation here: https://github.com/jeremyherbert/polyglot-turtle-xiao/blob/6135ec8e03812d8ec98ff66275096cd5d2241b42/src/rpc_openocd_jtag.c#L45

To flash an ECP5-12F, first connect TDO, TDI, TMS, TCK to the GPIO as per the line `with JtagServer` in the `openocd_jtag_fast.py` script near the end of the file. You can reorganise the pins and change them if you wish. The pin numbering can be seen in the picture at the bottom here: https://github.com/jeremyherbert/polyglot-turtle-xiao

Then, execute the `openocd_jtag_fast.py` script with the device connected to start the server. Then run `openocd -f flash.ocd`. An example `flash.ocd` file for flashing the ECP5-12F with a file `blink.svf` is below:

```
adapter driver remote_bitbang
adapter speed 1000
remote_bitbang port 33334
remote_bitbang host localhost
#remote_bitbang use_remote_sleep yes

reset_config none separate

jtag newtap ecp5 tap -irlen 8 -expected-id 0x21111043

init
scan_chain
svf -tap ecp5.tap blink.svf -progress  
shutdown
```

Also, just to let you know, I have submitted a patch to openocd which allows delays to be sent via this interface: https://review.openocd.org/c/openocd/+/7472 and I think it will be merged soon. This is what the `remote_bitbang use_remote_sleep yes` line is for (currently commented out). Without it, the ECP5 will load the bitstream correctly, but it will error out when verifying the write was successful because it doesn't wait for the FPGA to reboot properly. You will see an error like:

```
Error: tdo check error at line 28
Error:     READ = 0x0201f10
Error:     WANT = 0x0000000
Error:     MASK = 0x000b000
Time used: 0m39s592ms 
svf file programmed failed
```

The openocd patch adds 'd' to tell the remote host to delay for 1 microsecond and 'D' to delay for 1 millisecond. If you use the latest openocd with that patch, the error will go away.

trabucayre, 2023-02-16:
Thanks for details. I haven't this xiao board but I have an adafruit board with samd21 too: I suppose it's possible to adapt the code.
I think I have all I need to start this work.
Thanks

trabucayre, 2023-02-22:
I try to modify your code for my adafruit feather m0 but it's not a success... USB fails to be enumerated and no blink. I try a bit more and if remains uncessful I will order one official board to do the test.

jeremyherbert, 2023-02-22:
I am happy to donate you the cost of buying them, assuming it is like ~20 EUR or so for 2x units. Does that work?

trabucayre, 2023-02-23:
Thank for the proposition, but this board cost more or less 6 coffees so it's not too expensive :)
I try with my feather m0 because I have this one, it's based on the same chip and I have not really used before (and I'm curious to see why it's not working).
*glasgow* is announced to be shipped by the end of march, so in a near futur I will have too :)

trabucayre, 2023-02-25:
Finally my board is working after having modified two files:

- **CMakeLists.txt** to replace `-Og` by `-Os`
- **src/delay.h** to remove `inline` keyword.

I suspect something with the gcc's version I use (*12.2.1*).

Now I'm able to start `openocd_jtag_fast.py` and the FPGA connected is seen by *openOCD*. I'm consequently able to start developping client side. In a future I will try to modiffy a bit more the code to respect *feather* form factor but it's an other story :).

trabucayre, 2023-02-26:
Hi,
I have created a new branch and a [PR](#318).
I have, successfully, tested this implementation (SRAM) with an **ECP5-25k**.

jeremyherbert, 2023-02-27:
Great! I will try to test it in the next 24h or so.

trabucayre, 2023-03-01:
Any news? Thanks!

jeremyherbert, 2023-03-02:
Sorry, life got in the way :) 

I just tested it on an ECP5-12F (ECP5-Mini) with an SRAM flash and it works great! I Also tried `--detect` with an artix 35T and it detected correctly:

```
index 0:
	idcode 0x362d093
	manufacturer xilinx
	family artix a7 35t
	model  xc7a35
	irlength 6
```

And with an Altera MAX II EPM240:

```
JTAG init failed with: Unknown device with IDCODE: 0x020a10dd (manufacturer: 0x06e (altera), part: 0x10 vers: 0x0
```

(Looks like the MAX II is not supported by this tool, but the JTAG ID is correct for this part)

One small issue, the `--help` docs say that `--port` and `--ip` are for Xilinx only, but this would no longer be the case. Other than that, looks good to go!

trabucayre, 2023-03-02:
Thanks for your feedback.
Yep MAXII isn't supported (currently).
It's now merged and `--port` and `--ip` doc is fixed.

Thanks again

closed 2023-03-02 by jeremyherbert