trabucayre/openFPGALoader #304: Flashing M25P128 via SPI fails

bl0x, 2023-02-08:
openFPGAloader built from fresh git.

FPGA: Xilinx xc6slx150tcsg484
Flash chip: M25P128-VME6GB
SPI wires: 1

Flashing from Xilinx Impact works without a problem (using generated .mcs file).

I've compiled a new spiOverJtag bitfile for the FPGA model, with pinout according to the datasheet.
I'll add a copy of the relevant schematics of the device in a later comment.

This is the verbose output:

```
$ openFPGALoader -v -c digilent_hs2 --index-chain 1 --fpga-part xc6slx150tcsg484 -f exploder_bl.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 2 devices
index 0:
        idcode 0x6e5d093
        manufacturer xilinx
        family xa2c
        model  xa2c64a
        irlength 8
index 1:
        idcode 0x4403d093
        manufacturer xilinx
        family spartan6
        model  xc6slx150T
        irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/02/08
design_name: exploder_bl.ncd
hour: 11:33:07
part_name: 6slx150tcsg484
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: /usr/local/share/openFPGALoader/spiOverJtag_xc6slx150tcsg484.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
0 0 0 0 read 0
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
0 0 0 0 read 0
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
timeout: 0 0 0
0
wait: Error
write en: Error
Erasing: [==================================================] 100.00%
Fail
```

When running from Impact, it uses the bundled SPI Jtag core:

  /opt/Xilinx/14.7/ISE_DS/ISE/spartan6/data/xc6slx150t_spi.cor

I've zipped it up and disguised it for openFPGALoader by giving it the appropriate name/path and tried again:

```
$ openFPGALoader -v -c digilent_hs2 --index-chain 1 --fpga-part xc6slx150tcsg484 -f exploder_bl.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 2 devices
index 0:
        idcode 0x6e5d093
        manufacturer xilinx
        family xa2c
        model  xa2c64a
        irlength 8
index 1:
        idcode 0x4403d093
        manufacturer xilinx
        family spartan6
        model  xc6slx150T
        irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/02/08
design_name: exploder_bl.ncd
hour: 11:33:07
part_name: 6slx150tcsg484
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: /usr/local/share/openFPGALoader/spiOverJtag_xc6slx150tcsg484.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
ff ff ff ff read ffffffff
Detail: 
Jedec ID          : fe
memory type       : 2a
memory capacity   : c4
EDID + CFD length : fd
EDID              : fa3b
CFD               : 0a 77 f0 9f 31 45 ff 80 00 00 00 00 af d9 7a aa aa 00 00 05 b1 45 ff bf fe 00 00 65 92 87 3a aa aa 00 00 05 b1 45 ff bf fe 00 00 bd 92 87 3a aa aa 00 00 08 b1 45 ff bf fe 00 00 04 b7 d9 7a aa aa 00 00 04 b1 45 ff bf fe 00 00 08 18 ef 3a aa aa 00 00 80 00 00 00 0c 00 00 00 0d b1 45 ff bf fe 00 00 03 b1 45 ff bf fe 00 00 00 af d9 7a aa aa 00 00 10 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 01 b1 45 ff 80 00 00 00 0d a0 ef 3a aa aa 00 00 09 8d b9 7a aa aa 00 00 00 00 00 00 00 00 00 00 07 31 45 ff bf fe 00 00 02 e7 a9 7a aa aa 00 00 09 b1 45 ff bf fe 00 00 29 6f 6b 3a aa aa 00 00 66 66 66 66 66 66 66 66 02 e7 a9 7a aa aa 00 00 0d b1 45 ff bf fe 00 00 39 2f 6b 3a aa aa 00 00 02 e7 a9 7a aa aa 00 00 eb b1 45 ff bf fe 00 00 07 b1 45 ff bf fe 00 00 48 8f 
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
timeout: 0 0 0
0
wait: Error
write en: Error
Erasing: [==================================================] 100.00%
```

This gives some values, but I don't think that they make any sense. What do you say, is this method even supported?

I've also noticed the `spiOverJtag/xc6` directory. What is the purpose of that? Just a reimplementation in VHDL? I've also tried to produce a spiOverJtag bitfile from that directory (with appropriate device parameters), but same as above.

It would be really nice to get this working!

bl0x, 2023-02-08:
Connection of SPI flash chips (two!) to a CPLD, which should take care of relaying the signals and choosing between one or the other flash chip. Since flashing works from Impact, I assume that the CPLD does the right thing.

- sck: CCCLK
- sdi_dq0: SPI_SI
- sdo_dq1: SPI_Q
- wpn_dq2: not used (for SPI)
- hldn_dq3: not used
- csn: SPI_CS

[image]

Connection of FPGA to CPLD:

[image]

bl0x, 2023-02-08:
How does one choose between SPI / QSPI?

My constraints file looks like this:

```
CONFIG VCCAUX = "2.5";

NET "sdi_dq0"  LOC = AB17 | IOSTANDARD = LVCMOS33;
NET "sdo_dq1"  LOC = Y17  | IOSTANDARD = LVCMOS33;
# NET "wpn_dq2"  LOC = V13  | IOSTANDARD = LVCMOS33;
# NET "hldn_dq3" LOC = W13  | IOSTANDARD = LVCMOS33;
NET "csn"      LOC = AB5  | IOSTANDARD = LVCMOS33;
NET "sck"      LOC = W17  | IOSTANDARD = LVCMOS33;
```

trabucayre, 2023-02-09:
When openFPGALoader reads a jedec_id = 0 or 0xff it's because it can't communicate with the flash (and something is wrong during second try it read 0xffffffff but doesn't stop immediately).
subdirectory *xc6* is a residual and must be removed.

In fact I have never tried to access/program external flash when the jtag chain contains more than one device. Your board is available to order somewhere?
I have to find a way to create a complex jtag chain but, unfortunately all my xc6s have an onboard jtag interface...So it isn't possible to chain more than one FPGA

bl0x, 2023-02-15:
I've spent yesterday afternoon with @UweBonnes on this problem, and we've come to the conclusion that I need to look into the CPLD code to make sure that the CS for the flash chips is properly controlled by the FPGA and that there is no (additional) interference from the CPLD.

trabucayre, 2023-02-15:
I have to find a way to reproduce your issue if you can't find by yourself.
My first idea was an issue with spiOverJtag implementation when more than one FPGAs into the chain (it's the case for altera) but since you can't retrieve jedec id I'm not sure if it's your case (but maybe it also an issue you will discover after fixing current issue).
Thanks for your feedback.

bl0x, 2023-02-15:
I'm currently having more strange issues with one particular board of this kind, behaving differently from the others. Also I'm trying to get the original source code for the cpld, since that also drives the CS signals for the flash.

Would any changes be necessary to talk to this particular model of flash chip, or have you used one of these before?

Also, let me ask again: how does one differentiate between SPI and QSPI mode? I don't see anything related to that in the spiOverJtag code.

There's a second connector on board (not accessible from outside) as well that let's me connect to the chain in a slightly different way and I haven't tried that one yet with openFPGAloader. Will try soon.

trabucayre, 2023-02-15:
As far I remember I have never used this specific chip and since it's marked as obsolete and not available I can't order one. I have to read datasheet to see if it's differs to ones already supported.
for SPI vs QSPI: spiOverJtag support SPI: it's because all eeprom supports this mode and to have something at much as possible generic I have implemented this gateware using common mode (sometime only MISO and MOSI are connected). I'm agree it's slower than QSPI but it's the only mode sure to be available.

bl0x, 2023-02-15:
Ah, I see, so it is *always* SPI, even if some FPGA constraints also set d2, d3. These are then only used as wpn holdn?

trabucayre, 2023-02-15:
Yes. D2 & D3 are only used as data when chip is configured for non SPI mode. spiFlash class uses only SPI instructions to be ""universal"".

bl0x, 2023-02-16:
The chip appears to be [in stock (octopart)](https://octopart.com/m25p128-vme6gb-micron-20464300).
Datasheet is [here](https://media-www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/m25p/m25p_128.pdf).

I've spent another night trying to get this to work, to no avail. I'll still make a pull request with the development so far. I've also had a look in the xilinx, spi and jtag code and could not find any obvious problem. I could possibly put a logic analyser on the JTAG chain to see what Impact does differently, unless the spiOverJtag bitfile is already wrong.

Does the spiOverJtag work fine for other Spartan6 chips / boards?

One thing I noticed is that there is no STARTUP_SPARTAN6 primitive instantiated. It is however different from e.g. STARTUPE2 for series 7 devices, so may not be needed at all.

trabucayre, 2023-02-16:
Thanks  for the link: I will order a couple (not easy to solder without correct tools...) and have a closer look to the datasheet.
Yes spiOverJtag works with others boards based on spartan6: I used my pipistrello successfully and dumped and rewrite flash for [SPEC150](https://ohwr.org/project/spec/wikis/home).
I will try with boards I have at home to create a complex Jtag chain to see spiOverJtag behaviour is this situation...

trabucayre, 2023-03-04:
Sorry for the delay.
I have received a board based on xc6slx16ftg256: flash is correctly detected.
I have tried using octopart to order a couple of this chip: for most of store a request for quotation is required (and I have never received an answer: maybe because I have asked for a too few volume), one store allows a direct order (but with a price > 100$ for 2 component it seems a bit expensive).

One thing not really clear after re-reading the excerpt of schematic: CS seems driven by the CPLD and not connected to the FPGA.

trabucayre, 2023-03-04:
My brain sometime is slow: spi flash for the board I have received is a M25P16: more or less the same chip but small than your...