trabucayre/openFPGALoader #309: Issue with dirtyJtag cable

machdyne, 2023-02-14:
Hi. I'm trying to get [pico-dirtyJtag](https://github.com/phdussud/pico-dirtyJtag) working as an openFPGALoader cable to program an ECP5 board.

```
$ sudo openFPGALoader --cable dirtyJtag --detect
getVersion: read: usb bulk read failed -7
Jtag frequency : requested 6000000Hz -> real 6000000Hz
writeTDI: read: usb bulk read failed -7
JTAG init failed with: Unknown device with IDCODE: 0x0d422991 (manufacturer: 0x4c8 (), part: 0x6a vers: 0x0
```

When using a usb-blaster cable:

```
$ sudo openFPGALoader --cable usb-blaster --detect
index 0:
	idcode 0x1112043
	manufacturer lattice
	family ECP5
	model  LFE5UM-45
	irlength 8
```

I've tried programming the pico-dirtyJtag firmware to a [custom board](http://github.com/machdyne/werkzeug) as well as an official Pico board.

I've tried on one computer with libusb-1.0.23 and another computer with libusb-1.0.25.

I've built openFPGALoader from the latest version of this repo.

I have also opened an [issue with pico-dirtyJtag](https://github.com/phdussud/pico-dirtyJtag/issues/10) and there may be some additional details there.

Any ideas what the problem might be?

trabucayre, 2023-02-14:
Current dirtyJtag support was done for original firmware (stm32) and I'm not sure if someone has tested with pico variant.
ASAP(this evening) I will flash this firmware to an official pico board (I haven't Werkzeug)  I have at home to adding (if required) support for this variant.

trabucayre, 2023-02-14:
I can confirm: I have same issue.
First thing: reprogramming a **bluepill** with dirtyJtag to see if it's a global issue with this procol, something about libusb or something specific to pico board (maybe *pico-sdk*).

trabucayre, 2023-02-15:
As mentionned in [this message](https://github.com/phdussud/pico-dirtyJtag/issues/10#issuecomment-1430868790) dirtyJtag's firmware is working when using **pico-sdk** 1.4.0 release (and with a`git submodule update` after the `checkout`). It seems this issue is related to the **sdk**'s or **TinyUSB** changes after this tag.

closed 2023-02-17 by trabucayre