trabucayre/openFPGALoader #315: Missing link to firmware project for ch552_jtag cable

hansfbaier, 2023-02-20:
I found two projects while googling:
One firmware worked, but did not match the VID/PID,
and the other one matched, but the firmware did non even enumerate
in USB. What is the correct firmware URL?
Could that be included in the documentation?

trabucayre, 2023-02-20:
Hi,
it's already included [here](https://github.com/trabucayre/openFPGALoader/blob/master/doc/cable.yml#L32) and corresponding githubio page.
But this firmware is more or less my nighmare: it's unstable. This one is used for the tangNano board (first release) and I have lot of error when I try to write into the flash (an issue is opened about that).

hansfbaier, 2023-02-20:
Oh thanks, yes I tried that one, it doesn't even enumerate on my machine.
Thanks!
So Raspberry Pi pico is then the way to go for the cheapest JTAG>