trabucayre/openFPGALoader #317: issues loading Efinix Ti60

te-johan, 2023-02-23:
i've been trying to load Efinix Ti60 with openFPGALoader but have not been successful. Using litex or the efinix flasher works fine.

I run it with the following command `openFPGALoader -b titanium_ti60_f225_jtag efinix_platform.hex`
I've looked at the jtag activity and the transfered data looks correct. the efinix script do send some additional instructions after ENTERUSER and also seems to poll the device status afterwards.

trabucayre, 2023-02-23:
Thanks to point this issue.
I will reread efinix script and datasheet to try to fix it.

trabucayre, 2023-02-26:
I have received a board yesterday and fixed a serie of issue.
irlen was wrong, (4 instead of 5), .bit format must be used instead of hex and I have added the required Jtag states.
With the [commit](https://github.com/trabucayre/openFPGALoader/commit/567d854edf4be00096445508e44da0481907db4a) I am able to load bitstream into flash with a quite decent delay.

Unlike the Trion family: with the Titanium family a bridge is required to have access to the SPI flash through Jtag. It's my next task: adapting *spiOverJtag* to these FPGAs.

te-johan, 2023-02-27:
well done. with the new changes i'm able to load the fpga.
however the jtag device id is incorrect part.hpp says 0x00660a79 but the correct id should be 0x10660a79
does the board you tested with report 0x00660a79?

[image]

trabucayre, 2023-02-27:
Hi!
Great to see it's works also for you.
About idcode as mentioned [here](https://github.com/trabucayre/openFPGALoader/blob/master/src/part.hpp#L20) the higghest nibble (revision part) is always masked to avoid to have more than one entry for a device with more than one revision (and `detectChain` also mask this part during scan). But it's not always true and maybe it make more sense to mask this nibble only when it's relevant.

te-johan, 2023-02-27:
I see. now when i change it back i see that this change is actually not needed.
I added it to try and mitigate another issue I'm seeing. After powering up of the fpga the device id is 0x1066**4**a79 and not 0x0066**0**a79. However by manually resetting the board it works.

trabucayre, 2023-02-27:
Look like a glitch or something like this.
I have to test with the board I have to see if the same thing is visible (or not)

trabucayre, 2023-04-20:
Any news?
I have a [new branch](https://github.com/trabucayre/openFPGALoader/tree/efinix_jtagspi) with spiOverJtag support.
Thanks