trabucayre/openFPGALoader #319: Clarify `--offset` description

mole99, 2023-02-27:
The `--offset` argument is described as follows

```
  -o, --offset arg              start offset in EEPROM
```

But I used it successfully to store my firmware in flash at a specific offset with the iCEBreaker (ice40_generic) and ULX3S.

According to #45 I am not sure if this is now officially supported. But if it is, perhaps clarify that `--offset` can also be used for specifying the offset in flash :)

trabucayre, 2023-02-27:
Yes. It's true the description is bad.
What do you think If I replace *EEPROM* by *Non volatile Memory* ?

I have to reread #45 but I think this issue is completed.

mole99, 2023-02-27:
That would definitely be better!

Other programs describe similar functionality like this:

iceprog
```
  -o <offset in bytes>  start address for read/write [default: 0]
                          (append 'k' to the argument for size in kilobytes,
                          or 'M' for size in megabytes)
```

fujprog
```
  -f ADDR	Start writing to SPI flash at ADDR, optional with -j flash
```

Are Flash and EEPROM the only two cases where this argument is used? Maybe be a bit more verbose and list them there?

trabucayre, 2023-02-27:
I propose a mix
```
-o, --offset arg          Start address (in bytes) for read/write into
                                non volatile memory (default: 0)
```

Currently yes there is only eeprom or spi flash.

mole99, 2023-02-27:
Sounds good to me :)

trabucayre, 2023-02-27:
Thanks for your remarks and point of view!