trabucayre/openFPGALoader #324: Unknown command type 79 when parsing ECP5 bit file

jeremyherbert, 2023-03-04:
I am trying to build and flash the example here: https://github.com/orangecrab-fpga/orangecrab-examples/tree/main/litex

When I try to flash the file using `openFPGALoader -c orbtrace orangecrab.bit`, I get the following output:

```
Found 1 compatible device:
	0x1209 0x3443 0x3 Orbtrace
Open file: DONE
Unknown command type 79.

Parse file: FAIL
Error: Failed to program FPGA: std::exception
```

I am using the latest git master version of this tool. Please also see the bit file attached.
[orangecrab.bit.zip](https://github.com/trabucayre/openFPGALoader/files/10887319/orangecrab.bit.zip)

trabucayre, 2023-03-04:
True:
with this gateware `ecppack` is called with `--spimode qspi`. This option adds instruction `0x79` (SPI_MODE) followed by one byte (spi mode operation), followed by 2 blank bytes.
Fixed by [this commit](https://github.com/trabucayre/openFPGALoader/commit/c35f21a5be17e873ca5eca6894f12d1cffbaee68)

jeremyherbert, 2023-03-04:
Thanks! I can confirm that it is fixed by that commit.

closed 2023-03-04 by jeremyherbert

trabucayre, 2023-03-04:
Thanks for your feedback and to point this error!