trabucayre/openFPGALoader #325: External flash bulk erase fails on ECP5

jeremyherbert, 2023-03-05:
Hi,

When I try to use the `--bulk-erase` function with an ECP5 12F, the external flash erase fails:

```
$ openFPGALoader -c orbtrace --bulk-erase
Found 1 compatible device:
	0x1209 0x3443 0x3 Orbtrace
bulk_erase: Enable configuration: DONE
SRAM erase: DONE
Detected: Winbond W25Q128 256 sectors size: 128Mb
Done
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	SPIm Fail1
	Preamble ERR
	EXEC Error
```

If I power cycle the device, I can see the flash is definitely erased because the FPGA doesn't load a bitstream from the flash.

Also, if I try to flash some data which is not a bitfile at an offset, I get a similar error:

```
$ openFPGALoader -c orbtrace --write-flash --offset 0x40000 bios.bin
write to flash
Found 1 compatible device:
	0x1209 0x3443 0x3 Orbtrace
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detected: Winbond W25Q128 256 sectors size: 128Mb
00040000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	SPIm Fail1
	Preamble ERR
	EXEC Error
Error: Failed to program FPGA: std::exception
```

If I write a bitfile to SRAM and then try to write to the external flash at an offset like above, I still see the same error. The only way to make the error go away is to write a bitfile to the external flash, and then write data to my binary offset.

```
$ openFPGALoader -c orbtrace --write-flash --offset 0x40000 bios.bin --verify
write to flash
Found 1 compatible device:
	0x1209 0x3443 0x3 Orbtrace
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Detected: Winbond W25Q128 256 sectors size: 128Mb
00040000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
Refresh: DONE
```

trabucayre, 2023-03-05:
Hi,
It's a fails without being it :-/ 
Each time openFPGALoader access to the SPI flash it ask the FPGA to reread configuration from the SPI flash (when you flash a bitstream it make sense). But since you have erased the flash ECP5 will read a long sequence of '1' trying to check checksum, ... and will fails because it's not a valid bitstream.

Now the question is: how to avoid to displays false error message like here (for bulk erase I think it's "quite" easy but for "normal" operation the only way is to read flash's start to check if it's blank or not.

Thanks to point this behaviour!

jeremyherbert, 2023-03-05:
Thanks for your reply and explanation. Yes, I figured it was something like this as the flash is definitely erased.

trabucayre, 2023-03-05:
Thanks for your feedback.
With this [commit](https://github.com/trabucayre/openFPGALoader/commit/4a6e6f0d350adb662d9bae426bf78ed5e09985ee) error is only be printed when `loadConfiguration` fails and the flash isn't blank (ie first 256 bytes != `0xff`)

jeremyherbert, 2023-03-10:
Sorry for the delay in testing this, I just gave it a go and it fixes the problem. Thanks again.

closed 2023-03-10 by jeremyherbert